solvespace/solvespace #1012: Bug: Can not drag the datum point sometimes

zhulaojianke, 2021-04-14:
SolveSpace version 3.0~f4ad8205
Operating system: Win7 Ultimate Edition sp1

During I learing the tutorials at the https://solvespace.com/tutorial.pl , there is 6 video tutorials, in the No.4 video, just as shown below,
[image]

at 3:24, the teacher dragged the datum point, and then the circle and the sketch-in-plane was moved with the drag, just as shown below.
[image]

But I can not drag the point to move in SolveSpace version 3.0~f4ad8205! The svls doc is here.
(https://github.com/solvespace/solvespace/files/6310333/can.not.drag.the.circle.by.the.datum.point.zip)

I have tried the same operations in SolveSpace version 2.3-7c1ca460, the result is very OK! I can drag the datum point and circle pretty well in SolveSpace version 2.3-7c1ca460. **So it's must a bug in the SolveSpace version 3.0~f4ad8205.**

ruevs, 2021-04-14:
This is true. The reason is that at the center of the circle there are two points - the center of the circle and the datum point in `g002` that the plane is based on. When you click the sport (or drag it) in the old (2.3, 2.0) versions the point in the first (`g002`) group is selected. In the newer versions the point in `g003` is selected.

The behaviour most probably changed in commit f9529916c491387 . I can not test it now but @phkahler probably can confirm. You can read through the issues ([161](https://github.com/solvespace/solvespace/issues/161), [174](https://github.com/solvespace/solvespace/issues/174) ) mentioned in that commit to find out why it was changed.

One can reproduce the "new" behavior in 2.0 by hiding `g002` (unchecking the check box for it) - then the center of the circle in `g003` will be selected and it will be impossible to move the point.

In 3.x to move the point while `g003` is active you either need to temporarily hide `g003` (the circle will disappear and you will be able to move the point in `g002`) or switch to `g002`.

phkahler, 2021-04-14:
The ability to drag a point in a previous group is a convenience feature. You still need to select the prior group if you want to constrain a point in that group.

This feature - the ability to drag points in prior groups - was causing other problems.

For plain selection, priority is given to entities in the current group, followed by earlier groups. For dragging, the same priority is used, but points that are known to be undraggable are ignored, which allows dragging the underlying sketch of an extrude group for example.

There is a new function I think called CanBeDragged() which returns False for entities that are known to be tied to (directly copied from) one in a prior group.

We do not currently recognize newly constructed entities with points-coincident constraints as undraggable, so it tries to drag the point in the current group but can not due to the constraint.

Overall the change is an improvement, but some cases that could be dragged before can not. The logic is simpler now and someone with a good understanding of constraints (not me yet) could get that function to return False in these cases to "fix" the regression.

zhulaojianke, 2021-04-14:
OK, I hid the g003 after extrude (g004) and got the same effect as the tutorial. I also feel that the logic of the version 3.x is clearer from both of your explanations. :)

PR #1330 (2023-01-18): Don't try to drag points with pt-coincident constraint to a previous …
…group. Fixes #1012 and makes dragging as done in one of the tutorials possible again.
commit 4a34253a (2023-01-18): Don't try to drag points with pt-coincident constraint to a previous group. Fixes #1012 and makes dragging as done in one of the tutorials possible again.
closed 2023-01-18 by phkahler