solvespace/solvespace #1013: draw a ring use the lathe tool

zhulaojianke, 2021-04-14:
SolveSpace version 3.0~f4ad8205
Operating system: Win7 Ultimate Edition sp1

When I draw a ring, first draw a circle and a line on the plane, select the line and then click "New group rotating actives sketch", it appears as below:
[image]
It doesn't produce a solid entity.

The Revolve tool also has this problem.

I can see from YouTube videos that it looks like anyone else can draw a ring this way. 
https://www.youtube.com/watch?v=X9byxcHuins
But I have tried version 3.0-f4ad8205 and version 2.3-7c1ca460, all can not do this matter.   :(

So in order to draw a ring, I must draw one shape like below:
[image]
and then rotate it:
[image]
and then draw another one shape:
[image]
rotate and difference:
[image]

This is a very complicated operation! 

Am I really the only one with this problem?

And hope to improve the lathe tool and the revolve tool.

phkahler, 2021-04-14:
Make the rotation axis a construction line. Select it and press G. Then do your lathe or revolve.

The problem is that your sketch is not a closed contour. Construction lines don't count for that.

zhulaojianke, 2021-04-14:
Ah...really did I misunderstand!

One last stupid question, I promise this is the last one. How to fillet multiple edges at the same time? For Example, all the edges around a cubic. It seems that the "tangent arc at point" only could be use in 2D sketch, and no command can be used on solid edges.
So I also used the dumbest way to make multiple chamfered edges:
[image]
As you can see after the round Angle, the vertex of the cubic is sharp, not smooth.
Like this:
[image]

In OpenSCAD, there is a command "Minkowski" can do this work well, and how to implement such chamfering in Solvespace?

phkahler, 2021-04-14:
That can be done by using the Revolve tool to make some of the corners. I also used a rotate group to copy the corner 4 times in the attached sketch. Have a look. The green points from the original sketch can still be dragged to change the size. All curves share the same radius.
[image]
[rounded_cube.zip](https://github.com/solvespace/solvespace/files/6313242/rounded_cube.zip)

phkahler, 2021-04-14:
One more tip for your lathe. You can also select a point and one of those little grey arrows to define the lathe axis. The arrow pointing down in your original image would have worked instead of drawing a line.

zhulaojianke, 2021-04-14:
I have tried my best...
[image]
but there must be an easier way... I did this in the same clumsy method.

@phkahler Can you tell me about your drawing process? If it's not convenient to say it here, you can send me an email, zhu_xinyan@yeah.net.

phkahler, 2021-04-15:
@zhulaojianke did you open the sketch I attached in a zip file? Start at the first group - check the box in the text window for it. Then move through each group.

zhulaojianke, 2021-04-15:
Oh, sorry, I just found it.

zhulaojianke, 2021-04-15:
Your method is much more efficient than mine.

But filling in a feature along a path is still cumbersome to implement now. I think that Solvespce already have lathe and revolve and helix commands and if the Solvespace has a stretch along the path function, then all the three commands can be merged in it, and it's more flexible. 
Then I can stretch out an entity by simply drawing a feature pattern and an curve path with its endpoint perpendicular to the workplane where that pattern is located. 
When the curve is round, closed and in a flat plane, then the lathe command is done. 
And when the curve is round, opened and in a flat plane, then the revolve command is done. 
And when the curve is spiral extened with constant diameter, the helix command is done.

I hope you can take my suggestion into consideration.

ruevs, 2021-04-15:
There is an issue for loft already:
https://github.com/solvespace/solvespace/issues/439

closed 2021-04-15 by ruevs