solvespace/solvespace #1014: Add Export Options

dustinhartlyn, 2021-04-15:
When developing a design for a laser cutter, I export and test cut parts dozens of times. This makes every step in the export process seam tedious because I may be doing it every couple of minutes. In CorelDraw I even went through the trouble of making a macro to export a DXF to a preset location with a hotkey.

Request: add export options, set either in the dialogue window or in a preferences section. The first option allows a given file type to be exported without the constraints visible. Designers exporting a DXF or gcode to a machine will never want the constraints to be included. The 'section only' option allows only the selected lines/curves to be exported. The option to export only the selected lines is important because I usually design all the parts required for a design together in a single file, but only want to cut one at a time - either for a quick test or because I need additional copies. 

Although the two export options above are the most important to people doing rapid prototyping, I would love for the preference section to allow a file name, a location, and a file type to be set. A hot key combo (such as Ctrl+Shift+E) then instantly saves the file with the settings for quick export. The user can then quickly load the same file from the same location into the machine software. As someone who might repeat the process of exporting and then importing the file into the machine software several hundred times in a day, this feature can save a lot of mental energy clicking through a dialogue window.

robnee, 2021-04-15:
Are you using 2.x or one of the more recent 3.0 testing builds?  3.0 gives more control over which styles are exportable and may help with the first problem.  Constraints can be marked visible but not exportable.

dustinhartlyn, 2021-04-16:
I've been using a 3.0
After your message I looked around more intentionally though and don't see the export controls. Under Edit->Configuration I do see some options for export, but nothing to exclude the constraints from export. I full accept the possibility that I am blind though. Where should I be looking?

robnee, 2021-04-16:
This is controlled by the line styles.  In 3.0 there is a checkbox to specify if each style is exportable or not.   For the default styles this a global setting.  If you need to tweak this per drawing you can assign the constraints a custom style.

dustinhartlyn, 2021-04-17:
I looked into making a custom style for the excluding the constraints from export. I need further instruction on if I need to manually assign each constraint to a custom style, or if I can assign all constraints made to the style as a whole. I tinkered with the process for about fifteen minuets and could not figure out how to do anything beyond creating the custom style and specify that it be excluded from export. 

My impression as a general user who does not intimately know every corner of this wonderful program is that it is widely  complicated to accomplish the task. I have never encountered a situations in it was desirable to export the constraints in a DXF and the Solvespace even raises a warning when exporting the constraints in Gcode. 

I very much think the program should include a 'preferences' section for the program so users can use a simple and familiar interface to customize the program to their own needs. In this properties section, I think the default should be to not include constraints in the export of DXF, SVG or Gcode, while giving a check box to include them if there is ever a need for it. At the very least the default line styles should include an option to exclude them from export. 

I am very impressed by this program and all the thought invested into it.

robnee, 2021-04-17:
Did you try turning off exportable for the default constraint style vs creating and assigning a custom style for constraints?  This should certainly work for pdf and svg.  The settings for the default styles are global so you should only need to change this once.  Each format handles exports differently.

dustinhartlyn, 2021-04-17:
For the default constraint lifestyle I do not see an option to exclude it from export. I only see the 'export these objects' checkbox in custom lifestyles. Should I be looking somewhere else? 

I do think there are some situations in which I and others would want to export the constraints to a PDF, which is why I think the exclusion should be file specific. I might send an assembly drawing to a person in the form of a pdf with dimensions, but I will never send a dxf to a machine for cutting which includes the constraints. The same is true of Gcode. I would actually be interested if there was ever a situation in which a user desired to export the constraints in one of these two files.

Symbian9, 2021-04-18:
> Should I be looking somewhere else? I might send an assembly drawing to a person in the form of a pdf with dimensions, but I will never send a dxf to a machine for cutting which includes the constraints.

It is ***very ease to hide something just in one click before export***: if you need to exclude `normals` / `constraints` / `construction entities` / `constraints and dimensions` from 2D export (i.e. for DXF export) *just hide them via related buttons on "Property Browser" toolbar* (and click one more time to show it again for PDF / SVG export):

[image]

dustinhartlyn, 2021-04-18:
Thank you for the picture. I am aware of that feature, and the goal of my initial suggestion was to give feedback from a person using it in the 2d cutting world (lasers, waterjet cutting, plasma, etc). While effective it becomes a tiresome task to constantly click the constraints on and off when exporting so frequently. I am unaware of a situation in which the a user would desire to include the constraints when exporting DXF or gCode, so it seems like the appeal of the program could be increased by making it more intuitive. 

The program even warns when exporting gcode with constraints, and I think it far more likely a person will accidental forget to turn off constraints when exporting... and in turn ruin the material they cut.  Design work takes a lot of focus, and remembering to do an extra task that could be automated seems unnecessary.

I reiterate that I appreciate the work done by you. I am a seasoned engineer in the laser cutting industry and can saw with authority, that if the two suggestions in the original post are implemented, it will be one of the most user friendly options for people doing 2d cad work.

Symbian9, 2021-04-18:
>  it seems like the appeal of the program could be increased by making it more intuitive.

I may just recommend to read *SolveSpace docs* firstly before use *SolveSpace* app.

And if discussed here is missed in *SolveSpace docs*, then we can add such info into docs.

> I am a seasoned engineer in the laser cutting industry and can saw with authority, that if the two suggestions in the original post are implemented, it will be one of the most user friendly options for people doing 2D CAD work.

For batch export to various formats `solvespace-cli` should be a right way to do that, but need take a look on CLI options.

Also here is already a *feature request on "styles presets"*  for such cases where user need different export presets to be used:

- https://github.com/solvespace/solvespace/issues/902

dustinhartlyn, 2021-04-18:
Hi Symbian9 - Thank you for telling me about the solvespace-cli interface. I read through all the documentation on the SolveSpace site and completed the tutorials before posting anything. My goal is not to waste your time. I just looked and confirmed there is no reference to the CLI in the reference document (https://solvespace.com/ref.pl) but I will look elsewhere. 

I only posted this issue because [Paul encouraged me to](https://solvespace.com/forum.pl?action=viewthread&parent=3635). I see that several people are working very hard on this program, and it has accomplished some impressive features. However, I find that engineers (myself included) easily fall into the trap of thinking it is simple to use something they themselves designed, built and use daily. However, to new users the same thing may be preventively complex and confusing. I am not one of the developers, but I have put in the time to familiarize myself with the program. I think it is technically impressive but that the UI may not be intuitive enough for it to gain broad adoption by general users. 

As an outsider with no prior search history to bias Google, I only found this program as a note at the bottom of a post after an extended time of looking for different open source cad programs. That tells me that the learning curve for SolveSpace is too steep for the broader audience, and I think that is shame because it is much more capable than the alternatives. 

Symbian9, I would appreciate it if you point me to where I can learn more about the CLI interface.

I think we got into the weeds about excluding constraints from exports. You pointed out several ways it is possible, but my feedback is that it is not user friendly to accomplish a routine task for anyone doing 2d cad work. I am willing to help develop the UI, but was originally looking for direction from the core SolveSpace team first. Beyond the issue of exporting constraints, I still think that exporting only the selected lines is a very important feature for users.

ruevs, 2021-04-21:
For reference I am posting the forum thread where this discussion started:
https://solvespace.com/forum.pl?action=viewthread&parent=3635&tt=1618405421

phkahler, 2023-01-30:
Looking back at this now with an eye toward adding these features without a bunch of UI complexity. For example:

1) Make export only export the current selection if anything is selected, and the current behavior if nothing is selected. This eliminates the need for another UI or menu item for "export selection" but it is also not discoverable. Would need to be added to the user manual for sure.

2) g-code should NEVER include constraints. IMHO this could be hard coded with no UI change.

3) a checkbox for "export constraints" which would default to checked, and those exporting to machine tools could uncheck it. Not sure this needs to be file-type specific since exporting is already different from saving. I suspect most people are fairly consistent in what/how they export.

4) As suggested by OP, add a menu/hotkey for "ReExport" or Quick Export which would overwrite the previously exported file similar to how "Ctrl-S" saves the current sketch in the same location once it has been specified. Maybe the Export file name and type could be saved as a configuration option so it's reused even after restarting Solvespace?