solvespace/solvespace #1016: Specific design has error in STL export

someSven, 2021-04-18:
### System information

**SolveSpace:** version:3.0~3e3ccdca
**Operating system:** Raspian wth OpenGL active

### Expected behavior

### Actual behavior

### Additional information
[image]
[forearm_V1a.slvs.zip](https://github.com/solvespace/solvespace/files/6331502/forearm_V1a.slvs.zip)

phkahler, 2021-04-23:
@someSven  I can't seem to open that file to chase the issue. Until then you might try changing the chord tolerance settings in the configuration screen as a work-around. There are 2 settings, one for display and one for export. The display one is in percent, but also shows the equivalent in mm. The export one is in mm. You might be able to "fix" it with the display setting and then set the export to match, which should export the same mesh you see in SolveSpace.

ruevs, 2021-04-28:
@someSven the link your model is broken. Please attach it again.

someSven, 2021-05-02:
Hi, sorry for the late answer. I got distracted, had other things on my mind and completely forgot about this here.
[forearm_V1a.slvs.zip](https://github.com/solvespace/solvespace/files/6410125/forearm_V1a.slvs.zip)
The solution above by phkahler worked. Thanks. For me at least, because I don't need the model for anything than printing. It has many sharp edges, and I wasn't able to reduce them by increasing the number of exported linear segments. But this is not a problem, since it doesn't show up in the sliced model. Also I didn't look into the tutorial about that, which I plan to do now. 
[image]

phkahler, 2021-05-18:
>> It has many sharp edges, and I wasn't able to reduce them by increasing the number of exported linear segments.

@someSven  The first thing to do for that is lower the chord tolerance. That parameter determines how far a flat surface is allowed to deviate from the ideal curved surface. Small values lead to more triangles with less deviation. The maximum linear segments setting is a sort of upper limit on what can be achieved with the chord tolerance setting and will not have any effect unless you have already lowered the CT setting enough to bump into the limit.

I you're wondering why both exist... If you have a large model with tiny curved features, it may be necessary to lower the CT setting to get nice curves on the small features. But that might lead to a huge excess of segments on larger scale curves. At least that's what I think the reason is. The 20 segment default is enough to give a decent (IMHO) visual representation of any curve on-screen. The default CT setting also balances performance with appearance and is the first thing that should be adjusted. I hope people exporting STEP never have to change from those defaults. STL is another thing ;-)

phkahler, 2021-05-20:
This is due to the way ear and grid-triangulation work. The surface here is a spline (order 3) and when lathed that creates surfaces of order 3x2 which uses grid triangulation (make a grid of quads, each split in 2 triangles). The problem isn't with the grid itself, but that we don't use the grid over the entire surface. All quads on the perimeter are handled by the regular ear triangulation code, which sometimes does things like this.

ruevs, 2021-12-20:
@someSven can you please attach your model again. I'm not sure what happens, but both of the ones above a missing.