solvespace/solvespace #1017: Rotate about the specified axis at the specified angle

zhulaojianke, 2021-04-18:
Using the middle mouse button to rotate the view in the workspace is really convenient, but it gets tricky when I need to rotate the view to a specific perspective. Every time I need to adjust for a long time to get the right perspective, just like #1006 and the issues it refers to.

Let me make a suggestion: 
(1) Let user select a axis or a line, and press a hotkey (such as "K", this letter is not defined as a hotkey in the current version) and then a dialog box appear in the property browser;
(2) And the user input an angle data (>0, clockwise rotation, and <0, counterclockwise rotation) and press enter, the view will rotate about the axis or line at the specified angle;
(3) And if the user input 0 (the default value of the dialog box), then he can use middle mouse button to rotate about the axis or line freely.

This reduces the degree of freedom of rotation of the middle mouse button, and making it easier to control the rotation of the perspective.

I think such a function allows users to locate their desired perspective very quickly, thus saving a lot of time for the middle mouse button counterpoint operation.

zhulaojianke, 2021-04-18:
In addition to selecting a axis and line, the user can also select a point in workplane. When he does this, the view will be rotated about the axis perpendicular to the point, rotating the corresponding Angle either clockwise or counterclockwise.

phkahler, 2021-04-19:
Are you aware of the view options at the bottom of the toolbar? The "neatest isometric" is very useful, just drag to the approximate view you want and click that (or press F3). For the nearest  workplane (including the 3 primary predefined ones) press F2. This covers most of the orientations you'd get from a view cube.

ruevs, 2021-04-19:
`Ctrl + middle button (mouse wheel click)  drag`is also related to what you are asking for (even though not exactly. Try it out when a solid (3d) group is active.

zhulaojianke, 2021-04-19:
I have tried all of these features, and when I needed to rotate to a particular perspective, it took me a couple of times to get it right. It's not that Solvespace can't do it, but it's really hard to do it, it's really hard to control, and when you have rotated the solid shape at a certain Angle, then drag mouse with middle button, it will not rotate in the direction the mouse movese, it will rotates on a other certain axis. 

I wanted to make a video to illustrate this matter, but I felt that even making a video would not make it clear.

For a simple example:
I draw a extrusion cubic with it center on origin point, and I want to look at the right face of the cubic, how to do it? 
 Under current conditions, I will drag drag it to right, to show a solid shape, and press F3, and continue to drag to right, and press W or F2, and now the cubic is standing, not lying, I must press ctrl and drag again, adjust it so it's close to lying down, and press W.
Use the method I proposed, select the Z-axis, press K, and enter 90, enter, OK.

I only suggested to improve the efficency and accuracy of the perspective controlling. I don't know if I made myself clear.

zhulaojianke, 2021-04-19:
In the example above, the harder thing to do is to look at the back face of the cubic.

zhulaojianke, 2021-04-19:
Well, I admit I still need more practice. Once I got used to the spherical rotation from the center of the workspace and the use of the view-switching shortcuts, I wouldn't ask that question.

BTW, I found the "workplane" command in "Sketch" menu very handy and useful, when I just only want to create a reference surface not ortho without having to draw a graph on it. Why not give this command a shortcut key? Like using "K"?  :)

zhulaojianke, 2021-04-20:
I published a separate issue for the "workplane" shortcut key, #1019 .

closed 2023-01-16 by phkahler