solvespace/solvespace #1018: Change default file filter for linking

phkahler, 2021-04-19:
### System information

- **SolveSpace version:** ALL
- **Operating system:** ALL

### Expected behavior

When linking a file we should set the default file filter to include types other than .slvs if they are linkable.

### Actual behavior

The filter is set to only .slvs but that is not the only option.

### Additional information

SolveSpace can currently link .emn (IDF files) and should soon be able to link .stl files. The default filter for linking should include these other options to make it quicker for users, and to make that ability more discoverable.

kk6mrp, 2021-04-19:
I haven't checked on Linux but windows shows both options:
[image]

phkahler, 2021-04-19:
It does support both. What I mean is to have a single option (default) that shows both slvs and emn at the same time. And when STL is added we should show all 3 at the same time. The point is to show all files that could be linked so the user doesn't have to change the filter except to narrow the scope. We would keep the more specific options too.

closed 2021-07-29 by phkahler

Symbian9, 2021-07-29:
@phkahler, I highly recommends rename *"ALL"* to *"Supported formats"*, because *"ALL"* is commonly used in desktop software to list *just all available files in a folder path* in file chooser dialogs (aka *"All files"*).