solvespace/solvespace #1019: Hope to Specify a hot key for the "workplane" command

zhulaojianke, 2021-04-20:
During I learned and practiced Solvespace, I've found the "Workplane" command in "Sketch" menu is very useful in drawing a complex shape parts, it greatly simplifies the mouse operations required to define a new workplane. But it's still a little inconvenient to choose it from a drop-down menu every time, if specify this "Workplane" command a hot key, such as "K" key, this excellent " Workplane" feature will make the user feel more comfortable.

zhulaojianke, 2021-04-20:
Or perhaps it would be better to offer an option in the property browser to allow users to customize all of the Quickkeys for this software. This would prevent Solvespace from clashing with other software shortcuts and would allow users to maintain consistency between the shortcuts for the same function across different programs on their computer.

phkahler, 2021-04-21:
Shift->W

zhulaojianke, 2021-04-21:
"shift + W" and "Workplane" are two different commands. The "Workplane" command is in the menu "Sketch".

[image]

robnee, 2021-04-28:
@phkahler  @ruevs 'k' is the only unmodified letter remaining.  We could use that for sketch->workplane but I slightly prefer Shift+P instead.  There are also the new-ish commands sketch->Image and Darken Inactive Solids that are missing accelerators.  I propose this:

'k' - Sketch->Image
'<' - Darken Inactive Solids
'Shift-p' - Sketch->Workplane

I'm very open to other suggestions though...

phkahler, 2021-04-28:
@robnee  Not everything needs a key. Certainly not Sketch->Image or Darken Solids.  I'm not sure how often Workplane is used, so not sure how important that one is?

BTW I'm happy to drop the key for Helix if we ever need another one. It's not a super common thing to use.

BTW2 notice that many of the keys are "grouped" by the shift key. Most sketch tools are plain keys, more new groups use Shift, and everything in Analyze is Ctrl->Shift->Key.

BTW3 I have my eyes on Shift->S (for segmented extrusions) and Shift-F for Frustum extrusions ;-)

robnee, 2021-04-28:
I rarely use these three so have no preference but noticed that almost everything has a key so thought to propose this.

I noticed too that most of the new group commands are shift so, agreed, probably best to not consume that namespace.

Perhaps 'K' is best held for any new constraint that comes along although constraints also use punctuation and there are plenty of those left.

ruevs, 2021-04-28:
@phkahler @robnee In my opinion `Sketch | Workplane` in not used often enough to warrant a dedicated accelerator key, the same applies to image.

And once we assign an accelerator we are "stuck" with it - someone will inevitably complain if we remove or repurpose it later.

And by the way it would be pretty convenient to do `Alt+S` -> `W` (the normal menu accelerators) to create a new work plane in the current group. But there are conflicts between the accelerators in the Sketch menu:

- W
  "In &Workplane"
  "&Workplane"
- A
  "&Arc of a Circle" 
  "Tangent &Arc at Point"
- I
  "&Image"
  "Split Curves at &Intersection"

All accelerators used in that menu: ["W", "3", "P", "W", "S", "O", "R", "C", "A", "B", "T", "I", "G", "A", "I"](https://github.com/solvespace/solvespace/blob/709dc31f7865573cbf23c09dbd1946b017e7dc80/src/graphicswin.cpp#L122)

If we remove the collisions it would be more convenient - even though (at least on Windows) pressing the accelerator key multiple times jumps between the different items and `Enter` "runs" the action.

So overall - I think we should **_not_** dedicate a global single key accelerator to "Workplane", but we should consider rearranging the Sketch menu accelerators.

robnee, 2021-04-28:
@ruevs yes, I like your suggestion so perhaps something like:

"wor&kplane", "Ta&ngent Arc at Point", and "Split Cur&ves at Intersection"

zhulaojianke, 2021-04-28:
If solvespace allows users to customize hotkey in the properties browser, I think all the problems are no problem.

ruevs, 2021-04-28:
> If solvespace allows users to customize hotkey in the properties browser, I think all the problems are no problem.

That is true, but it is out of scope at this point. It is a rather large refactoring and a fair amount of work and bloat for a fairly small benefit. And it goes somewhat against the small fast and simple ideology of Solve space - it is after all a single 6.7MB exe.

> @ruevs yes, I like your suggestion so perhaps something like:
> 
> "wor&kplane", "Ta&ngent Arc at Point", and "Split Cur&ves at Intersection"

Looks good to me. @phkahler shall I do it? @jwesthues ?

zhulaojianke, 2021-04-28:
@ruevs Indeed, as you said, one of the reasons I like using Solvespace is that it's small, quick to start up, and fast to load models.

But what about making Solvespace no longer a single-file program? For example, if there is another .cfg or .ini file, read from that document when solvespace start up. And I think many of today's new feature requirements can be easily addressed if users are allowed to create and invoke macros, or plug-ins. Let the users themselves consider Whether they need or not to sacrifice some startup speed and software volume for a better use of the experience.

For example, add a function to Solvespace to read and load external files on startup (the external files, such as .ini file, must be stored in the same folder in the solvespace), skip external files when they don't exist, and read and load external files when they exist. I think this is maybe the least possible modification to the current version of Solvespace.

phkahler, 2021-05-18:
@zhulaojianke  What are you doing with Workplane?  When sketching I always use existing lines or add construction lines to define the orientation of a new workplane created via Shift->W.  It looks like the method you're using is much harder to orient a new workplane, so either I'm using it wrong or it doesn't seem to be a usedful feature. Is this why you opened issue #1017 ?? To get better control of the camera angle since workplanes created the other way seem to align to the current view?

zhulaojianke, 2021-05-19:
@phkahler Hello, I am not a professional mechanical engineer, sorry, and my job in my team is to be the project leader, sometime like a teacher. I need to explain my plan and ideas to our mechanical engineer. And I have test several softwares such as Solidedge cracked, Design Spark mechanical, sketchup, GeoGebra and many others. But many softwares are too big to work on my poor notebook computer, or too complex to test my method quickly. I have learned solvespace on YouTube, and I found this software is what I need after  I finished the application tutorial. 

My interaction with mechanical engineers is that I first draw a sketch, and test my idea on the sketch, and then explain my idea to them, I must rotate my sketch over and over to show the whole things to my workmate. Occasionally I need draw some geometry on my sketch to show them some design possibilities or how to use these design possibilities, these geometry are often more arbitrary and free, not necessarily with the exact constraints of the original sketch in order to avoid design misunderstandings. So I found that almost all of the features in solvespace were useful to me, and almost without waste! Such as easy comments, the "workplane", change design by contrains and so on.

The previous time I wanted to explain a CAM mechanism to my colleagues, but it seems solvespace could not support CAM design, that I have writen some issues in this forum. And then, I drew on paper and show it to my workmate, that's a hard way to explain it.

I have use Microsoft PowerPoint to draw mechanical structure diagram before, it use easily. But ppt can not draw and operate 3D shape, and I can not test my design methods on it.

ruevs, 2023-07-14:
https://github.com/solvespace/solvespace/pull/1381

commit f0912d42 (2023-07-14): UI: Remove shortcut key collisions in the Sketch menu

"&Workplane" -> "Wor&kplane"
"&Image" -> "I&mage"
"Tangent &Arc at Point" -> "Ta&ngent Arc at Point"

Closes #1019
PR #1381 (2023-07-14): UI: Remove shortcut key collisions in the Sketch menu
"&Workplane" -> "Wor&kplane"
"&Image" -> "I&mage"
"Tangent &Arc at Point" -> "Ta&ngent Arc at Point"

Closes #1019
closed 2023-07-14 by ruevs