solvespace/solvespace #1020: Create an automated performance benchmark

phkahler, 2021-04-21:
We should have a performance benchmark that can be run in an automated way, probably using solvespace-cli. Maybe this is just loading an existing sketch that stresses a lot of code paths and produces an image?  Frame rates are not relevant because sketch regeneration when geometry changes takes way longer than rendering.

If there is an existing benchmark it's probably out of date with all the recent improvements. We should have something that takes on the order of seconds (10's of seconds?) where variations of a few percent to 5x can be observed.

This is based on an inquiry from Michael over at Phoronix, so keep in mind the kind of testing he's known for. If we're lucky he'll start doing benchmarking for us, but it should be easy/trivial for him ;-)

ruevs, 2021-04-21:
This would be extremely useful. However some performance tests depend on changing something. E.g. changing constraints in the fist group of this https://github.com/solvespace/solvespace/issues/841#issuecomment-742153838

solvespace-cli can not do that.

vespakoen, 2021-04-29:
I have a vague memory of seeing a benchmark function in someones fork at some point, I think it added something to the "test" functions that allowed one to see how long solving took.

Was it in your fork maybe @Evil-Spirit ?

Profiling on macOS is also pretty good, showing how long some lines of code took to run, I should probably open a whole bunch of big files and see where the bottlenecks are & share back.

phkahler, 2021-04-29:
@ruevs agreed. It's probably a good idea to stress the solver as well as everything else. Maybe we need a benchmark build (since we dont have scripting) that programmatically creates something and stresses everything?

Do groups get solved when opening a file? That should be easy though because it should be saved in a solved state.