solvespace/solvespace #1023: Feature Request: "Angle Ratio" and "Angle difference"

zhulaojianke, 2021-04-22:
I have find a better method to solve this problem. :)
[Angle ratio.zip](https://github.com/solvespace/solvespace/files/6352976/Angle.ratio.zip)

And I hope you could add "Angle Ratio" and "Angle difference" commands in the "constrain" menu, just like "length ratio" and "length diffrerence" commands. These two commands are very useful for driving the "motor" turn many rounds in step dimension mode.

_Originally posted by @zhulaojianke in https://github.com/solvespace/solvespace/issues/996#issuecomment-824254025_

Evil-Spirit, 2021-04-22:
There is no point to support things like this. There is branch with supporting for arbitrary equations / parameters, so you can easily input any expression you want

Evil-Spirit, 2021-04-22:
Also, you can check how I use [thing like this in NoteCAD](http://notecad.xyz/?filename=romantic_steam_new.xml) for creating slider to rotate engine.

ruevs, 2021-04-22:
Indeed! I totally agree with @Evil-Spirit, that parametric sketches is the next logical step Instead of adding more specific constraint types (even though I implemented length difference back in 2015 in b23336b58).

See the issue for the whole discussion of parametric sketches https://github.com/solvespace/solvespace/issues/77

closed 2021-04-22 by ruevs

zhulaojianke, 2021-04-22:
Indeed, if Solvespace had variable parameters and computational capabilities (and preferably some system constant, such as π), then it would probably be the best lightweight windows CAD software in the world. 

But it's not clear to me why the Solvespace development team didn't include such great features. I also saw the discussion on adding this feature in this issue forum a few years ago. It seems that this feature will be added in version 4.0, right? Although I don't know when version 4.0 will be released, after all the version 3.0 has just been released. Therefore, I settle for the next best thing, lower the level of feature requirements, and put forward the "Angle Ratio" and "Angle diffrence" feature requirements.

I've seen a fascinating presentation and programming live stream that you and another friend did earlier on YouTube, and I also have visited notecad.xyz. Perhaps due to the online usage, the use of Notecad web version in China is not very smooth. And maybe because of the lack of help files, I also have some matters on using Notecad.

I also want to come up with several other feature requirements, some of which seem to have been mentioned in this issue forum, such as "Group Mirror", I've seen a demo of this on YouTube, and it seems a very good ideal.

There are also multi-points symmetry constrain, multi-lines or angles equal constrain, as well as extruding along paths, line segment welding, edge fillet, the broader point on surface constraints that I had suggested in the post earlier (currently Solvespace seems to be limited to point-in-plane or cylinder surface constraints), and so on.

And I would also like to see the solvespace development team add a "group edit" feature that manually sorts and copies and pastes groups events that have occurred in the properties browser, instead of renaming and deleting groups.

As a freeware user, am I asking for too much? In the end, everyone on the development team is working for free, no pay, and we're all very busy on our own jobs, so despite all the features I'd like to ask for, I'm not too upset with Solvespace as it is now. Solvespace is the one of the loveliest pieces of CAD software I've found in my decades of work.