solvespace/solvespace #1024: New feature idea? Missing functionality? Look here first?

ruevs, 2021-04-22:
# Commonly Requested and Discussed Features and Enhancements to SolveSpce

This list is not exhaustive. Please search the issues before opening a new one.

## [Features already "planned" for a future "4.0" version](https://github.com/solvespace/solvespace/milestone/4) 
 
## Constraints
- Parametric sketches, parametric constraints, variables. inequality constraints etc. </br>Examples: `x=2` `y=3` `z=x+5` `3*x+z` `x`  `<7`</br>https://github.com/solvespace/solvespace/issues/77 https://github.com/solvespace/solvespace/issues/81   https://github.com/solvespace/solvespace/issues/422 https://github.com/solvespace/solvespace/issues/80 https://github.com/solvespace/solvespace/issues/166 (duplicates https://github.com/solvespace/solvespace/issues/273 https://github.com/solvespace/solvespace/issues/542 https://github.com/solvespace/solvespace/issues/931)
- "Angle Ratio" and "Angle Difference" - covered by parametric constraints https://github.com/solvespace/solvespace/issues/1023
- Constrain multiple objects simultaneously, batch constraints https://github.com/solvespace/solvespace/issues/573 https://github.com/solvespace/solvespace/issues/921 (pull requests https://github.com/solvespace/solvespace/pull/530 https://github.com/solvespace/solvespace/pull/836)
- Tangent constraints on lines and curves with no common points https://github.com/solvespace/solvespace/issues/937

## Construction tools, solids etc.
- Extrude at an angle or skewed? It is already possible. See here https://github.com/solvespace/solvespace/issues/453#issuecomment-642508274
- Loft, extrude along arbitrary path https://github.com/solvespace/solvespace/issues/439
- Chamfer, fillet https://github.com/solvespace/solvespace/issues/149
- Mirror tool https://github.com/solvespace/solvespace/issues/72
- Hole tool https://github.com/solvespace/solvespace/issues/955
- Spiral (probably not happening) https://github.com/solvespace/solvespace/issues/426

## User Interface
- Navigation Cube https://github.com/solvespace/solvespace/issues/70 https://github.com/solvespace/solvespace/issues/1006
- Suppressing some solid models? Already possible, but see here https://github.com/solvespace/solvespace/issues/1007

## Export
- Export STEP files as solids instead of as surfaces https://github.com/solvespace/solvespace/issues/206
- Export "blueprints" https://github.com/solvespace/solvespace/issues/143
- DXF export problems? Comment here https://github.com/solvespace/solvespace/issues/361
- Bill of materials https://github.com/solvespace/solvespace/issues/24

## Import, Link
- SVG import https://github.com/solvespace/solvespace/issues/245
- STL or OBJ import https://github.com/solvespace/solvespace/issues/724
- STEP files https://github.com/solvespace/solvespace/issues/88
- DXF DWG import problems? https://github.com/solvespace/solvespace/issues/326

## Problems with solid geometry? Red lines and missing surfaces in your model?
- Look here first https://github.com/solvespace/solvespace/issues/738. New **_simple_** examples are always welcome as new issues.
- Edges have "see through" dotted lines? Yes they do. https://github.com/solvespace/solvespace/issues/379 https://github.com/solvespace/solvespace/issues/378

## Parts model library  https://github.com/solvespace/solvespace/issues/330 https://github.com/solvespace/solvespace/issues/793

## Donating crowdfunding
https://github.com/solvespace/solvespace/issues/265 https://github.com/solvespace/solvespace/issues/716 https://github.com/solvespace/solvespace/issues/788 https://solvespace.com/forum.pl?action=viewthread&parent=1590&tt=1618728484

ruevs, 2021-09-08:
https://ltnr.ca/Lib/solvespaceTips.html a nice list of useful hints (slightly outdated in places, because it was written for 2.3)

barkYboY, 2022-01-22:
A configurations feature would be very useful.  With this feature a part(imported geometry) in a model could shift things around without having to open, modify, save, then reload the imported geometry.  Since the part is not changing, just the used configuration, the part doesn't need to be modified.  This feature not only would increase workflow productivity but could also be used to solve some other implementation issues(mentioned in other threads I'm not quite up to date on yet).
  Configurations could help enable parts libraries by being able to adjust a standard model without having to change it.  Say screws for example could have different configs for at least lengths, but potentially many other things.  A hole wizard implementation could use different screw configs for tap diam, clearance, counter sink etc, to encapsulate the geometry differences in solids...  I've used configurations in Solidworks to solve tolerance stack problems before and it worked well.  Configurations could be used to hold versions of parts too.  So they could evolve and be rolled back to simpler states with less confusion.
  If in addition to configurations a locking (key based encryption) feature was added, Solvespace could potentially get much more adoption from larger organizations, if not become the standard.  Giving the ability to protect and segregate part IP(how the part was designed) from how to implement it, is one of the main hurdles of hurdles for a mainstream 3D CAD standard.  Anybody that has tried to design around a broken step file knows what I'm talking about.
  Here's an analogy.  Suppose a cad standard is the road and the parts are the vehicles that travel on roads.  The majority of those making vehicles will make them comply with standard roads.  You'll still have niche special vehicles for special roads (like racing and high end cad systems).  Yet the majority of vehicle makers will adopt the road parameters that enable the largest sales market.  If some of the details of the road are hidden from the masses so one vehicle maker has an unfair advantage, their road will probably remain proprietary and likely cannot become the agreed upon standard.  In my opinion, Solvespace has the opportunity to become the standard road for 3D modeled cad.  A road that can lead to an 80% or more increase in efficiency of parts usage for people(on demand manufacturing).  Thank you to all who contribute.

ruevs, 2022-01-22:
@barkYboY please open a new issue with your suggestion. Then delete your comment here. This is intended as a summary and a starting point only.

barkYboY, 2022-01-23:
ruevs,
  Completed as requested, issue #1196 presents a case for a configuration feature.
Cheers

ruevs, 2022-01-23:
@barkYboY thanks! I will delete both our comments now :-)

Kreijstal, 2025-06-02:
please make a release, a tag or something its been 3 years

ruevs, 2025-06-03:
> please make a release, a tag or something its been 3 years

One more really annoying issue I need to debug, the Chinese translation update and a regression in STEP file export that we just introduced last week (I'm currently fixing it).

After those we should be able to release 3.2.

LJH-DESIGN, 2025-06-04:
Please explain your email to me.

Thanks,

Larry J. Hill

912.547.0005
***@***.***

LJH-DESIGN, 2025-07-16:
I'm still here.

"Sandbagging" as it used to be called.

Larry J. Hill

912.547.0005
***@***.***