solvespace/solvespace #1027: Default style for hidden edges is inconsistent with the "stipple occluded lines" icon

jkrei0, 2021-04-23:
### System information

- **SolveSpace version:** 3.0~0e0b0252
- **Operating system:** Windows 10

### Expected behavior

Toggling the hidden lines icon to "Stipple occluded lines" should stipple hidden lines by default, as the icon shows.

### Actual behavior

The default style for hidden lines is a solid line.

[image]

phkahler, 2021-04-24:
As I recall, the text when you hover over the tool indicates what it will do when clicked, not the current state of things. If the appearance and hover text are consistent then all is "correct". I found it strange too and still dont know if it should change. It's also a 3-state button which makes it more odd.

Symbian9, 2021-04-25:
@phkahler, Gess actual ***issue here*** is that *"stipple type"* in default `s00e-#def-hidden-edge` style set to "not dashed" (see screenshot) — seems like it affected by update from previous SolveSpace versions (pre-release) builds to `v3.0` (probably this introduce few commits before `v3.0`).

Also *on clean* install `v3.0~` it seems like `s00e-#def-hidden-edge` style by default set to "not dashed":

[image]

> As I recall, the text when you hover over the tool indicates what it will do when clicked, not the current state of things. If the appearance and hover text are consistent then all is "correct".

No, icon show actual state of the hidden edge representation in main window, so *its correct here*.

> I found it strange too and still dont know if it should change. It's also a 3-state button which makes it more odd.

For curios reason order of  this "3-state button" is not always step-by-step — if click it a little bit faster it jumps in unordered manner (but this is another issue, not relate to actual).

robnee, 2021-04-25:
Will do...

robnee, 2021-04-25:
> For curios reason order of this "3-state button" is not always step-by-step — if click it a little bit faster it jumps in unordered manner (but this is another issue, not relate to actual).

@Symbian9 @phkahler This is a problem with all of the toolbar buttons.  Not just the tri-state but the toggles as well.  I can't reproduce it on Windows but I can on Linux.  I'm using a VM for Linux with a remote display (vs native for Windows) so it may be related to that and not a SS issue.  I will try to look into this too.

robnee, 2021-04-25:
#974 introduced this problem when looking up the default styles for the very first time after an install.  i.e. before the config settings.json or registry entries have been created.  When exiting SolveSpace for the first time the config would be created and the incorrect stipple settings written so this problem persists.  Manually resetting the factory defaults and restarting SolveSpace repairs the problem.  I will do a PR to fix this for the first run.

Symbian9, 2021-04-25:
>  I will do a PR to fix this for the first run.

@robnee Also, please, check does SolveSpace save actual styles into config on each session
Actually it seems like SolveSpace does NOT saves configuration/styles changes on closing SolveSpace/saving file — as a result all styles & configuration from actual session lost for the each next session (but it should be saved on closing app on each session).

robnee, 2021-04-25:
> Actually it seems like SolveSpace does NOT saves configuration/styles changes on closing SolveSpace/saving file — as a result all styles & configuration from actual session lost for the each next session (but it should be saved on closing app on each session).

SolveSpace does save the default styles when exiting the GUI by calling `Style::FreezeDefaultStyles` but, surprisingly, it does not save them on File -> New so the subsequent sketch reverts to the styles from the start of the session.  This is due to the odd way SolveSpace works, it _reads_ settings directly from the config but does not update to the config until exit.

1. Open SolveSpace
2. Draw a line segment
3. Change the line style for active-grp to dashed.  current drawing reflects change.
4. File -> New (save or don't save, doesn't matter)
5. Draw a line segment (this will have reverted to continuous not dashed)

@phkahler @ruevs I suppose this can be fixed but is it intentional?  It does not seem to be what the user would expect nor is it documented.  Since default styles are not saved in drawing they should probably be saved to the config immediately or via a "save to config" option.

Symbian9, 2021-04-25:
> SolveSpace does save the default styles when exiting the GUI by calling Style::FreezeDefaultStyles but, surprisingly, it does not save them on File -> New so the subsequent sketch reverts to the styles from the start of the session.

I just tried to:

1. Start SolveSpace;
2. Change styles;
3. Exit from SolveSpace (both ways: via menu "File > Exit" and via "Close" button on window title);
4. Restart SolveSpace;
5. As result — ***styles NOT saved***.

>  or via a "save to config" option.

Adding `save to config` option into "Property Browser > line stiles"* tab would be the best solution as for now ­— users should not redefine styles each session if they want use own style presets in next sessions too.

Also, its related to adding *"Import / Export line styles"* feature request:

- https://github.com/solvespace/solvespace/issues/902#issuecomment-771048837

robnee, 2021-04-25:
@Symbian9 
> 1. Start SolveSpace;
> 2. Change styles;
> 3. Exit from SolveSpace (both ways: via menu "File > Exit" OR via "Close" button on window title);
> 4. Restart SolveSpace;
> 5. As a result — _**styles NOT saved for next session**_.

It's not clear what you did in "2. Check styles;" or if you modified a default style.  

This works for me:

1. Start Solvespace
2. Open line styles page
3. Change `s008-#def-hovered` to a different color like blue
4. Exit SolveSpace
5. Restart SolveSpace
6. Confirm hovered entities are now still blue

Can you confirm this works for you?

Symbian9, 2021-04-25:
> It's not clear what you did in "2. Change styles;" or if you modified a default style.

Here is what I did:

1. Start SolveSpace (clean install);
2. Open `line styles` page;
3. Change `s008-#def-hovered` stipple type from "not dashed" to "dashed" (default by design in previous versions);
  [image]
4. Exit SolveSpace;
5. Restart SolveSpace;
6. Styles not saved and wrongly "not dashed"  `s008-#def-hovered` still persists in next session.

> - Change `s008-#def-hovered` to a different color like blue
>
> Can you confirm this works for you?

@robnee, Nope, this also ***does NOT works*** for me.

FTR, I use SolveSpace `v3.0~`https://github.com/solvespace/solvespace/commit/9e8c3db856b8fa9eeb807b42239cf9d404739acf under Linux / Debian 10 buster.

robnee, 2021-04-25:
If settings.json is not created then that is the problem.  Is there any reason SolveSpace might not be able to create files in your home directory?  Are either of these env variables set?  If so to what?

$HOME
$XDG_CONFIG_HOME

Is there anything strange about your setup?  Can you try on a different install of Linux?  Are you starting SolveSpace from the command line?  If not please do so and repeat the above tests and report any messages displayed.

Symbian9, 2021-04-25:
> Is there any reason SolveSpace might not be able to create files in your home directory? Are either of these env variables set? If so to what? Is there anything strange about your setup?

@robnee, I can't detect when SolveSpace 3.x nightly builds stopped to create `settings.json`, but I just tried "clean install" of few previous nightly builds — seems like issue introduced before SolveSpace `3.0~`https://github.com/solvespace/solvespace/commit/2939abf5f876a7454e0afbc06b32c888989b6546 (as it is latest build I has stored & just tested) already has NOT creates `settings.json`, but could create empty `~/config/solvespace` on first install.

robnee, 2021-04-25:
And what messages were displayed to the console if any?

Symbian9, 2021-04-25:
Here is output ("Start SolveSpace" ; "Change stipple type in hidden syle" ; "Exit"):

```
$ solvespace
SolveSpace!

json_object_from_file: error opening file /home/user/.config/solvespace/settings.json: No such file or directory

(solvespace:8102): Gtk-WARNING **: 22:14:17.677: gtk_window_present_with_time() should not be called with 0, or GDK_CURRENT_TIME as a timestamp, the timestamp should instead be gathered at the time the user initiated the request for the window to be shown
connect failed: No such file or directory
Segmentation fault
```

robnee, 2021-04-25:
It looks like it's crashing on exit before writing the settings.json.  Can you please remove ~/.config/solvespace and retry?  I'd like to see if it recreates that directory.

robnee, 2021-04-25:
@Symbian9 Any chance you can run with valgrind?  i.e.

```
valgrind solvespace
```

You may need to install it.

Symbian9, 2021-04-25:
> Any chance you can run with valgrind?

@robnee, Yes.

0. Empty `~/.config/solvespace/` removed;
1. Started SolveSpace (just after fully "clean install" of `v3.0~9e8c3db8`):
2. Changed *stipple type* in `s008-#def-hovered` & closed "Poperty Browser";
3. "Exit" from SolveSpace;

<details><summary>Terminal Log:/summary>

```Shell
$ valgrind solvespace
==9289== Memcheck, a memory error detector
==9289== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==9289== Using Valgrind-3.14.0 and LibVEX; rerun with -h for copyright info
==9289== Command: solvespace
==9289== 
SolveSpace!

json_object_from_file: error opening file /home/user/.config/solvespace/settings.json: No such file or directory
Generate::ALL (for bounding box) took 280 ms
Generate::ALL took 315 ms

(solvespace:9289): Gtk-WARNING **: 23:27:30.124: gtk_window_present_with_time() should not be called with 0, or GDK_CURRENT_TIME as a timestamp, the timestamp should instead be gathered at the time the user initiated the request for the window to be shown
connect failed: No such file or directory
==9289== Conditional jump or move depends on uninitialised value(s)
==9289==    at 0x134D9C7C: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x1347F021: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x135BE6A4: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x135B5E14: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x134A66DF: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x134A9180: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x134A92B1: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x139C1477: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x139C1A47: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x139C3FF4: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x139C4105: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x135AA250: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289== 
==9289== Conditional jump or move depends on uninitialised value(s)
==9289==    at 0x134D9C82: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x1347F021: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x135BE6A4: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x135B5E14: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x134A66DF: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x134A9180: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x134A92B1: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x139C1477: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x139C1A47: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x139C3FF4: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x139C4105: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x135AA250: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289== 
==9289== Conditional jump or move depends on uninitialised value(s)
==9289==    at 0x134792D5: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x134D88E7: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x13479821: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x134731CC: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x13473789: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x135B2C99: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x139C1AA5: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x139C3FF4: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x139C4105: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x135AA250: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x12F09610: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
==9289==    by 0x168FFB: ??? (in /usr/bin/solvespace)
==9289== 
==9289== Conditional jump or move depends on uninitialised value(s)
==9289==    at 0x134793A1: ??? (in /usr/lib/x86_64-linux-gnu/dri/r300_dri.so)
[... 69 lines cut ...]
==9289==    by 0x59A15F7: g_object_run_dispose (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.5800.3)
==9289== 
==9289== HEAP SUMMARY:
==9289==     in use at exit: 68,508,585 bytes in 104,450 blocks
==9289==   total heap usage: 688,477 allocs, 584,027 frees, 218,096,178 bytes allocated
==9289== 
==9289== LEAK SUMMARY:
==9289==    definitely lost: 26,992 bytes in 11 blocks
==9289==    indirectly lost: 12,343 bytes in 508 blocks
==9289==      possibly lost: 8,201,085 bytes in 46,990 blocks
==9289==    still reachable: 59,819,117 bytes in 54,023 blocks
==9289==                       of which reachable via heuristic:
==9289==                         length64           : 37,448 bytes in 494 blocks
==9289==                         newarray           : 2,368 bytes in 68 blocks
==9289==                         multipleinheritance: 14,352 bytes in 172 blocks
==9289==         suppressed: 0 bytes in 0 blocks
==9289== Rerun with --leak-check=full to see details of leaked memory
==9289== 
==9289== For counts of detected and suppressed errors, rerun with: -v
==9289== Use --track-origins=yes to see where uninitialised values come from
==9289== ERROR SUMMARY: 1055 errors from 7 contexts (suppressed: 0 from 0)
Segmentation fault
```

</details>

robnee, 2021-04-25:
@Symbian9 thanks.  This seg fault seems unrelated to saving the config on exit.  You can try adding `--num-callers=50` to valgrind to get a deeper stack trace but it won't be that helpful without debug symbols.  Does SS _always_ crash for you on exit regardless of what you do?  Can you build SS yourself with Debug or do you have access to debug builds?  Can you run X with GL or hardware acceleration disabled to see if that makes a difference?  It seems like a problem with your Linux install vs SolveSpace at this point.

Symbian9, 2021-04-25:
> Does SS always crash for you on exit regardless of what you do?

I guess I spotted `Segmentation fault` in terminal on "Exit" for a while, but can't tell when it was introduced for the first time.

> Can you build SS yourself with Debug

I has SolveSpace `3.0~`https://github.com/solvespace/solvespace/commit/10ca307d986ab0c52faaab4fcaedb26bd42d1fd6 build with debug symbols.

@robnee, may you take a look on my previous backtrace log (https://github.com/solvespace/solvespace/issues/960#issuecomment-812181934) — maybe it already include needed info; and if not, tell me what should I do with `gdb` and/or `valgrid` after "clean install" `solvespace-dbgsym_3.0.0~git1710-10ca307-1_buster_amd64`.

robnee, 2021-04-25:
We need to know if SS always seg faults on exit regardless of what you do.  Does it seg fault if you simply start and then immediately exit?  If so does the one with debug symbols do the same?  If so run valgrind --num-callers=50 on that.  Maybe try on a separate clean install of Debian or another Linux disto.

Symbian9, 2021-04-25:
> Does it seg fault if you simply start and then immediately exit?

Seems like yes.

Here are logs for SolveSpace `3.0~10ca307` with debug symbols on "clean install":

<details><summary>GDB Output</summary>

```Shell
user@user:~
$ gdb solvespace
GNU gdb (Debian 8.2.1-2+b3) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from solvespace...Reading symbols from /usr/lib/debug/.build-id/03/16c1010af7886af3059da7b2af8402e0f297f5.debug...done.
done.
(gdb) run
Starting program: /usr/bin/solvespace 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffedaeb700 (LWP 11960)]
[New Thread 0x7fffed2ea700 (LWP 11961)]
SolveSpace!

json_object_from_file: error opening file /home/user/.config/solvespace/settings.json: No such file or directory
Generate::ALL (for bounding box) took 37 ms
Generate::ALL took 43 ms
[New Thread 0x7fffe6339700 (LWP 11962)]
[New Thread 0x7fffe59f7700 (LWP 11963)]
[New Thread 0x7fffe51f6700 (LWP 11964)]
[New Thread 0x7fffe49f5700 (LWP 11965)]
[New Thread 0x7fffdaeea700 (LWP 11966)]

(solvespace:11956): Gtk-WARNING **: 01:07:04.994: gtk_window_present_with_time() should not be called with 0, or GDK_CURRENT_TIME as a timestamp, the timestamp should instead be gathered at the time the user initiated the request for the window to be shown
[New Thread 0x7fffd8e71700 (LWP 11976)]

Thread 1 "solvespace" received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()
(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x00007fffe6b5e19b in ?? () from /usr/lib/x86_64-linux-gnu/dri/r300_dri.so
#2  0x00007fffe65f117e in ?? () from /usr/lib/x86_64-linux-gnu/dri/r300_dri.so
#3  0x00007fffe65f244c in ?? () from /usr/lib/x86_64-linux-gnu/dri/r300_dri.so
#4  0x00007fffe65cc8be in ?? () from /usr/lib/x86_64-linux-gnu/dri/r300_dri.so
#5  0x00007fffe6b5c7b3 in ?? () from /usr/lib/x86_64-linux-gnu/dri/r300_dri.so
#6  0x00007fffec18114f in ?? () from /lib/x86_64-linux-gnu/libGLX_mesa.so.0
#7  0x00007fffec170c09 in ?? () from /lib/x86_64-linux-gnu/libGLX_mesa.so.0
#8  0x00007ffff6fb654d in ?? () from /lib/x86_64-linux-gnu/libgdk-3.so.0
#9  0x00007ffff6e47db3 in g_object_unref () from /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#10 0x00007ffff723f612 in ?? () from /lib/x86_64-linux-gnu/libgtk-3.so.0
#11 0x00007ffff6e495f8 in g_object_run_dispose () from /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#12 0x00007ffff7beeec8 in Gtk::Object::_release_c_instance() ()
   from /lib/x86_64-linux-gnu/libgtkmm-3.0.so.1
#13 0x00007ffff7b4ccbc in Gtk::GLArea::~GLArea() () from /lib/x86_64-linux-gnu/libgtkmm-3.0.so.1
#14 0x00005555555ceaa7 in SolveSpace::Platform::GtkGLWidget::~GtkGLWidget (this=0x555555d23e18, 
    __in_chrg=<optimized out>, __vtt_parm=<optimized out>) at ./src/platform/guigtk.cpp:443
#15 SolveSpace::Platform::GtkEditorOverlay::~GtkEditorOverlay (this=0x555555d23df0, 
    __in_chrg=<optimized out>, __vtt_parm=<optimized out>) at ./src/platform/guigtk.cpp:610
[... 426 lines cut ...]
==12308== HEAP SUMMARY:
==12308==     in use at exit: 68,869,106 bytes in 106,293 blocks
==12308==   total heap usage: 772,239 allocs, 665,946 frees, 243,631,282 bytes allocated
==12308== 
==12308== LEAK SUMMARY:
==12308==    definitely lost: 30,884 bytes in 19 blocks
==12308==    indirectly lost: 26,419 bytes in 1,075 blocks
==12308==      possibly lost: 8,215,200 bytes in 47,099 blocks
==12308==    still reachable: 60,141,027 bytes in 55,127 blocks
==12308==                       of which reachable via heuristic:
==12308==                         length64           : 37,448 bytes in 494 blocks
==12308==                         newarray           : 2,368 bytes in 68 blocks
==12308==                         multipleinheritance: 14,448 bytes in 173 blocks
==12308==         suppressed: 0 bytes in 0 blocks
==12308== Rerun with --leak-check=full to see details of leaked memory
==12308== 
==12308== For counts of detected and suppressed errors, rerun with: -v
==12308== Use --track-origins=yes to see where uninitialised values come from
==12308== ERROR SUMMARY: 1106 errors from 7 contexts (suppressed: 2 from 2)
Segmentation fault
user@user:~
$ 
```

</details>

robnee, 2021-04-26:
@Symbian9 This looks like it has something to do with destroying the editor overlay for the Property Browser when tearing down the GUI.  This may be a waste of time but can you try the following with gdb and the debuggable 3.0~10ca307:

1. start SolveSpace
2. go to config screen
3. click on any of the [change] buttons to bring up the floating text field (editor overlay)
4. dismiss it by hitting esc or clicking away
5. exit SolveSpace

I'm wondering if displaying the editor overlay at least once will prevent it from causing a crash on exit.

Symbian9, 2021-04-26:
> I'm wondering if displaying the editor overlay at least once will prevent it from causing a crash on exit.

Nope. Still `Segmentation fault` on "Exit".

```Shell
$ solvespace
SolveSpace!

json_object_from_file: error opening file /home/user/.config/solvespace/settings.json: No such file or directory

(solvespace:19516): Gtk-WARNING **: 10:22:36.126: gtk_window_present_with_time() should not be called with 0, or GDK_CURRENT_TIME as a timestamp, the timestamp should instead be gathered at the time the user initiated the request for the window to be shown
Segmentation fault
```

JFTR, In backuped `~/.config/solvespace/` directory I found that `settings.json` config file last time saved/overwritten on March 29, 2021.

So, some previous nightly builds (I always use latest fresh builds from @AlexanderP's PPA) was able to at least overwrite existed configs `settings.json` from earlier installs.

> This may be a waste of time

Not for me — I'm interested in discovering what has happen, so I may do more test.

Any test which should I do with actual install?

Also I will try to install SolveSpace in MX-Linux LiveUSB mode with "clean distribution" ISO.

Symbian9, 2021-04-26:
@robnee, I just tested "clean install" *SolveSpace* `3.0~9e8c3db8` in LiveUSB mode on "clean install" *MX-Linux `19.3` (`MX-19.3_ahs_x64.iso` - https://sourceforge.net/projects/mx-linux/files/Old/MX-19.3/) — SolveSpace does NOT reach `Segmentation fault` on "Exit" & ***could*** create `settings.json`!

BUT *on my actual system* (*MX-19.3* with installed *regular updates of system packages*) *SolveSpace* reaches `Segmentation fault` and as a result it could NOT create `settings.json`.

So, `Segmentation fault` "probably" introduced due to some system packages update (maybe graphics drivers, etc.) — I'm not sure how should I investigate which package update introduced this "issue".

Symbian9, 2021-04-26:
@robnee, So, I just fully reinstalled my actual *MX Linux* `MX-19.3` with a "clean install" distribution of `MX-19.4`.

Now *SolveSpace* issue with `Segmentation fault` on "Exit" disappear &  *SolveSpace* `3.0~9e8c3db8` on "clean install" successfully creates`settings.json` file!

@ruevs @phkahler, After all, merging of https://github.com/solvespace/solvespace/pull/1029 is required for fix initial hidden style issue & probably SolveSpace `v3.1` bugfix release should released after its merging — and this issue thread could be closed now ;)

P.S. According regular updates on fresh installed distribution I has pending updates of some firmware packages, so if after its update issue with `Segmentation fault` on "Exit" would back — I would create separate issue on this "bug" (then we could detect which packages affects SolveSpace `Segmentation fault`).

robnee, 2021-04-26:
> Now _SolveSpace_ issue with `Segmentation fault` on "Exit" disappear & _SolveSpace_ `3.0~9e8c3db8` on "clean install" successfully creates`settings.json` file!

That's great!  Does it also fix the issues opening the model of the sunglasses from an earlier issue?

Symbian9, 2021-04-26:
> Does it also fix the issues opening the model of the sunglasses from an earlier issue?

@robnee, There are NO `Segmentation fault` on opening testcase, but still are warnings about timers  in terminal output, see: https://github.com/solvespace/solvespace/issues/960#issuecomment-827068539

closed 2021-04-28 by ruevs

ruevs, 2021-04-28:
@jkrei0 and @Symbian9 please try try with the latest edge, The bug should be fixed in 709dc31f7865573cbf23c09dbd1946b017e7dc80.
Please note that you should delete your  `HKEY_CURRENT_USER\Software\SolveSpace\Style_HiddenEdge_StippleType` registry key on Windows or the equivalent in the .json file on Linux to reproduce the problem (and thus test the fix).

jkrei0, 2021-04-28:
Yep. It works as expected with the latest edge!

Symbian9, 2021-04-28:
>  `@Symbian9` please try try with the latest edge

@ruevs, I will try when it fresh DEB package would be built by @AlexanderP's PPA (as I avoid to use Snap/Flatpak builds, and there are no AppImage bulds yet).

PR #1029 (2021-04-28): Fix lookup of stipple pattern before config is available
Fix a bug with the lookup of stipple pattern for default styles when the config settings are not yet available (install/upgrade/first run). This caused hidden lines to display as continuous rather than dashed.

This fixes #1027