solvespace/solvespace #1028: Are usable tarballs not more maintained since 2.3?

nick87720z, 2021-04-24:
### System information

- **SolveSpace version:** 
- **Operating system:** 

### Expected behavior

Versions ending with 2.3 had some non-automatic tarball named solvespace-source.tar.gz or so.

### Actual behavior

solvespace-openmp.exe windows executable looks to be on place of previously existing tarball.

### Additional information

There could be separate, simple (probably hand-written) makefile for maintainance jobs, like `make dist` (for such a tarball), like maint.makefile (which could be also excluded from releases).

phkahler, 2021-04-24:
What kind of tarball are you looking for? Someone suggested we tar all the dependencies/submodules. Is that what you mean?

nick87720z, 2021-04-25:
Yes, release tarball should include submodules in order to be usable. For now version 2.3 - last, for which it exists.

Symbian9, 2021-04-25:
> Yes, release tarball should include submodules in order to be usable.

Guess, `solvespace-3.0-source.tar.gz` (with all submodules) should be attached manually to `v3.0` release as asset.

rpavlik, 2021-06-24:
yeah this is an annoying thing about github and submodules.

bobertlo, 2021-12-18:
A vendored release tarball would be ideal, but tagging the sub-module repositories would be a work-around without cutting a new release for now.

bobertlo, 2021-12-20:
I'm just going to leave this here in case anyone else ending up here finds it useful.

For the 3.0 release:

```
_libdxfrw_hash=0b7b7b709d9299565db603f878214656ef5e9ddf
_mimalloc_hash=4e643b6d3178e0ea2a093b7e14fe621631a91e4b
https://github.com/solvespace/libdxfrw/archive/${_libdxfrw_hash}.tar.gz
https://github.com/microsoft/mimalloc/archive/${_mimalloc_hash}.tar.gz"
```

closed 2022-02-06 by phkahler

phkahler, 2022-05-24:
@nick87720z  and @vuokko  does the current tarball in 3.1-rc1 suite your needs?

rpavlik, 2022-05-25:
Yeah, see the Debian package source for how we handle it there, the instructions are right there in the comment in the CMake file.

PR #1209 (2022-02-06): CI: Create tarballs with submodules included for releases
Closes #1028 

See https://github.com/ppd/solvespace/releases/tag/v8.5 for an exemplary release
commit 43795f5d (2022-02-06): CI: Create tarballs with submodules included for releases

Closes #1028