solvespace/solvespace #1030: Parametric Template Components

giannissc, 2021-04-27:
This is more of a question than a feature request. Is there a way to define a component that follows a pattern that can be exploited using repeat translate/rotate and have access to the repeat parameter when linking to that component?

Right now I have the following component which I import into reparate files to 2 step repeat, 3 step repeat, ..., 7 step repeat
[resistance-test-structures.zip](https://github.com/solvespace/solvespace/files/6385252/resistance-test-structures.zip)

I would have liked to be able to create the component just once and just set the repeat parameter as an output to my part to be specified when importing it.

phkahler, 2021-04-27:
What you want is related to issue #77 and #80 which are still a long way off. Unfortunately there is currently no way to override anything in a linked file via input from the assembly file.

You might be able to simplify things by starting with the repeating pattern in a 2D sketch, do the repetition in a step-repeating group, and then add the end features while still in the 2D repeat group. You could then extrude the whole thing - repeats and end features. You could then make copies of that one file only changing the repeat count for each variation.

I've attached a crude approximation to demo the construction method I mentioned.
[example.zip](https://github.com/solvespace/solvespace/files/6386039/example.zip)

The 2nd copy in the repeat group is constrained to the first copy to set the spacing. You can still change the g002 sketch and the rest will connect properly. The end features are preserved correctly as the number of copies changes because they are constrained to copy number 1 and copy number Last - which is recognized as such for constraints!

giannissc, 2021-04-27:
Ok then I will close this for now :)

closed 2021-04-27 by giannissc