solvespace/solvespace #1031: Feature request: use copy and paste to insert images in workplace

zhulaojianke, 2021-05-01:
When I need to insert a picture annotation in the workplace, if solvespace can read the image file directly from the clipboard, it can be inserted by simply copying and paste, which simplifies a lot than now. Then users don't have to save the screenshot for PNG files, and click the "sketch image from a file" icon to select the image file and put it in. This is a bit cumbersome.

It seems that if I insert an image directly from the comment box, then I'll can not resize the image. This is my guess, because I can't insert an image into the comment box at all. So I think it's a good idea to just use the current method of inserting images into the workspace.

ruevs, 2021-05-01:
Highly unlikely that this will happen. Cross platform clipboard handling of image data...

phkahler, 2021-05-01:
@ruevs  and it doesnt save image data in the .slvs file does it? Just a reference to the file?

ruevs, 2021-05-01:
Of course!

closed 2021-05-01 by ruevs

zhulaojianke, 2021-05-01:
Oh, I see. It means that when publishing some .slvs files, I must make sure that the relative path of the image file to the source file is the same. That's good, too.