solvespace/solvespace #1032: The comments can't move with the shapes

zhulaojianke, 2021-05-01:
I have made a 2D geometry sketch, and used the comments to mark the point name. When I moved the shapes by selecting and dragging them, the comments around the points didn't move at the same time. When I use the mouse to drag a box to select the whole shape, only the shape itself is selected, and the comment is not selected. And even I click the comments one by one and then selecting the whole shape, the shape and the comments still moved seperately.

I think if I could contrain the shape and the comments as one "group", or constrain the comments lock on the related points, this problem maybe be solved. I hope you can provide this feature.

Attached is the 2D geometric representation that I have drawn. I am trying to provide some basic parameter computing power in SolveSpace by means of plane geometry.
[image]

robnee, 2021-05-08:
@phkahler @ruevs I took a look at this.  When selecting with the marquee constraints are ignored.  The presumption is that if one moves the entities then the constraints will move with them anyway.  In fact, moving constraints with entities would double-count the distance change and end up moving them twice as far. Comment constraints are unique in that they are not associated with any entity.  They can be moved individually like other labels (such as distance constraint labels) but not as a group.  Changing how this works turns into a bunch of irritating edge-cases to deal with everywhere.

The more I looked at this though the more I think the solution is to turn this issue into a feature.  It's fairly simple to allow comment constraints to be associated with an entity.  I just needed to tweak how comment constraints are created and drawn.  I don't change any of the existing functionality but now permit one to _optionally_ select a point prior to creating a comment constraint to associate it with that point.  Dragging the point moves the constraints with it.  It seems like it could be a useful feature.

I've attached a video as a quick demo.  Please let me know what you think.  Also, I use the Gnome screen recorder to create the webm video.  Please let me know if it doesn't work.  GitHub doesn't seem to like .webm but accepts the video if renamed .mp4

https://user-images.githubusercontent.com/3731923/117518670-b7368900-af6e-11eb-8b8c-23b5a0769c40.mp4

ruevs, 2021-05-09:
@robnee your approach is exactly how I would have done it.

How would it affect backward compatibility of the files?

Symbian9, 2021-05-09:
> How would it affect backward compatibility of the files?

There no needs in backward compatibility — newer SLVS files should not be opened in older version of SolveSpace (actually we has implemented *warning message* for that).

As for opening older versions of SLVS in newer version of SolveSpace it could be solved just by adding some *warning message* on opening SLVS with comments that *some comments should be assigned* OR *some comments would not be imported*.

robnee, 2021-05-09:
@ruevs  I was thinking about compatibility too.  This does not need to introduce any new members to the `Constraint` class and hence does not change the file format.  I used `ptA` to hold the reference point so the introduction of a `Constraint.ptA` line is the only thing that changes as compared to a comment saved in version 3.0.
```
Constraint.h.v=00000002
Constraint.type=1000
Constraint.group.v=00000002
Constraint.workplane.v=80020000
Constraint.ptA.v=00040001
Constraint.other=0
Constraint.other2=0
Constraint.reference=0
Constraint.comment=NEW COMMENT -- DOUBLE-CLICK TO EDIT
Constraint.disp.offset.x=-27.99999879023074811357
Constraint.disp.offset.y=-17.16666592496290633107
AddConstraint
```
As luck would have it 3.0 already understands `ptA` and recreates the comment internally correctly with `ptA` holding the reference point.  It will even save the file out again with `Constraint.ptA`.  The only thing 3.0 doesn't do is render the comment offset for the reference point, since it assumes the offset is from the origin of the workplane.

This works about as gracefully as we could hope I suppose.  I will try to push a PR (as a work in progress) for you to take a look at later today.

PR #1035 (2021-05-12): Allow comments to be associated with point entities
If a single point is selected when a "Constrain | Comment" (`;`) is added
then the comment is associated with the point and its position becomes
relative to the point. In this way the comment will move with the point.

If nothing is selected or more than a single point is selected then
the behaviour is as before and the comment is "floating".

Closes #1032

Alternative to https://github.com/solvespace/solvespace/pull/1034
commit 7e08b02d (2021-05-12): Allow comments to be associated with point entities

If a single point is selected when a "Constrain | Comment" (`;`) is added
then the comment is associated with the point and its position becomes
relative to the point. In this way the comment will move with the point.

If nothing is selected or more than a single point is selected then
the behaviour is as before and the comment is "floating".

Closes #1032
closed 2021-05-12 by phkahler