solvespace/solvespace #1036: Windows error in solvespace 3.0 for windows 10.

orisvaldo19, 2021-05-12:
### System information

- **SolveSpace version:** 
- **Operating system:** 

### Expected behavior

The program open.

### Actual behavior

The error message to open solvespace im my windows 10 as is in attached image. The vc_redist.x86.exe is installed.

### Additional information

[image]

kk6mrp, 2021-05-12:
Your computer may not support a high enough OpenGL version. Can you try the attached version and report if this solves your issue?
[solvespace.zip](https://github.com/solvespace/solvespace/files/6463481/solvespace.zip)

Symbian9, 2021-05-12:
> Can you try the attached version and report if this solves your issue?

@kk6mrp, Please, describe what changes included in your build.

orisvaldo19, 2021-05-13:
Really. I don´t have a graphic card in this PC (is a reserve machine cause
my principal is with my son, a student and gamer), but the version that you
send me solves. Its open.
Thanks!

Orisvaldo Silva.

phkahler, 2021-05-13:
Maybe we need to update the web site to indicate an OpenGL / graphics requirement?

Another option is to make a release build for XP or whatever. That will eventually die but it's still possible apparently :-)

ruevs, 2021-05-13:
> Another option is to make a release build for XP or whatever.

I'll make a Windows XP compatible build of 3.0 with the Open GL1 renderer. Will ping you when I get to it (matter of of sitting on a machine which very rarely gets used these days) and you can decide how "official" you want to make it.

Adding the Open GL requirement on the download page is a good idea.

orisvaldo19, 2021-05-28:
Cool, thanks!

Symbian9, 2021-05-29:
> I'll make a Windows XP compatible build of SolveSpace 3.0 with the OpenGL1.x renderer.

@ruevs, It would be cool! What about OpenGL1.x-copmpatible build for Mac OS X? See: https://github.com/solvespace/solvespace/issues/102#issuecomment-261479046

> Adding the OpenGL requirement on the download page is a good idea.

Additionally, add info on compiling SolveSpace `3.x` with `OpenGL1.x` for all supported platforms with older GPUs.

phkahler, 2021-05-29:
@Symbian9 OpenGL 1.x is deprecated by all platforms. MacOS works fine with GL3 going way back. On Linux GL3 is supported on some pretty old hardware. The fact that @ruevs likes to keep it going on Windows XP is nice, but is possible as a result of SolveSpace moving slowly. We have no plans to move backward :-)

Things I consider deprecated:

OpenGL less than 3.0
X-windows
32 bit CPUs

Symbian9, 2021-05-29:
While SolveSpace 3.x could be built for deprecated OpenGL1.x, I suppose there should be a note in `README.md`.

akauppi, 2021-06-07:
@phkahler You mention X-windows (X11) as deprecated, but it's an essential part of [WSLg](https://devblogs.microsoft.com/commandline/wslg-architecture/).

phkahler, 2021-06-07:
> @phkahler You mention X-windows (X11) as deprecated, but it's an essential part of [WSLg](https://devblogs.microsoft.com/commandline/wslg-architecture/).

Yes, even in the Linux world X-windows is deprecated. I'm not saying its dead yet, and if people want to submit fixes for X-issues I'm fine with that. I personally don't see any reason to support it for WSL because we have native Windows builds, but again if someone wants to use it that way and it works that's fine. Pull Requests to support old stuff are welcome too, as long as they're not invasive.

akauppi, 2021-06-07:
@phkahler I was also wrong in my claim. Now, having read that page I referred to, it’s obvious that Microsoft and WSLg are seeing Wayland as the way, and X11 is nicely legacy supported. So my link proved my text wrong! 

I would like to have a discussion on wider project goals, but hijacking an Issue is the wrong way. I’ll get back, after checking the source code a bit more..

phkahler, 2021-06-08:
> I would like to have a discussion on wider project goals, but hijacking an Issue is the wrong way. I’ll get back, after checking the source code a bit more..

Another good place to look is at the 4.0 milestone issues. Those are ones I consider high impact even if they can't all make it. Some of them were originally targeting 3.0. The current goal is to make Solvespace better at what it already does and not try to turn it into a massive toolkit like FreeCAD. At some point we will have to make invasive breaking changes (hopefully with code for backward compatibility) but not for now.

rpavlik, 2021-06-24:
Interestingly, I don't think I've ever seen Solvespace run on Windows without using ANGLE, at least not in modern versions.

ruevs, 2021-08-08:
I created an unofficial Windows XP release with OpenGL 1.
https://github.com/ruevs/solvespace/releases/tag/WindowsXPPost3.0

Symbian9, 2021-08-08:
> I created an unofficial Windows XP release with OpenGL 1.

Great! Is it reproducable? If yes, it would be cool to add it to `edge` GH Actions.

It could be useful not only for Windows XP users, but also for those who use SolveSpace using WINE under older Linux & macOS on older hardware.

ruevs, 2021-08-09:
It can be reproducible, but as it is it is not.

To make it reproducible I need to create a patch that removes c4ca4be and this patch needs to be applied before compiling for XP. This is the commit that replaced the old memory allocator with mimalloc.

As is is there are a few Vista and later Windows features,  that mimalloc uses, that I had to stub out. I described them in the release.

Another alternative is to try to upstream Windows XP compatibility for mimalloc, but considering Microsoft's stance on XP, the mimalloc maintainers may not be interested.

ruevs, 2021-08-09:
And of course to compile for XP one needs the compatible platform toolset, the Visual Studio 2017 v141_xp toolset.

https://docs.microsoft.com/en-us/cpp/build/configuring-programs-for-windows-xp?view=msvc-160

In not sure what version of Visual Studio the GithubActions provide (or ours use?) but if it is 2017 or 2019 it needs the above add-on and I do not know if it can be installed/configured. @ppd ?

If it is 2015 it supports XP out of the box.

Symbian9, 2021-08-15:
> To make it reproducible I need to create a patch that removes https://github.com/solvespace/solvespace/commit/c4ca4be91e0f38646fa391aaca6fd58e505be4c4 and this patch needs to be applied before compiling for XP.

Here is "How To" generate a git patch from commit & apply it via git command:

- https://gist.github.com/emmanueltissera/19d0a8852f000cde13768dd7420a0906
  - https://stackoverflow.com/questions/6658313/generate-a-git-patch-for-a-specific-commit

rpavlik, 2021-08-24:
The GitHub Actions stuff tends to install basically every toolset, iirc. I don't remember what version we're building with. May just need to add a build that passes the toolset version to cmake: https://cmake.org/cmake/help/v3.21/manual/cmake.1.html  suggests it should be `-T v141_xp` though I haven't tested this.

ruevs, 2022-01-22:
Here is an updated unofficial Windows XP release with OpenGL 1.
https://github.com/ruevs/solvespace/releases/tag/WindowsXPPost3.0_2

Symbian9, 2022-01-23:
> Here is an updated unofficial Windows XP release with OpenGL 1

@ruevs, thanks! What about adding this build script & patches to SolveSpace's GitHub Actions CI for continuous "edge" release?

ruevs, 2022-01-23:
> What about adding this build script & patches to SolveSpace's GitHub Actions CI for continuous "edge" release?

@Symbian9 sometimes you are annoying https://github.com/solvespace/solvespace/issues/1036#issuecomment-850894499 :-)

I do the XP build, because I still can and because I like XP, but it is slowly dying. 

Presuming you will get an email notification with this I'll delete it in a few hours together with your comment :-)

Symbian9, 2022-01-24:
> sometimes you are annoying

Yeah, I know. Sorry me for that, please (^_^;)

Symbian9, 2022-02-18:
> Our missing dependencies on Windows 2000 are:

@ruevs, (*just theoretically*) could be those dependencies implemented as *module inside SolveSpace itself*?

ruevs, 2023-03-24:
Here is an updated unofficial Windows XP release with OpenGL 1.
https://github.com/ruevs/solvespace/releases/tag/WindowsXPPost3.1

ruevs, 2023-08-08:
> Visual Studio 2017 v141_xp toolset.

(Note to self) It seems the v141_xp toolset is not part of the default build environment.

https://github.com/actions/runner-images/blob/main/images/win/Windows2019-Readme.md

ruevs, 2024-10-26:
Here is an updated unofficial Windows XP release with OpenGL 1. https://github.com/ruevs/solvespace/releases/tag/WindowsXPPost3.1_2024

ruevs, 2025-10-15:
# SolveSpace for Windows 2000:

[SolveSpace_Win2000_ruevs_ae82fdf0.zip](https://github.com/user-attachments/files/22924921/SolveSpace_Win2000_ruevs_ae82fdf0.zip)

As "promised" [three and a half years ago](https://github.com/solvespace/solvespace/issues/1036#issuecomment-1044348682) :-)

Built from here: https://github.com/ruevs/solvespace/commits/Win2k/ https://github.com/ruevs/solvespace/commit/fb0b4554b418a01920d25adc34ba9b7faff97635 (based on this master https://github.com/solvespace/solvespace/commit/ae82fdf0bdd5956975a0b80c199abce6582a902c).

It works by linking against [my fork](https://github.com/ruevs/EnlyzeWinCompatLib/tree/ThreeMoreFunctions) of [EnlyzeWinCompatLib](https://github.com/enlyze/EnlyzeWinCompatLib) and [patching the OS version](https://building.enlyze.com/posts/modern-visual-studio-meets-ancient-windows/) in the [PE header](https://github.com/datadiode/PatchPE) to 4.0 (Windows 2000 is 5.0 but perhaps someone will try it on NT4?).

Of course [Dependency Walker](https://www.dependencywalker.com/) is the one tool to rule them all when attempting to do such things.

## Next pointless goal: Windows 95 :-)

[image]

[image]

ruevs, 2026-01-03:
@Neustradamus https://github.com/YuZhouRen86/VxKex-NEXT/issues/98#issuecomment-3706851021

ruevs, 2026-10-21:
Version 3.3 for Windows 2000 and later:
https://github.com/ruevs/solvespace/releases/tag/v3.3