solvespace/solvespace #1037: Rebase the emscrupten port to master.

phkahler, 2021-05-14:
The emscripten port needs work but is out of date. We should merge it and write some build instructions so it will be easier for people to work on.

The bit-rot is mostly in the build related files. The UI is in entirely new files due to the platform abstraction layer.

Even though it's not usable yet, I think it should be merged to help prevent such divergence from happening again.

akauppi, 2021-06-08:
Before merging, the branch should compile. 

Comparison here: 
https://github.com/solvespace/solvespace/compare/emscripten

If the aim is to merge, maybe someone does a PR between `emscripten` and `master`. That way, it would be easier to comment the right locations, line-wise.

Here are my findings, on Ubuntu 20.04 LTS (under WSL2).

## `README`

The section is now added under Linux build instructions, but has the same header level as "Building on Linux". Either add one `#` to `## Building for web` or - I would - move the section below all the OS specific things, to be its own, separate `##` level sectoin.

### Build instructions

I needed to change this:

```
./emsdk install latest
./emsdk activate latest
```

The tool `upstream-latest` did not exist. `esmsdk` 2.0.23 (`f26ebb...`).

**Use same terminal**

I would add a notion that the `emsdk` commands need to be given in the same terminal as where SolveSpace is going to be compiled. It's currently not stated.

**`git submodule update --init`**

I would add `--init` to the instruction that now only says `git submodule update`. Struggled with this for a while - not that familiar with git submodules.

---

The rest works, and I get:

```
 $ ls bin/
CDemo.html        libpixman-1.a              solvespace-benchmark.js    solvespace-debugtool.wasm  solvespace.wasm
CDemo.js          libpng16.a                 solvespace-benchmark.wasm  solvespace-testsuite.html  solvespaceui.css
CDemo.wasm        libslvs.a                  solvespace-cli.html        solvespace-testsuite.js    solvespaceui.js
libcairo.a        libsolvespace-core.a       solvespace-cli.js          solvespace-testsuite.wasm
libdxfrw.a        libsolvespace-headless.a   solvespace-cli.wasm        solvespace.data
libflatbuffers.a  libzlibstatic.a            solvespace-debugtool.html  solvespace.html
libfreetype.a     solvespace-benchmark.html  solvespace-debugtool.js    solvespace.js
```

Launching `solvespace.html` in Edge gives:

[image]

..but I'm personally mostly interested in the build chain, so this might be enough for me, for now.

## "emscripten" is misspelled in the issue title

kk6mrp, 2021-06-08:
> Launching solvespace.html in Edge gives:

I built the Emscripten branch recently and had the same result. Something changed...

phkahler, 2021-06-08:
@akauppi  whitequark is no longer working on solvespace, but you can still try an older version of the emscripten version linked in   [this tweet](https://twitter.com/whitequark/status/1020156780201107463?lang=en)

This is totally out of my territory, but I'd like people to be able to build and run like that with master, obviously not for real use but to make it easy to improve that version to make it usable. If anyone offers a PR that brings emscripten up to the level of master I'd like to merge it. I didn't try to merge the branch myself because I'm really not familiar with all the build stuff and that's where the merging needs to happen.

akauppi, 2021-06-09:
I'm not knowledgable about `cmake` builds, either.

My sole interest is to be able to use the (2D) solver within SolveSpace, within a JavaScript app. But the UI would come from web mechanics (normal web app).

FYI; I set up a Dockerfile to help in the compilation. That repo's not public, at the moment but I'm happy to share it if someone wants to see the steps, for the good of merging the branches here.

Using Docker has some good consequences: one can easily see which libraries are really needed, and which are not. Collecting them to a `README` is prone to break, since it's not executable and relies on the libraries the developer already happened to have installed.

phkahler, 2021-06-09:
@akauppi  you might want to have a look at a similar solver written in Rust:  https://github.com/Michael-F-Bryan/constraints

Rust to WASM is supposed to be easy, but I've not tried.

vespakoen, 2021-06-24:
If you want to work on emscripten, checkout my branch, I got it half-compiling? again with a newer emscripten version (in december 2020):

https://github.com/vespakoen/solvespace/commit/91df00be073cc29ac04602ae8e9231beb521174a

https://github.com/vespakoen/solvespace/commits/vespakoen-emscripten

Better to diff this branch with the current master, looks like I rebased it at the time...

mmiscool, 2021-07-02:
I compiled and got the same error. 
Also some thing looks weird as the output after running the dev server thing to host the wasm.emrun if you run "build/bin/solvespace.html" I got the following output.
```
127.0.0.1 - - [02/Jul/2021 20:21:45] code 400, message Bad request version ("`)÷\x11¬Æ\\wçhÃ\x91ö#¾Qî'ÝK%\x06Ó\x98R\x992¿/r@\x00>\x13\x02\x13\x03\x13\x01À,À0\x00\x9fÌ©Ì¨ÌªÀ+À/\x00\x9eÀ$À(\x00kÀ#À'\x00gÀ")
127.0.0.1 - - [02/Jul/2021 20:21:45] "2.&xÊCP}¡w>-à¥:"¬¤çû  `)÷¬Æ\wçhÃö#¾Qî'ÝK%Ó2¿/r@>À,À0©Ì¨ÌªÀ+À/$À(kÀ#À'gÀ" 400 -
127.0.0.1 - - [02/Jul/2021 20:21:45] code 400, message Bad request version ("1F\x99\x12Ãû/\x99öKßé\x11\x83Y½\x8d¢¶B\x00>\x13\x02\x13\x03\x13\x01À,À0\x00\x9fÌ©Ì¨ÌªÀ+À/\x00\x9eÀ$À(\x00kÀ#À'\x00gÀ")
127.0.0.1 - - [02/Jul/2021 20:21:45] "2.FgQãÁTÜÀ[H¾ÀIÃÖÌè(iÖêp1Y 4Ä½³0+v1c1FÃû/öKßéY½¢¶B>À,À0©Ì¨ÌªÀ+À/$À(kÀ#À'gÀ" 400 -
127.0.0.1 - - [02/Jul/2021 20:22:01] code 400, message Bad request version ("Â\x9a\x12\x8f-\x00ªÇl6\x00Ñ¡¿\x86Ó¤\x17X·Id9\x80\x19¿úM\x94\x9fío\x00>\x13\x02\x13\x03\x13\x01À,À0\x00\x9fÌ©Ì¨ÌªÀ+À/\x00\x9eÀ$À(\x00kÀ#À'\x00gÀ")
127.0.0.1 - - [02/Jul/2021 20:22:01] "2.¥kñæb £¨ÉeÎ
¡+;-ªÇl6Ñ¡¿Ó¤X·Id9¿úMo>À,À0©Ì¨ÌªÀ+À/$À(kÀ#À'gÀ" 400 -
127.0.0.1 - - [02/Jul/2021 20:22:01] code 400, message Bad request version ("oÇ\x015jÉ6Ð1\x15\x83ÖÖ\x00ªÿ\x9f;P¸Âc\x9cz¨PXv@!8;\x00>\x13\x02\x13\x03\x13\x01À,À0\x00\x9fÌ©Ì¨ÌªÀ+À/\x00\x9eÀ$À(\x00kÀ#À'\x00gÀ")
127.0.0.1 - - [02/Jul/2021 20:22:01] "2.Ëx× Ão¡âkmh_e²_«ö^}; oÇ5jÉ6Ð1ÖÖªÿÂcz¨PXv@!8;>À,À0©Ì¨ÌªÀ+À/$À(kÀ#À'gÀ" 400 -
```

akauppi, 2021-07-05:
@phkahler Thanks for the Rust link. I'll take it in to the [Solver shootout](https://github.com/akauppi/solver-shootout) that I just started. 

The aim: to find a 2D solver suitable to be used in JS. :)

phkahler, 2021-07-06:
> @phkahler Thanks for the Rust link. I'll take it in to the [Solver shootout](https://github.com/akauppi/solver-shootout) that I just started.
> 
> The aim: to find a 2D solver suitable to be used in JS. :)

That's awesome!  In the link I provided, he created an algebraic constraint solver. I'm not sure if he ever completed geometric constraints on top of that, but it was his goal to use it in his "arcs" 2D drawing program.

phkahler, 2022-08-04:
@akauppi  @mmiscool  @kk6mrp  @vespakoen  Check out this new PR #1275 

I'm not even sure how to review/evaluate it ;-) His demo link does run in my browser with an unfortunate mouse offset. It would be nice to get confirmation that others can build it successfully.

akauppi, 2022-08-04:
@phkahler I’ve forgotten why I needed this. While still active with CAD and JavaScript, I doubt I will have a go..

ruevs, 2022-08-21:
Done here https://github.com/solvespace/solvespace/pull/1275 and here f7415048a5769c78fc4bb02afa426d101059374d

closed 2022-08-21 by ruevs