solvespace/solvespace #1039: angle constraint can not work

lelehehele, 2021-05-16:
[image]
there is a problem, when I change the angle from 20 to 60, or big, it cant not work, but if from 20 to 30, then 30 to 40..... it can work
thanks!
[image]

zhulaojianke, 2021-05-18:
I cannot reproduce this issue in win7  version e74e2024 :(

phkahler, 2021-05-18:
@lelehehele  I created a similar diagram and was able to change the angle from 20 to 80 degrees in one step, but not from 10 to 90. I'm not sure what other constraints you've got in there, but this is just the solver having difficulty finding the solution due to the large change and other constraints present. You've already found the solution - make several smaller changes to reach your desired angle. Another option is to drag the sketch close to how you want it prior to adding the constraint.

closed 2021-05-20 by phkahler

xionghuijiang, 2021-05-24:
Actually，there are other constraints