solvespace/solvespace #1041: Add dialog to re-setup keyboard shortcuts

Symbian9, 2021-05-18:
### System information

- **SolveSpace version:** `3.0`
- **Operating system:** All

### Expected behavior

There should be next:

- Dialog to re-setup keyboard/key shortcuts in SolveSpace (user may customize it);
- All operations should be able to has a key shortcut, even if by default there are no shortcut (so user may set one).

### Actual behavior

Actually there are no ways to resetup keyboard/key shortcuts in SolveSpace (as all are hardcoded) and in same time some operations has no key shortcuts at all.

### Additional information

- https://github.com/solvespace/solvespace/issues/1019

closed 2021-05-18 by phkahler

ruevs, 2021-05-18:
https://github.com/solvespace/solvespace/issues/1019#issuecomment-828540005

Symbian9, 2021-05-19:
> That is true, but it is out of scope at this point.

The cost in few additional megabytes is nothing in comparison to such a "must have" feature for SolveSpace users to be able to customize shortcuts.