solvespace/solvespace #1042: A method to solve the problem about chinese character displaying wrong in comments

zhulaojianke, 2021-05-18:
SolveSpace version 3.0 b5fb1dd4
Windows 7 ultimate sp1

I've found chinese character displaying correctly in the property browser, but wrong in the comments. Like the picture below, the same chinese character displayed in the browser and in the comment.
[image]
I think if you change the chinese character encoding in the comments as same as in the property browser, this problem may be solved.

Because today I draw a machine drawing, and my workmate ask me to write some chinese comment on the drawing, and I told him that the solvespace's comment is not support chinese characters, and it was embarrassing for me.

By the way, how to input multiline comments in one comment input box? Or how do I wrap a line in a comment?

phkahler, 2021-05-18:
The text window uses the GNU Unifont which contains every glyph in the unicode Multilingual Plane. This is a bitmap font so it is not really resizable. The sketch window comments use a vector font that came from the LibreCAD project and does not have so many glyphs defined (I recently added 3 new ones for hole annotations). Fonts in that vector format are not common so getting better character coverage would probably require changing how Solvespace renders fonts on the sketch (a big change).

Multi-line comments don't seem to be supported, but I could be wrong (anyone know how?). This is something I'd like to add if it is not already possible. Hole annotations are frequently multi-line, and it's obviously useful for other things too.

ruevs, 2021-05-19:
> Multi-line comments don't seem to be supported, but I could be wrong (anyone know how?).

They are not. Changing the vector font is not worth it in my opinion. At least not at this point. This has been discussed in the past, but I can not find it now.

phkahler, 2021-05-19:
@zhulaojianke  There is a new feature where you can select a point and then press semicolon ;  And the comment will be positioned relative to that point. You may create multiple comments and move them all together by moving the point. It's not as nice as a real multi-line comment but it may help.

This is not available in verison 3.0, you'll need to use an Edge build:
https://github.com/solvespace/solvespace/releases/tag/edge

zhulaojianke, 2021-05-19:
Haha, thanks a lot! I've downloaded and used it a day ago. This new feature allows me to explain my ideas more clearly on the drawings. Thank you for your help! :)

ruevs, 2021-05-20:
The original [unicode.lff](https://github.com/LibreCAD/LibreCAD/blob/master/librecad/support/fonts/unicode.lff) has some updates:
  https://github.com/LibreCAD/LibreCAD/commits/master/librecad/support/fonts/unicode.lff
after ours: 
  https://github.com/solvespace/solvespace/commits/master/res/fonts/unicode.lff.gz

Perhaps we should at least update it and maybe try to upstream our fixes and additions?

phkahler, 2021-05-20:
> Perhaps we should at least update it and maybe try to upstream our fixes and additions? Not that it will help with the missing Chinese characters...

I'm in favor of that. LibreCAD could probably benefit from having the glyphs for depth, countersink, and counterbore. Maybe those should be contributed back to them and then we pull the fully updated font again?

phkahler, 2021-05-20:
Oh look:  https://github.com/LibreCAD/LibreCAD/tree/master/tools/ttf2lff
They have a ttf to lff converter!

I'm not sure the lff file can go beyond 0xFFFF but filling in more characters seems like a good idea.

Symbian9, 2021-05-20:
> Oh look: https://github.com/LibreCAD/LibreCAD/tree/master/tools/ttf2lff

`unicode.lff` is a single stroke font — that is why the best way to adding missed characters to single stroke `unicode.lff` fonts is by editing LFF-file in LibreCAD.

`ttf2lff` tool in most cases used for as TTF fonts ("filled fonts") into double stroke fonts ("outlined fonts") suitable for plotters / CNC, where single stroke fonts are required; also there are more converters such as `jhf2lff`:

- https://github.com/fontforge/fontforge/issues/3187
- http://web.archive.org/web/20200912190715/https://github.com/whitequark/solvespace/issues/12

> Perhaps we should at least update it and maybe try to upstream our fixes and additions? Not that it will help with the missing Chinese characters...

There is already `wqy-unicode.lff` which include mostly all CJK characters and is *double stroke font* (NOT a single stroke!) for LibreCAD, converted from [WenQuanYi](http://wenq.org/wqy2/index.cgi?action=browse&id=Home&lang=en)'s TTF-fonts using `ttf2lff` tool. But it is huge in uncompressed form due to `.lff` format (of course, it may be not an issue for those who who need CJK support for comments and would like to compile SolveSpace itself):

- https://blog.librecad.org/2011/11/complete-cjk-font-support-in-librecad/
  - https://sourceforge.net/projects/librecad/files/Resources/fonts/wqy-unicode.zip/download

### REFERENCES

> I ***highly recommend*** to read next linked pages & discussion threads firstly!

- https://forum.librecad.org/wqy-unicode-lff-td5712970.html
  - https://github.com/LibreCAD/LibreCAD/issues/957
- https://wiki.librecad.org/index.php/Fonts
  - https://librecad.readthedocs.io/en/latest/appx/fonts.html
- https://wiki.archlinux.org/title/Localization/Chinese
  - https://en.wikipedia.org/wiki/WenQuanYi

Symbian9, 2021-05-21:
### Bitmap font to LFF?

As for me, the best what we can do at this moment — try to convert CJK characters from "single stroke" bitmap *GNU Unifont* font (available in BDF and HEX) into *LFF font* using some additional tools/converters.

- https://en.wikipedia.org/wiki/GNU_Unifont (CJK characters merged from WenQuanYi's Unibit bitmap font )
  - https://en.wikipedia.org/wiki/WenQuanYi#Unibit

For example, here is GNU Unifont in a Hebrew font format (?) & in [PCB File Format]() + `pcbGlyphTool.c` (tool used for converting bitmap fonts from BDF format into Hebrew fonts & PCB File Format):

- http://vk5hse.blogspot.com/2014/09/geda-pcb-font-utilities-and-options.html
  - http://users.on.net/%7Eesh/geda/pcb/src/fonts/

I'm not sure are there is there `bdf2lff` or at least `bdf2jhf` converter (`jhf2lff` already exists), but that would be a good start point for *"Bitmap-to-LFF"* converting research... (need google more on this subject).

### P.S.

The weird tricky way:

1. try to convert WenQuanYi's or GNU Unifont's TTF file into Hershey font (`.jhf`) using next converter (instead of `ttf2lff` which create double stroke font from TTF by outlining characters paths, linked project tries to "vectorize TTF fonts" into single stroke font):
  -  https://github.com/LingDong-/chinese-hershey-font 
2. convert resulted Hershey font into LFF using `jhf2lff` tool.

BUT I'm not sure that a result would be satisfiable for CJK users.

ruevs, 2021-05-21:
Yeah, I also found wqy-unicode.lff but it is indeed 42MB uncompressed and almost 8MB as an "ultra" compressed `gz` file.
After the mentioned discussions it was eventually removed from LibreCad in 2016:
  https://github.com/LibreCAD/LibreCAD/commit/382744d4ca1c151ce911cdf516c2b6d073b67b0e
  https://sourceforge.net/projects/librecad/files/Resources/fonts/

@Symbian9 when you say that "unicode.lff is a single stroke font" do you mean that the glyphs have not "thickness"? And in "wqy-unicode.lff [] is double stroke font" they have a contour?

Symbian9, 2021-05-21:
> when you say that "unicode.lff is a single stroke font" do you mean that the glyphs have not "thickness"? And in "wqy-unicode.lff is double stroke font" they have a contour?

@ruevs, Yep, `unicode.lff` and `wqy-unicode.lff` are rather different.

See: https://wiki.librecad.org/index.php?title=File:LibreCAD_Fonts_v2p1p3.png

liuxilu, 2023-05-03:
I just composed a lff font form [Mingti-basic.hf.txt](https://github.com/LingDong-/chinese-hershey-font/blob/master/dist/hershey/Mingti-basic.hf.txt) (CJK URO and Ext-A)  and current `unicode.lff`.

The current `unicode.lff.gz` is 472kB, and my font is 1160kB with default compression, adding 688kB. I think it's affordable. Converting from other fonts should have similar size as well.

Many Chinese chars in current unicode.lff are even worser than resampling a bitmap font (although unnoticeable on small size). So I replaced them all. For example 悔:
[image]
left: unicode.lff; middle: normal font; right: new font

[new unicode.lff.gz](https://github.com/solvespace/solvespace/files/11379045/unicode.lff.gz)
[executable with new font](https://github.com/solvespace/solvespace/files/11379033/solvespace-font.exe.gz)

phkahler, 2023-05-04:
@liuxilu  I'm waiting for clarification on the font license. They show MIT for the software, so if that applies to the font we should be able to (I think) just incorporate this change. When you say it's an increase of 688kb is that the effect on our executable size as well? That seems reasonable to me for a significant increase in usable characters. @ruevs what do you think?

BTW I showed a Chinese coworker the above 3 characters and he said the first one "I don't know what that is" the new one he's fine with. If the improvements to other glyphs are similar this is good!

ruevs, 2023-05-04:
Judging from this https://github.com/LingDong-/chinese-hershey-font#pre-compiled-fonts
[Mingti-basic.hf.txt](https://github.com/LingDong-/chinese-hershey-font/blob/master/dist/hershey/Mingti-basic.hf.txt) is based on https://zh.wikipedia.org/wiki/%E8%8A%B1%E5%9C%92%E5%AD%97%E9%AB%94
or https://fonts-jp.translate.goog/hanazono
And this page says:
> About Hanazono
> This font is a free kanji font.

[unicode.lff.gz](https://github.com/solvespace/solvespace/blob/master/res/fonts/unicode.lff.gz) is directly linked in our executable so it will indeed grow. After re-compressing with maximum compression the new font is 1079732 bytes; the old one is 482559 bytes. So ~583KiB more - not so bad.

I'll do some (very superficial) testing and sanity checking and create a pull request.

liuxilu, 2023-05-04:
I'm currently working on a new lff font converter, the display and size may be better if I got it work.

ruevs, 2023-05-04:
I did a few random spot checks and it indeed looks "similar enough".
No idea what it says :-)
[image]

seamt, 2023-12-01:
@liuxilu
> I just composed a lff font form [Mingti-basic.hf.txt](https://github.com/LingDong-/chinese-hershey-font/blob/master/dist/hershey/Mingti-basic.hf.txt) (CJK URO and Ext-A) and current `unicode.lff`.

I just tried your new unicode.lff in LibreCAD 2.2.0 and found it freezed LibreCAD.
I also found a workaround, that is, add an empty line ("\n") between each glyph paragraph.
Now, you font file can be used in LibreCAD.
Very exciting. Thanks for your good work.

A suggestion, if your new font be included in any apps, the file name should better use "mingti-basic.lff" than current"unicode.lff". The latter is confusing for 2 reasons.
1. as I know, all .lff fonts use unicode table.
2. your new font and current all .lff fonts have not included all unicode glyphs.

A question, how did your convert hershey font into lff? I cannot find a download of "jhf2lff" tool by Rallaz. If you have it, or other alternatives, could you share me a download link?

A hint, the below project is converter from AutoCAD shx font to LibreCAD lff font. It is still in progress. For now, it cannot handle CJK encodings, and have accidental alignment error for some fonts.
[](https://github.com/Kharabadze/SHX_font_converter)

ruevs, 2023-12-06:
> I'm currently working on a new lff font converter, the display and size may be better if I got it work.

@liuxilu did you make any progress with that. Or should we just use the new font you attached here? https://github.com/solvespace/solvespace/issues/1042#issuecomment-1532409515

seamt, 2023-12-16:
> @seamt I wrote [jhftool.py](https://github.com/solvespace/solvespace/files/13577502/jhftool.txt). License: public domain. Note: Everything hardcoded; Load into REPL to use.

Quite thank you. Your script really works.

ruevs, 2024-08-21:
@liuxilu your screen shot from December looked pretty nice, and even the first unicode.lff was already a dig improvement.

Would you be willing to generate a "current" version (and perhaps also post the current version of your tool for future reference) - we can merge it as it is. In the future when you make further improvements we can update the font again.

liuxilu, 2024-08-21:
@ruevs Please merge the font.
My program still not working on many chars, unable to generate a font.

ruevs, 2024-08-21:
I updated the font https://github.com/solvespace/solvespace/pull/1478 with all [upstream changes from LibreCAD](https://github.com/solvespace/solvespace/issues/1042#issuecomment-1534907994) and all [Chinese characters that @liuxilu updated and added](https://github.com/solvespace/solvespace/issues/1042#issuecomment-1532409515).

closed 2024-08-21 by ruevs