solvespace/solvespace #1046: Ability to independently show/hide: H/V constraints, dimensions, REF and comments

baryluk, 2021-05-30:
### System information

- **SolveSpace version:** 3.0-rc2
- **Operating system:** Debian testing

### Expected behavior

I really love SolveSpace, and it is my tool of choice for 2D and 3D designs, metal, 3D printing, woodworking, sketching, and even planning space use for many years, but there is one thing that bother me often, that is visual noise of all the constraints on the sketch.

There is a button to hide / show "All constraints and dimensions", and option to change the line style (including text style) of constraints and dimensions. But it is all or nothing. I wish there was a bit more flexibility here.

For example:

I would like to keep the dimensions, just hide H and V constraints. Often I have very rectilinear design, with many rectangles, that I already know have H and V on their side segments. 99% of the I don't need to see them. Even the line segments that I do manually, I often will see that it has H or V constraints, and once I know it has one (either automatic or manual), I really don't want to see it again, ever, until maybe the very last verification at some point later.

Similarly, I sometimes might want to see H and V constraints, but not dimensions or references. This would be exactly for the remaining 1% of cases, where I want to ensure that H and V constraints are where they should be, but other things should be hidden so I can easily verify it.

I want to hide dimensions and all constraints, but keep the "Comment" texts, still visible. This is again, for clarity, and sketching.

I also want to be able to hide equal length constraint markers (small perpendicular magenta line on line segments and arcs) . They are nice, and I love them, but there are times, where I want to hide them and only them.

And be able to hide point on line constraint markers (square around the point). Again, I love them, but often when I am done with them, I don't care, and wish to not see them 95% of the time, to focus on other things.

And I think it would be also nice to be able to separately show/hide "reference dimensions" independent from "normal dimensions".

On a related note, I wish there was easy way to set color and text size of Comments, independently of other constraints and dimensions style. Often I want Comment to be slightly bigger and bolder, but dimensions be smaller. And sometimes H/V constrain marker, even smaller (if shown). Just some ideas.

It would be best to be able to set this per SolveSpace model file.

Symbian9, 2021-05-30:
This feature request is what I'm itself wanted for SolveSpace!

phkahler, 2021-06-03:
@baryluk  There is a lot covered in this request. I think it's good that you put it all in one place because it's not really a series of specific features, but wants a rather comprehensive change in displaying constraints. The "On a related note" part about comments is the only part I'd consider separate ;-)

On reading it sounds like you want a lot of fine-grained ability to show/hide categories of constraints, perhaps H/V as one type, and points-on-X and equal length hash marks as another, with length and angles as another. I don't like the idea of adding little toolbar icons for each of those - lots of clutter - so I kept trying to think of something better.

What if we make all constraints relating to a selected entity visible and a different color? So even if all constraints are not-shown, selecting a line would show any length, perpendicular, parallel, angle, H, V, whatever applies to that line.  By showing these in a different color you could also take a drawing with all constraints shown and still easily see which ones apply to a given entity just by clicking it. This doesn't cover all of your use-cases but does it seem like a good start?

An add-on to the above idea would be to also show any new constraints that are created automatically. If you have constraint display turned off, you won't know if a new H or V was actually created (sometimes the mouse moves a little when you click), but if we showed them until the next mouse click you'd know.

Related: There are IMHO 2 significant uses for SolveSpace.  1) creating 3d models (and assemblies)  and 2) creating technical drawings. My suggestion above is clearly for use 1. The notion of more selectively handling the display of constraints is more of use 2.  BTW one thing I learned from Symbian9 for making great drawings is to create a 3D model, then make an extra group where you add all needed dimension constraints (with the allow redundant constraints checked for this group). It's not perfect but it covers use-case 2 fairly well and should work for 2D sketches as well.

baryluk, 2021-06-03:
> @baryluk There is a lot covered in this request. I think it's good that you put it all in one place because it's not really a series of specific features, but wants a rather comprehensive change in [...]

Yes, it is more of a idea / set of features in UI, because I don't know how to best implement it in UI, or if other people thing there are good reasons to not implement it at all.

> On reading it sounds like you want a lot of fine-grained ability to show/hide categories of constraints, perhaps H/V as one type, and points-on-X and equal length hash marks as another, with length [...]

Yes, many little icons in toolbar are not a good idea, not only it makes it cluttered, it makes it hard to decode what they do, and you start hovering over all icons to know what they do, wasting a lot of time. Maybe a list of few types in the configuration, or a group, listed by words, would be better.

Personally H/V markers hide/show would solve 90% of my clutter issues, and would make me happy. But, then, I don't see reason for not having fine grain control over few other types. That is why I mentioned them. I guess, having consistency would be nice.

> What if we make all constraints relating to a selected entity visible and a different color? So even if all constraints are not-shown, selecting a line would show any length, perpendicular, [...]

That is an interesting idea. I didn't think about doing it this way. It isn't really what I am looking for, but maybe it could cover some cases indeed.

> An add-on to the above idea would be to also show any new constraints that are created automatically. If you have constraint display turned off, you won't know if a new H or V was actually created [...]

Agreed. When can show H and V temporarily as the next line segment is being drawn, as they are being drawn, but after you click, we can stop showing them.

>  I learned from Symbian9 for making great drawings is to create a 3D model, then make an extra group where you add all needed dimension constraints (with the allow redundant constraints checked for this group). 

Yes, I have seen this method utilized, and it can produce really nice and cool drawings. But, it is usually something that can only be done when the design is finished. It can not be utilized to when doing the design itself.