solvespace/solvespace #1048: Feature request: workplace creation and keyboard keys to rotate

simrim1, 2021-05-31:
This piece of software is unbelievably simple, efficient and robust. I haven't come across any other 2d/3d design software with such a small footprint and an output of such level. I have been using it for 3d design ever since I downloaded and tried it out and never looked around for any other software except for toolpath generation. Once it is equipped with toolpath generation ( I hope it will), I doubt there will be any need to use another software.

Working with solvespace I have faced a dire need of two capabilities though which will make working with it very easy and much more productive.

1. Creating a new workplane based on the selection of a line and its end points. The workplane created this way will be perpendicular to the line at the end point which is closer to the viewer.
2. As we can create workplane in 3d, there is no way ( or at least I don't know If there is any ) to precisely rotate in 3d to create a workplane at the desired precise place so keyboard shortcuts will be extremely helpful as mouse rotation is not as precise and efficient as keyboard shortcuts.

Thanks.

phkahler, 2021-06-01:
The usual thing to do is make a pair of construction lines starting at a point (the end of your line). You can constrain them perpendicular or at a specific angle relative to something else. Then select the 2 lines and point and press Shif->W.

Without at least on line in the workplace there is no clear way to orient the workplane - left/right or up/down.

simrim1, 2021-06-02:
Yes we can create a workplane by adding two lines ( most of the times construction lines) at a particular point and adding angle constraints to get the required surfuce on which to create a workplane. This suggestion is to increase productivity without adding lines and constraints on them for a surface which is perpendicular to the desired line. This will help to decraese a bunch of lines in a complex design. The less number of lines is less clutter in 3d view. 

Selecting a single line is enough to create a workplane perpendicular to it. The suggestion for selecting its endpoints as well is to ensure the user intention of creating workplane perpendicular to it.

Again, this is just a suggestuion. I think this can be tagged as UI?

As for the keyboard shortcuts for creating workplane in 3d Is actually, in my humble oinion, a necessity. because you keep moving mouse left/right, up/down and in the corners to attain the visual surface, in 3d, where you want to create a workplane. Even then you may not be able to get the precise position. With keyboard shortcuts for rotation in 3d will help pinpoint the desired plane. This will obviously be tagged an enhancement.

Thanks.

MaxiPaille, 2021-06-06:
Hello guys,

I agree with simrim1, this software seems wonderfull. Very light, very powerfull. But to be honest, it's a bit painfull to use, especially with workplanes. I'm pesevering to learn it because it's the closest software i'm looking for right now.

I'm wondering why we can't use **Shitft + W** when selecting a point and a normal ?

[image]

Is this a simple lack of implementation? Or is there a technical constraint anywhere that i'm not aware of?

Thanks

simrim1, 2021-07-02:
I am very grateful to everyone who responded especially @phkahler who actually tried to implement it, although, not exactly what I requested but it gave a clue to make another feature to create a new workplane. I hope that actual request will be implemented sooner or later.

As for the other request for keyboard keys to rotate 3d view will extremely be useful. I think numpad keys can be used 1,2,3,4,6,7,8,9 for bottom left corner of the current view, down, bottom right corner of the current view, left, right, upper left corner, up and upper right corner of the current view respectively.

Individual keys for movement in size of pixels and in combination with Ctrl, Shift, Alt with rotation of 15, 30, 45 degrees. I don't if I have described it properly but it can be done differently as long as small movements with keys are possible.

I think this is easiest thing to for developers as rotation with mouse is already in place, it just need to be bound with keyboard keys.

If someone else think it will help in increasing productivity plz give a thump up.

Thx

phkahler, 2021-07-02:
> Selecting a single line is enough to create a workplane perpendicular to it.

This is not true. A workplane for sketching has a coordinate system (u,v). It is not just the standard mathematical definition of a plane. We need more information for the orientation of the u,v.  If not for that, I would agree that a line and point should be sufficient.

You can always use construction lines, which can have visibility turned off via the toolbar in the text window.