solvespace/solvespace #1050: Custom TTF fonts not showing up on Windows 10

piedoom, 2021-06-02:
### System information

- **SolveSpace version:** 3.0~0e0b0252
- **Operating system:** Windows 10 (Version 10.0.19041 Build 19041)

### Expected behavior

Installing any TTF font via opening and clicking "install", or dragging to the `%windir%/fonts` folder should make those fonts accessible through the type tool

### Actual behavior

The list of fonts remains unchanged, showing what appears to either be embedded or system default fonts

PS, thank you for maintaining solvespace, it's an awesome tool!

sokol07, 2021-12-02:
Stumbled upon the same issue, it is still valid.
The thing is that if you install a font clicking on it and selecting "install" (automatic Windows installation, Windows will place the .ttf file somewhere deep in the system:
[image]

Meanwhile, **Solvespace is fetching only the fonts placed in folder C:\Windows\Fonts** (not Users\<Username>\AppData\...)

You can make the font work in solvespace by opening a windows command prompt as administrator (you can't do this with file explorer as entering the C:\Windows\Fonts will turn the explorer into a font manager!) and manually copying the font to the desired folder, for example:
`copy <path to the downloaded font.ttf> C:\Windows\Fonts`

ruevs, 2021-12-02:
It never occureed to me to "install" a TTF font in Windows. One always simply copies them to C:\Windows\Fonts (since at least Windows 95).

sokol07, 2021-12-02:
It may depend on Windows version or configuration (accounts?) but on my Windows 10 Home when I open C:\Windows\Fonts I get such preview and there is no way to open "normal" view of the folder:

[image]

If I drag and drop (copy label when dropping) the .ttf file to this window the font is installed in directory C:\Users\<user>\AppData\... same as when I open the font preview (doubleclick) and hit "install" or click with right mouse button and select "install" from the context menu. It works the same if I copy the .ttf file to the C:\Windows\Fonts folder from the C:\Windows view. Because of this it is not possible for me to just copy the new font to the Fonts folder. (However, I remember that I have been doing this on one of my previous PCs so it must have been different on either Win7 or Vista).

I decided to look deeper into this "feature" and I found this topic on stackoverflow:
https://stackoverflow.com/questions/55264642/how-to-force-win10-to-install-fonts-in-c-windows-fonts

Choosing "install for all users"  works for me - the fonts is added to C:\Windows\Fonts.
However, this option is available only using the right mouse button and "install for all users" option or command line - installing from preview or drag-and-dropping into the fonts folder won't install the font in a way which would make it available for Solvespace.

And this is, I believe, the bottom line which could be added to the docs of Solvespace to clarify the font installation.

ruevs, 2021-12-03:
Indeed! At some point - hard to find exactly when - it became impossible to copy directly to C:\Windows\Fonts with Windows Explorer. It certainly works in XP. Here is another discussion https://inkscape.org/forums/questions/installed-fonts-do-not-appear/

# `Right-click the font | Install for all users` is the solution.

closed 2021-12-03 by ruevs

ruevs, 2021-12-03:
And here for the programmers is the "real" explanation http://archives.miloush.net/michkap/archive/2006/06/25/646701.html

piedoom, 2021-12-04:
Fantastic, thanks for the discussion and answers!