solvespace/solvespace #1051: Identical (overlapping) edges of different parts/volumes are collapsed

gnbl, 2021-06-02:
solvespace-openmp.exe 3.0~e74e2024

Edges that are identical seem to be suppressed, so multiple parts in an assembly appear to have one face (under the mouse cursor):

[image]

Is this behavior intended and why?
Personally, I'd like to see that edge.

gnbl, 2021-06-02:
Indeed, if the length of either edge is different, it is displayed.
Maybe a remnant of "union" behavior for modelling?

phkahler, 2021-06-02:
I have vague memory of causing this as a result of fixing something else that was worse. It's still a valid issue.

Somewhere in the edge classification and rendering code...

gnbl, 2021-06-03:
In case it helps, I have attached a small test case with an assembly of a cube and a boolean union.
[tmp.zip](https://github.com/solvespace/solvespace/files/6589827/tmp.zip)

- This only affects the resulting solid model edges, not individual group edges/outlines.
- It does not affect the faces, i.e. the union merges faces as expected, while the adjacent face of another entity remains separate.