solvespace/solvespace #1052: Question: Extrude along a surface

giannissc, 2021-06-03:
Can something like this be done with existing version of solvespace or will this be needed #439?

[fluidic-ink-display.zip](https://github.com/solvespace/solvespace/files/6591478/fluidic-ink-display.zip)

phkahler, 2021-06-03:
That is hard to do now. You could possibly make channels like that using a series of revolve groups combined as difference. See the path.zip file attached to one of the comments in #439. It won't be very easy because you'll need to have construction geometry including the center point of each curve, and each curved or straight segment will have to be a new group. The tool I hope to have one day will make this much easier.

You might also want to extend the sketch used to cut the channel above the top surface of the base to avoid coincident surface issues with NURB differences.

giannissc, 2021-06-03:
Is there a timeframe for this feature and is there a way I can help?

phkahler, 2021-07-02:
> Is there a timeframe for this feature and is there a way I can help?

No time frame. The way I want to implement this is quite complex but should be very satisfying to use.