solvespace/solvespace #1053: Application does not start

aelio, 2021-06-04:
### System information

- **SolveSpace version:** 3.0~0e0b0252
- **Operating system:** Windows 10 home 2004 build 19041.985

### Expected behavior

Application working

### Actual behavior

Application Error (message box) The application was unable to start correctly (0xc000007b) Click OK to close the application

### Additional information

Fresh windows installation/
Vc redist installed.

kk6mrp, 2021-06-05:
I think I might have seen that error related to OpenGL versions, can you try the executable I attached from the link below?

https://github.com/solvespace/solvespace/issues/1036#issuecomment-839428790

phkahler, 2021-06-09:
@aelio  any update on this? Did you try the OpenGL 1 version linked above?

aelio, 2021-06-09:
> @aelio any update on this? Did you try the OpenGL 1 version linked above?

Hi, yet had no chance. I guess it could work, there is Intel 630 CPU graphics, I'll let you know ASAP

ruevs, 2021-08-08:
I created an unofficial Windows XP release with OpenGL 1.
https://github.com/ruevs/solvespace/releases/tag/WindowsXPPost3.0

ruevs, 2022-01-22:
Here is an updated unofficial Windows XP release with OpenGL 1.
https://github.com/ruevs/solvespace/releases/tag/WindowsXPPost3.0_2

ruevs, 2022-08-08:
@aelio do you still have this problem?

ruevs, 2023-03-08:
Closing since there is no reaction from the original reporter.