solvespace/solvespace #1055: Compress slvs files

kekscode, 2021-06-07:
### System information

- **SolveSpace version:** 3.0˜0e0b0252 
- **Operating system:** macOS 11.4 

### Expected behavior

A lot of plain text file formats are nowadays compressed when written on disk. Modern
machines are very good at (de)compression and `.slvs` files benefit a lot from that. I am
aware about the "storage is cheap" mantra, but `slvs` files can get quite big and cloud
syncing and regular backups among computers over LAN/WAN/Internet is a thing.

Compression could be an option when saving for the first time and/or a setting in the configuration.

Compressed files should have another extension like `.slvsz` in order show that the file has been
saved using lossless compression.

### Actual behavior

A simple model with less than 6 sketches and less than 6 extrudes is over 100Kb on disk.
With a simple `gzip [filename]` this goes way down (quick test: 135Kb => 10Kb).

### Additional information

This relates to #75, because having a modern format like protobuf would benefit
file size anyways.

phkahler, 2021-06-09:
This really falls under the umbrella of new file format. That's not going to happen for a while.

Evil-Spirit, 2021-06-24:
Just need to invent a new format, but support of loading of slvs files.

Evil-Spirit, 2026-02-27:
> A simple model with less than 6 sketches and less than 6 extrudes is over 100Kb on disk.
> With a simple `gzip [filename]` this goes way down (quick test: 135Kb => 10Kb).

Just get rid of storing of remapped id or somethings like this. SolveSpace  should having stable-id generation (persistant id), based on additional structures, so it should not store all the remapping, just generate it on fly, so we could save a lot of data. See NoteCAD, it having partially implemented this feature, but in reality a lot of job should be done.

wolfpackmars2, 2026-03-02:
> named parameters working correctly soon

This would be totally bad-a--.  Really look forward to named parameters.

> 1. The text based format is _very_ useful for keeping the .slvs files under version control systems.

Yes I love text files too for their flexibility. If someone is inclined, for example, it's relatively trivial to write scripts or utilities to work on text files. 

Using a widely known compression method is more of a means to an end. Text files are great but they don't work well with embedded binary data. Using something like zip/etc. offers the flexibility of a well-known file standard and allows working with a mix of data types. But I'm not trying to argue for or against compression. Compression has downsides too, including CPU and memory overhead, on top of the other points already made.

The save file format has not changed during the life of the project. In the beginning, it looks like some thought was given to try to provide a unique identifier to .slvs files, but it has never been used and IMO is a potential for problems more than any utility it might provide.

Here's my wishlist for improved save files.

Firstly,

> we can (optionally?) eliminate the shell and triangle mesh from save files.

Yes - I think this can be done right away without breaking any backwards compatibility.  Basically, the data keys that we know are ignored can be totally removed from the save files. They are:

- The File Signature line `"±²³SolveSpaceREVa\n\n\n"`
- Lines exactly equal to "AddEntity\n", "AddSurface\n", or "AddCurve\n"
- Lines beginning with "Triangle ", "Surface ", "SCtrl ", "TrimBy ", "Curve ", "CCtrl ", or "CurvePt ". Anything between these line beginnings and newline is ignored.

Beyond this, most changes would break backwards compatibility once implemented. In that case, we should use the opportunity to improve the format from that point.

Improvements:
- Remove keywords that will just be ignored anyways.
- Implement comments. These provide utility beyond just human readable comments - for example, a comment line that is ignored by an older build could still contain information that can be used in newer builds. I don't have any good examples off the top of my head, but the key is future flexibility while minimizing maintenance and complexity. Comments should be ignored functionally, but they should be preserved between save files. There are many options to implement comments, from using the common "#" to using single and multi-line comments like // and /* ... */ 
- ASAP implement better handling of unexpected information in the save file. The current go-to is to force a crash when encountering an unexpected line in the save file. At least the user will be informed as to which line number caused the crash, but it's overall not very useful. Can we just abort instead of crashing? Can we provide better feedback? Can we distinguish between levels of errors? Categories of errors such as "unexpected data, but doesn't affect anything and can be largely ignored, unexpected results that may affect the final result and the user should be warned but we can try to ignore and continue processing the file, and errors that should abort the process and return SS to a previous state (or a blank slate at least).
- Also ASAP settle on an encoding standard, probably UTF-8, and get rid of the unique file signature. Or implement a signature that is compatible with UTF-8. As it is now, the current signature would cause a crash if, for example, the file were encoded as UTF-8.
- If a magic number/file signature is desired, let it actually be used.
- Start tracking format revisions. Probably a counter. E.g. SolveSpaceFormat=0. We can easily implement a check to prevent older versions from opening files of a higher format than expected and causing a crash.
- The format revision should be incremented to signal changes that are not compatible with previous builds. Where possible, new or unknown data keys should be preserved but ignored where possible. As an example, a key "Description" which might allow the author to add some text to help them identify the sketch contents. Other metadata which may be added in the future which doesn't affect the sketch but is useful to future builds.

The focus is to make maintenance easier, not harder. The simple act of adding a comment line would allow us to, for example, add the version line of the original SS build as a comment.

```
// Savefile generated by SolveSpace 3.2~abcd0123
```

It would also be nice to incorporate linked files as a single save file, but this will be more complicated. It should be a goal to strive towards eventually. Maybe distinguish between sketch files and assembly files if needed. Future versions could eventually allow to edit/modify/update/sync of embedded/linked sketches.

wolfpackmars2, 2026-03-13:
> [@phkahler](https://github.com/phkahler) [@wolfpackmars2](https://github.com/wolfpackmars2) why to write so many worlds here. instead, write all your wishes to copilot, I wonder why you haven't done it yet.

I'm getting there. I will get there eventually. When I have more confidence that the main team is comfortable with me, I won't feel the need to explain myself as much - hopefully.

Copilot can't tell me how the devs will react to my proposals or guarantee they won't dismiss my ideas/proposals out of hand.

I've done a lot of work reviewing the save files. I don't want to waste my and their time working on something they don't want, or maybe they have current plans that I don't want to interfere with which aren't presently known.

I should be wrapping up the AppImage stuff tonight, and I hope the appimages will be helpful to the core developers as well. I want to focus on the savefiles next, but only if the team wants my help, and only if I am moving in a direction they are comfortable with.

Evil-Spirit, 2026-03-14:
There are fundamental limitations in the SolveSpace basement. At this point isn't it better to redesign it from scratch? While Copilot is unlimited and free, you can just make you own SolveSpace in a week (if you have enough time on endless codereview and testing)

ruevs, 2026-10-31:
> 13\. There are fundamental limitations in the SolveSpace basement. At this point isn't it better to redesign it from scratch? While Copilot is unlimited and free, you can just make you own SolveSpace in a week (if you have enough time on endless codereview and testing)

It's called [BearCAD](https://github.com/iffy/BearCAD) :-) And a few others "these days"... finding them and evaluating them is left as an exercise for the reader ;-)