solvespace/solvespace #1058: SolveSpace crashes when moving a sphere into a

kekscode, 2021-06-11:
### System information

- **SolveSpace version:** Version 3.0 (3.0~0e0b0252) 
- **Operating system:** MacOS 11.4 (20F71) 

### Expected behavior

Reproduce by opening the attached slvs file and move the sphere from the top
right (by clicking+holding its mid point with the mouse cursor) and drag the
selected sphere into the X-shaped body.

This should work without crashing and show solver issues visually.

### Actual behavior

SolveSpace crashes.

Sometimes SolveSpace crashes very fast, sometimes the outlines are highlighted
red and it crashes shortly after this visualization happened.

### Additional information

I attached:

- The .slvs file
- Stderr output from commandline as text file (program start until crash while doing exactly the one action which causes the crash)

[model_and_stderr_output.zip](https://github.com/solvespace/solvespace/files/6635484/model_and_stderr_output.zip)

ruevs, 2021-06-11:
It is interesting that if I fix the sphere not to be "double walled" - in other words if it is made by a lathed half circle instead of a whole one then it does not crash. See the attached file.

It makes sense - the "distance" between the two "walls" of the sphere is 0 (zero) by definition, so any edge that is being intersected with the sphere will be cut into **three** segments - one outside of the sphere, one inside, and a zero length one "between" the two shells that are coincident.

But we should not crash anyway.

It should be possible to make a much simpler reproduction case for this.
[exoskelett-base_good_sphere.zip](https://github.com/solvespace/solvespace/files/6636976/exoskelett-base_good_sphere.zip)

ruevs, 2021-06-11:
Sure enough here is a much simpler model to reproduce the bug:
[DoubleWalledSphereZeroLengthEdgeInstersectionCrash.zip](https://github.com/solvespace/solvespace/files/6637019/DoubleWalledSphereZeroLengthEdgeInstersectionCrash.zip)