solvespace/solvespace #1059: Crash when creating lathe group and extrusion from triangle

nowakf, 2021-06-17:
### System information
- **SolveSpace version:** 
3.0~0e0b0252
- **Operating system:**
Arch Linux
### Expected behavior
1. You create the triangular section.
2. You lathe it around its tip.
3. You select the sketch again, and extrude it.
4. You save it / select the extrusion, etc

### Actual behavior
It crashes at 4 with: 

File ./src/srf/surfinter.cpp, line 103, function AddExactIntersectionCurve:
Assertion failed: !(sb->Start()).Equals(sb->Finish()).

### Additional information
Unfortunately, it crashes if you try to save it with the problematic geometry. If you extrude the first sketch though, it should crash upon the next thing you do (saving, etc).
[crash.zip](https://github.com/solvespace/solvespace/files/6667857/crash.zip)

ruevs, 2021-06-17:
Most.probably a duplicate to this https://github.com/solvespace/solvespace/issues/1058
I'll confirm later.

ruevs, 2022-08-27:
Definitely a duplicate to #1058 but I'll keep this open since it is a different way to reproduce it.