solvespace/solvespace #1060: Crashes with error LSOpenURLsWithRole() failed with error -10810 for the file

littlebritaly, 2021-06-23:
### System information

Solvespace 3 on macOS Sierra 10.12.6

I downloaded the latest version from the website (3.0 stable) and as soon as I try to open SolveSpace.app, i get a "SolvesSpace quit unexpectedly" dialog.

I tried from the command line (open SolveSpace.app), what i then get is "LSOpenURLsWithRole() failed with error -10810 for the file"

even after removing the quarantine (xattr -d com.apple.quarantine SolveSpace.app/) the app wont launch.

Any ideas anyone ?

phkahler, 2021-06-23:
@vespakoen  haven't heard from you is a while, any ideas on this mac issue?

vespakoen, 2021-06-24:
The error you got was from the "open" command not finding SolveSpace.app, the correct way to use the "open" command with an app name is this:

```
open -a SolveSpace
```

However, that won't show a log, if you want to see a log, open SolveSpace like this (from the Terminal.app):

```
/Applications/SolveSpace.app/Contents/MacOS/SolveSpace
```

Hopefully it shows what the problem is, if not, you could try installing XCode and compiling SolveSpace yourself:

```shell
# get solvespace source code
git clone https://github.com/solvespace/solvespace.git
cd solvespace
git submodule update --init
# install homebrew (package manager)
/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)"
# install openmp
brew install libomp
# compile solvespace
.github/scripts/build-macos.sh debug x86_64 xcode
# it should create and open a xcode project, now you can click the "Run / Play" button and it should jump into the debugger in case it crashes.
```

littlebritaly, 2021-06-25:
Thank you for the help, I dont get a log, just this one line 

LSOpenURLsWithRole() failed for the application /Applications/SolveSpace.app/Contents/MacOS/SolveSpace with error -10810.

I'll try downloading and compiling and will report back

littlebritaly, 2021-06-25:
Ignore the above, i found "SolveSpace.app" in the solvespace/build/bin/Debug" folder, copied to Applications folder and it launches instantly, thank you so much.

vespakoen, 2021-06-26:
Glad that worked!

We probably dropped support for your macos version in our build process a while ago, but im glad that manual compilation still seems to work.
I cannot test on that old of a macos version  so I cannot help more than I already did i'm afraid.

closed 2021-06-26 by vespakoen

ruevs, 2021-06-26:
> We probably dropped support for your macos version in our build process a while ago...
> I cannot test on that old of a macos version...

Well the _newest_ Mac OS version that I have is  a 32 bit [Snow Leopard](https://en.wikipedia.org/wiki/Mac_OS_X_Snow_Leopard) on a MacBook Pro from 2008 - with a similar vintage XCode :-D  I'm not using it :-)

Now there is a system that will be mighty hard to support - C++ 11 did not exist when that XCode version was released :-D

Maybe someday (when I retire?) maybe I should try compiling SolveSpece on that :-D

Something tells me it will have to start wit porting clang :-D

Or even better - a Power PC based Mac? :-D

ruevs, 2021-06-26:
> macOS Sierra
> Initial release | September 20, 2016; 4 years ago
> Latest release | 10.12.6 (16G2136)[2] / September 26, 2019; 20 months ago

Yep, that's "a bit old" in Apples' twisted "forced obsolescence" world :-D