solvespace/solvespace #1061: Symmetry can't be about a vector? Why does this line segment act like a vector?

rpavlik, 2021-06-24:
### System information

- **SolveSpace version:**  only flatpak changes on top of 00533a0fb22c85293d253f998c8d6f30029061ea
- **Operating system:** debian buster

### Expected behavior

I'm trying to mirror the location of two screw holes across an object. I click the two points I want symmetric, then I click a line to mirror across. I should be able to add the symmetry constraint and have it work.

### Actual behavior

When I have two points plus a line selected, the property browser calls the line a vector, even though it has endpoints.

[image]

Thus, when I press "Y" to make the symmetry, I get this error:
[image]

### Additional information

[two-side-for-coated-tape2 (copy).zip](https://github.com/solvespace/solvespace/files/6710374/two-side-for-coated-tape2.copy.zip)

I was in the "g010-translate" group when I tried to do this.

phkahler, 2021-06-25:
@rpavlik  What would symmetry about a line mean in 3D? For a point I could see it being put on the opposite side of the line, but it looks like that would get the position wrong in this case.  I had no idea this feature exists but symmetry about a line works fine in a 2d sketch group.

On a related note, my hope is that a mirror groups can be used for this. You'd select 3 points to define a "mirror plane" and the entire hole group here would reflect across it. Need to work on that feature some more maybe this weekend.

rpavlik, 2021-06-26:
Yeah you're right, it would get things wrong in this case. I think I was thrown off by the fact the line segment started being called a vector once I also had two points selected.