solvespace/solvespace #1066: Runtime error 'Length / Arc Ra&tio' missing translation

josedgm, 2021-07-11:
### System information

- **SolveSpace version:** edge
- **Operating system:** ubuntu 20.04

### Expected behavior

After compile, the program must open.

### Actual behavior

It just gives this error:
Missing (absent) translation for 'Length / Arc Ra&tio'
Missing (absent) translation for 'Length / Arc Diff&erence'
connect failed: No such file or directory
Killed

### Additional information

The 3.0.0 stable version 0e0b0252 runs perfectly fine when installed using the DEB from relaseses.

phkahler, 2021-07-11:
Perhaps we should put that back to the way it was. I think Lenght Ratio and Length Difference are clear even when applied to an Arc. Did the old text include the word "Line"?

The translation issue should not be causing the killed process though.

Symbian9, 2021-07-11:
I use @AlexanderP's PPA builds, and latest *SolveSpace* `3.0~` https://github.com/solvespace/solvespace/commit/ddb76324afb99376c1bf5cccec2db7cb06a023dc works well for me.

@josedgm, may you try its DEB on your Ubuntu 20.04?

- https://notesalexp.org/packages/en/focal/amd64/solvespace/
  - https://notesalexp.org/packages/en/focal/amd64/solvespace/download.html

ruevs, 2022-11-06:
@josedgm your problem was pretty long ago. Do you still have it or can we close this issue?

josedgm, 2022-11-19:
I'm sorry. 

Yes. Close it, please.

closed 2022-11-19 by josedgm