solvespace/solvespace #1067: Unable to open/save a file

eaglefred, 2021-07-14:
### System information

- **SolveSpace version:** 3.0~0e0b0252
- **Operating system:** Xubuntu 

### Expected behavior

Opening a dialog box to choose a file

### Actual behavior

Nothing

### Additional information

Error reported in terminal:

(solvespace:6053): GLib-GIO-WARNING **: 06:20:50.978: Error creating IO channel for /proc/self/mountinfo: Permission denied (g-file-error-quark, 2)
Missing (empty) translation for file-type'SolveSpace models'

(solvespace:6053): Gtk-WARNING **: 06:20:51.050: Can't open portal file chooser: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.portal.Desktop was not provided by any .service files

ruevs, 2021-12-20:
@eaglefred is this still a problem? @ppd at the time you assigned it to yourself - do you have an idea?

ppd, 2021-12-20:
Ah, I forgot, sorry. This looks very much like a missing desktop portal on the host. That's not really a supported configuration for flatpaks or snaps. GTK would have to fall back to its own file chooser in this case, but it doesn't do this by default.

A workaround for users would be to set `GTK_USE_PORTAL=0`. That should get rid of the portal requirement. Of course the paths will look strange and bookmarks might be missing etc because that file chooser will only run in the confinement of the snap/flatpak.

Symbian9, 2021-12-22:
> Ah, I forgot, sorry. This looks very much like a missing desktop portal on the host. That's not really a supported configuration for flatpaks or snaps.

@ppd, as alternative, to avoid of Flatpak and Snap issues, try SolveSpace DEB builds from @AlexanderP's PPA:

- https://notesalexp.org/packages/en/focal/amd64/solvespace
- https://launchpad.net/~alex-p/+archive/ubuntu/solvespace-daily

closed 2023-12-31 by phkahler