solvespace/solvespace #1068: Link DXF and DWG files

Symbian9, 2021-07-19:
### System information

- **SolveSpace version:** `3.0~master`
- **Operating system:** All supported

### Expected behavior

Add feature suport for linking AutoCAD DXF and DWG files via *"Link / Assemble..."* (as 2D/3D wireframe).

In such way all entities from DXF / DWG would be inserted as fully constrained ("read-only" enteties), which allows.

There should be one dot with a `3` normals (for X, Y & Z) according each linked DXF / DWG origin (`0;0` for 2D & `0;0;0` for 3D).

### Actual behavior

Actually there are no way to link DXF and DWG files  — instead the only way is to import its 2D/3D wireframe (as "editable" enteties) directly into SolveSpace *as single group*, which in case of DXF / DWG often produces "too many unknowns" due to a lot of unconstrained entities imported with a lot of DOFs.

### Additional information

DXF Title Blocks / Sheets:

- https://github.com/qcad/qcad/tree/master/libraries/templates/metric
- https://github.com/LibreCAD/LibreCAD/tree/master/librecad/support/library/sheets

phkahler, 2021-07-25:
As a workaround you can open them and immediately save the drawing as a .slvs file which can be linked.

I think this should be looked at after the major DXF DWG work is completed.

Symbian9, 2021-07-26:
> As a workaround you can open them and immediately save the drawing as a .slvs file which can be linked

In most cases if *"Import > DXF"* with more that just few entities *SolveSpace* immediately "red screened"  with `To many unknowns in a single group!`