solvespace/solvespace #1075: 🔮 How burdensome are Windows and macOS builds to maintain? (+ suggestion)

akauppi, 2021-07-25:
I'm only tangentially (pun) following the SolveSpace project, but installed it today.

As I'm seeing it, the ecosystem for this kind of tools is changing. The introduction of [WSLg](https://devblogs.microsoft.com/commandline/the-initial-preview-of-gui-app-support-is-now-available-for-the-windows-subsystem-for-linux-2/) makes it possible to run Linux GUI programs on updated Windows 10, by the end of this year.

Would this be something that SolveSpace could embrace?

**Benefits**

All of the `README` is dedicated to how to build the source in multiple systems. While this is admirable, it must take away time from feature development.

---

For Mac, I would suggest following the WSLg's lead and making sure people can run Linux applications, also graphical ones, in the desktop. This means installing Docker and XQuartz. No big deal, and one would be on-par with the Windows experience in late 2021, once WLSg is out.

An added benefit (for both these approaches) is that SolveSpace runs sandboxed, not in the main host machine. For a niche engineering application I do find this comforting.

So... what say you?

phkahler, 2021-07-25:
>> How burdensome are Windows and macOS builds to maintain?

Not that bad? Getting the CI setup took a lot of work by some great people, but now it mostly just works although macOS builds are currently failing (@vespakoen or @ppd  any idea why?)

If someone wants to run the linux version on another OS they are certainly welcome to do that. If the maintenance burden were to get too high it might be worth considering, but so far it's OK and we get to run native on all platforms which is kind of nice - windows saves the configuration parameters in the registry for example. The Windows version also has no installer, just run it from wherever which is as easy as it could possibly be.

ruevs, 2021-08-08:
Requiring WSL on Windows is downright crazy. SolveSpace works on Win 7 and later out of the box and can still be built for XP.

And you want to force people to have the latest Windows 10 and download and install gigabytes of a full extra OS (the WSL) just to to run a program, which natively built is a 6MB single EXE?!

And maintaining the readme is close to 0% of the effort.

closed 2021-08-08 by ruevs