solvespace/solvespace #1076: Building from sources on Debian `buster-slim`

akauppi, 2021-07-25:
### System information

- **SolveSpace version:** git `master`, currently `a97b77c1e552c89aee873562694e7ac733056094`
- **Operating system:** Debian `buster-slim`

### Expected behavior

I followed the Linux Debian/Ubuntu instructions for [Building for Linux](https://github.com/solvespace/solvespace#building-for-linux) and found some deviations from the instructions.

This is one thing I'd like to report.

The other is that after building, trying to launch says:

```
$ build/bin/solvespace
Sorry, only UTF-8 locales are supported.
```

I have all `LC_ALL` etc. set correctly (I think); `locale` is installed and so forth. 

"Expected behavior":

- I can compile the git `master`, for latest stable Debian, according to the instructions on the main `README`.

### Actual behavior

- I needed to tweak compilation settings, adding one package (`pkg-config`) 
- some packages listed as requirements were not missed by the compilation (`libfontconfig1-dev`, `libglu-dev`, `libspnav-dev`). I am fine adding them; maybe they came up indirectly through some others.

### Additional information

https://github.com/akauppi/docker-solvespace

- has the Dockerfile

I am aware of the following discussions:

- [solvespace 3..0 on Debian Linux](https://solvespace.com/forum.pl?action=viewthread&parent=3764) (SolveSpace Forum, 2021)
- [Sorry, only UTF-8 locales are supported](https://solvespace.com/forum.pl?action=viewthread&parent=1649) (SolveSpace Forum, 2016)

akauppi, 2021-07-26:
I ended up being stuck with "Unable to create GL context" - which can be a Docker / macOS thing. https://github.com/akauppi/docker-solvespace/issues/1

Symbian9, 2021-07-27:
@AlexanderP, may you get us your DEB-build scripts?

- https://notesalexp.org/packages/en/buster/amd64/solvespace/

phkahler, 2021-07-31:
For the packages that are not missed, are they required to do a debug build? They are -dev packages.

Symbian9, 2021-07-31:
FTR, It would good to make it usable via GitHub Actions, for example:

- https://github.com/nylas/build-dpkg-buster

ruevs, 2022-01-24:
@akauppi [pkg-config](https://packages.debian.org/sid/pkg-config) sounds like a pretty "basic" package needed for development - maybe that is why it was not noticed? Everyone already had it?

And perhaps your "Debian `buster slim`" did not include it - which is untypical?

Someone more familiar with Linux distributions will have to decide whether we should add it to the README.

Symbian9, 2022-01-24:
> Someone more familiar with Linux distributions will have to decide whether we should add it to the README.

@AlexanderP, may I ask You for help here?

kk6mrp, 2022-01-25:
> I needed to add one package: pkg-config

It appears like pkg-config support has been required ever since the GTK port was added here: https://github.com/solvespace/solvespace/commit/2c39f259db0c6b6f1eeef7fa95125311daadc991 and my best guess is that the distribution normally pulls it in with the `build-essential` package or something like that. I couldn't find any other information on it but there is pkg-config and pkgconf, pkgconf appears to be a replacement for pkg-config. I probably wouldn't add it as a dependency for that reason and cmake does a good job of telling you when a package is missing from your system as well as what it is.

> two packages listed as requirements were not missed by the compilation: libfontconfig1-dev, libglu-dev. Are these necessary? Maybe they came up indirectly through some others.

If you noticed under the https://github.com/solvespace/solvespace#building-for-linux link, it states:

> To build the GUI, you will need **fontconfig**, gtkmm 3.0 (version 3.16 or later), pangomm 1.4, OpenGL and **OpenGL GLU**, and optionally, the Space Navigator client library. On a Debian derivative (e.g. Ubuntu) these can be installed with:

AlexanderP, 2022-01-25:
libfontconfig1-dev and pkg-config are needed, cmake checks them, and installed as dependencies of `libpangomm-1.4-dev`
```
libpangomm-1.4-dev -> libpango1.0-dev ->  libfontconfig1-dev
libpangomm-1.4-dev -> libpango1.0-dev -> pkg-config 
```

ruevs, 2022-01-27:
So shall we close this without adding `pkg-config` to the build instructions or should we add it?

Symbian9, 2022-02-04:
Could be there created `build-deb.sh` script for easily compile & build DEB package locally or via CI/GitHub actions?

- https://github.com/jtdor/build-deb-action
- https://github.com/zlig/deb-builder-packagecloud

closed 2022-02-10 by ruevs

ruevs, 2022-02-11:
https://tracker.debian.org/pkg/solvespace