solvespace/solvespace #1079: Show XYZ normals for linked STL and EMN origin points

Symbian9, 2021-07-31:
### System information

- **SolveSpace version:** `3.0~` https://github.com/solvespace/solvespace/commit/f6bb0a2d35ba6c67a72170413207d0f28bac216d
- **Operating system:** Debian 10.x buster

### Expected behavior

Linked STL and EMN objects *origin point* should be represented with three *XYZ normals* shown for easy rotating, same way as for linked SLVS objects (if linked SLVS file saved with shown `[v] g001-#references`).

### Actual behavior

Linked STL object (mesh) has NO normals at its *origin point*.

In same time linked EMN object (solid model) has a *single normal* at *origin point*.

### Additional information

There should be a way to show in SolveSpace *origin points* with turned OFF visibility of all other point types.

phkahler, 2021-08-08:
Awe, c'mon I gave you a bounding box for STLs !   ;-)

But seriously, I don't know how the "same orientation" constraint even works. I thought a normal was something other than it turned out to be. There is one (one of 3) on linked IDFs. Might need to add more, and for STL.

ruevs, 2021-08-08:
> But seriously, I don't know how the "same orientation" constraint even works.

Ehh, I don't know how it works in the code either, but (as you probably know) it is extremely powerful for assemblies since it constrains all three rotational degrees of freedom of two objects relative to each other by just roughly orienting them, selecting two normals and pressing "X".

phkahler, 2021-12-11:
OK @Symbian9 and @ruevs  an IDF has one normal shown, does that fully behave the way a normal is supposed to? If so I'll make 3 of them for linked STL and maybe add two more for IDFs.

Symbian9, 2021-12-11:
> does that fully behave the way a normal is supposed to?

Probably. I guess there also should be origin point at XYZ normals crossing point for each linked/assembled STL/IDF/etc. parts after import (but not for linked/assembled SLVS parts)

NOTE: In SLVS its possible to decide show or not XYZ and origin point for each SLVS part by toggling `g001-#refernces` in  each SLVS part before saving — some time it may be really useful for assembly creating, so I may just recommend to keep such behaviour for linked/assembled SLVS parts only.

ruevs, 2021-12-12:
> OK @Symbian9 and @ruevs an IDF has one normal shown, does that fully behave the way a normal is supposed to?

@phkahler  As far as I can tell the lone normal of a linked IDF works just like any other. Gets hidden when you "Hide normals". `X` works as expected. So  if you add normals to the STL that wold be great.

In the meantime a solution for constraining and STL is to add a construction line(s) between two "convenient" point(s) of the STL and use that to constrain. It is not as convenient as a normal since `X` (same orientation) is not possible but anything can be achieved with the other constraints.

phkahler, 2022-01-20:
@Symbian9  I'm going to merge a change that implements the normals for STL files. If you've got sketches with constraints on an STL I expect there to be problems (and I don't care because I warned about that when I added STL linking). But see how this works for you. It will close this issue on merge.

Thanks!

closed 2022-01-20 by phkahler

Symbian9, 2022-01-20:
> But see how this works for you. It will close this issue on merge.

Thanks, @phkahler!

I will checkout for fresh DEB packge from @AlexanderP and then will test it & let you know my new experience ;)

Symbian9, 2022-01-24:
> But see how this works for you.

UPD: Just checked — it works as expected now. Thanks! ;-)

[image]