solvespace/solvespace #1082: Add "step length" option for "Step Translate" groups

Symbian9, 2021-08-07:
### System information

- **SolveSpace version:** `3.0~` https://github.com/solvespace/solvespace/commit/1b8e1dec652eef7916a94b6c9db404457dacb0db
- **Operating system:** Debian 10.x buster

### Expected behavior

"Step Translate" group should has `step length` (in some way similar to `pitch` in Helix groups) for continuous copies drawing with constant step length:

```
`step length - distance between copies origins`
{step_length_value}
[ ] fixed
```

While distance between origins of the *1st-copy* and the *N-copy* is `>= N × step_length_value` & `< (N+1) × step_length_value` draw only `N` copies.

### Actual behavior

Actually there are no way for drawing copies with a constant step length alongside axis between the *1st* and *last* copies, according distance between origins of the *1st* and *last* copies.

### Additional information

phkahler, 2021-08-08:
You can do a distance constraint between first and second copy. Or first and last copy.

ruevs, 2021-08-08:
And one can constrain points on (any of the) copies using the vertical, horizontal, on line, midpoint, symmetric etc. constraints.

The requested feature is completely superfluous.

ruevs, 2021-08-08:
> While distance between origins of the _1st-copy_ and the _N-copy_ is `>= N × step_length_value` & `< (N+1) × step_length_value` draw only `N` copies.

@Symbian9 and otherwise? Draw more than N copies? I find it hard to imagine the use cases in a CAD program where one does not know exactly how many copies are needed.

Symbian9, 2021-08-08:
>  I find it hard to imagine the use cases in a CAD program where one does not know exactly how many copies are needed.

Lets talk about parametric: I has a template of long bar (which may warry in size) and want duplicate some objects along this bar depending on bar length (i.e. two points from this bar would be used to constraint 'total length' of "Step Translate" group) with a constant step length - CAD program already know total distance between two points which should be used for Step Translate.

> You can do a distance constraint between first and second copy. Or first and last copy.

Yeah, but it could not be automatically changed if distance between origins of `1` and `LAST` copies of "Step Translate" is constrained to variable from previous groups.

So, each time I change some variable lenght in previous group, which used for constraint distance between origins of `1` and `LAST`, the number of copies should be changed manually - this feature request about adding some way to automatically calculate number of copies with constant step length depending on 'total length' of "Step Tranlate" group.

May be there should be an additional construction line added in "Step Translate" group where one end is constrained to origin of `1` copy, and origin of `LAST` copy should by just constrained as `point-on-line`.

ruevs, 2021-08-08:
OK - I understand. This is a valid use case. Typical example - the mounting holes along a linear rail.

Essentially it would be a "variable N (copies) translate" with set distance between them and the ends (length) constrained to some entities on a (parametric length) object or just a distance.

I'm not sure this can be implemented as part of the current step stranslating group while keeping backward compatibility - most probably not.

This would probably require a new group type.

phkahler, 2021-08-08:
I've run across this type of thing before with step rotating where I want the number of copies and an angle to be related. IMHO this is worth keeping in mind while noodling on bigger changes.

ruevs, 2021-08-10:
Definitely.

A possible solution in a future where [parametric sketches](https://github.com/solvespace/solvespace/issues/77) are implemented is to literally do `NumberOfCopies=TotalLength/CopyDistance`.

ruevs, 2024-08-18:
And parametric sketches now have a draft implementation where this is totally possible. https://github.com/solvespace/solvespace/issues/77#issuecomment-2191629118