solvespace/solvespace #1083: Distance constraint between origins of 1st and 2nd copies removed after decreasing Step Tramslate to 2 copies

Symbian9, 2021-08-07:
### System information

- **SolveSpace version:** `3.0~` https://github.com/solvespace/solvespace/commit/1b8e1dec652eef7916a94b6c9db404457dacb0db
- **Operating system:** Debian 10.x buster

### Expected behavior

If in "Step Translate" group with `3` or more copies there is ***non-reference*** *"Distance constraint between origins of 1st & 2nd copies"* and number of copies then decreased to `2` — convert this constraint into *"Distance between origins of 1st and last copies"*.

### Actual behavior

Actually by default new "Step Translate" group creates with `3` copies.

If "Step Translate" group has `3` *or more* copies it may has few constraints:

1. Distance between origins of two consecutive copies (1st & 2nd, 4th & 5th, etc.);
2. Distance between origins of the 1st and last copies;
3. Distance between origins from various copies;
4. Distance between any entities from various copies.

But if decrease number of copies in such group then:

- if decrease to `2` copies: any constraints would be removed (except "Distance between origins of the 1st and last copies", which would be preserved);
- if decrease to `1` copy: all constraints would removed.

### Additional information

Feature Request: *Add "step length" option for "Step Translate" groups* (alternative solution to this issue)

- https://github.com/solvespace/solvespace/issues/1082

phkahler, 2021-08-08:
I'll give it some thought but I dont think this is fixable without braking things. The copies are internally numbered 1..n-1 and LAST. That's because constraining against the last one is common, and it allows changing the number of copies without braking those constraints.

When you reduce the number to 2, they get numbered 1 and LAST. Any constraint on 2 is lost.

This was IMHO a really good way to handle it, but it's not perfect.

Symbian9, 2021-08-08:
> When you reduce the number to 2, they get numbered 1 and LAST. Any constraint on 2 is lost.

So, in this issue I just propose: if there is constraint between origins of `1` and `n-1` in a group with 3+ copies, after decrease to 2 copies this constraint should be automatically replaced with a constraint between origins of `1` and `LAST` copies.

phkahler, 2021-08-08:
Sorry, I'm not going to accept changes that try that hard to figure out what the user wants to do in special cases. If I think of another general solution I'll post it as an enhancement suggestion.

closed 2021-08-08 by phkahler