solvespace/solvespace #1084: SolveSpace runs out of RAM and swap on linking STL files

Symbian9, 2021-08-07:
### System information

- **SolveSpace version:** `3.0~`
- **Operating system:** Debian 10.x buster

### Expected behavior

SolveSpace should detect any potential issue on linking 'dangerous' STL files to prevent overload RAM.

### Actual behavior

SolveSpace fully blows my RAM with linking next STL files:

- `break_openscad.stl` - https://forum.freecadweb.org/viewtopic.php?t=60912&p=522689#p522689
- `Suzanne.stl` - https://en.wikipedia.org/wiki/File:Suzanne.stl
- `Utah_teapot_(solid).stl` - https://commons.wikimedia.org/wiki/File:Utah_teapot_(solid).stl

I just kill SolveSpace when it stopped to be responsible and its process loaded RAM at 100% and started to overload swap.

### Additional information

In same time here are STL files of Blender's default "Cube" (`cube.stl`) & "Suzanne" (`monkey.stl`) mesh objects, which I exported from Blender 2.49b myself and *which are linking successfully* into SolveSpace:
 
- https://twitter.com/i/status/1421386197100793858

ruevs, 2021-08-08:
This is true - there are no provisions in the code to restrict RAM usage. Probably due to the fact that it would become unusably slow, when trying to construct complex objects in the "normal" way, long before RAM usage became excessive.

Linking STL is the first time we have been able to "trick" SolveSpace into importing something "too complex".

Symbian9, 2021-08-08:
>  Probably due to the fact that it would become unusably slow, when trying to construct complex objects in the "normal" way, long before RAM usage became excessive.

Issue here is not about its complexity, but about 

Just compare next files:

- `break_openscad.stl` (blows out RAM) VS `cube.stl` (not issued RAM)
- `Suzanne.stl` (blows out RAM) VS `monkey.stl` (not issued RAM)

Maybe there are some digit precision or big distance values in those STL files which triggers SolveSpace to blows RAM? Or maybe there some difference in format structure, characters, etc (in comparison to `cube.stl` and `monkey.stl`).

phkahler, 2021-08-08:
I can open that monkey file too, but the suzanne fails for me with increasing memory usage and general unresponsiveness - I didn't wait for it to crash or anything like that.

I've got no tools for analyzing an STL file to figure out what might be wrong. Maybe adding some debug output to the STL linker.... @ruevs you're a 3d printing kind of guy, how do you like the STL linking? Any ideas why it might choke on a ~30k file?

Symbian9, 2021-08-08:
>  Any ideas why it might choke on a ~30k file?

FTR, SolveSpace tottally freeze on my PC with `break_openscad.stl`, which has only 39 verticies / 13 triangles (but works fine with a `cube.stl`, which has 36 verticies / 12 triangles)

phkahler, 2021-08-10:
I'll have to verify, but the cube with 13 verticies is probably not valid in some sense. It has edges that are not shared 1-to-1 by two triangles. It's not clear why that's causing such huge problems - it may be triggering issue #320 which would be interesting. I'll confirm later that its not the mesh reader (very unlikely).

It is interesting that CGAL (used by openscad) can't handle it properly either.

ruevs, 2021-08-10:
I'll try to debug it as well - when I get some spare time.

ruevs, 2021-08-11:
Well when importing "break_openscad.stl" we are  importing 538976288 (0x20202020) triangles on line 123 of importmesh.cpp

[image]

This is certain to cause problems... 

Now on to the hex editor to see why that `f.read((char*)&n, 4);` reads 0x20202020 and to learn what the STL format looks like :-)

ruevs, 2021-08-11:
Well `break_openscad.stl` is not an STL file it is a text file containing OpenSCad source!

```
solid OpenSCAD_Model
  facet normal 0 0 -1
    outer loop
      vertex -1 -1 -1
      vertex 1 1 -1
      vertex 1 -1 -1
    endloop
  endfacet
  facet normal -0 0 -1
    outer loop
      vertex 1 1 -1
      vertex -1 -1 -1
      vertex -1 1 -1
    endloop
```

No wonder stupid things happen. Interesting whether there is a way to validate that a file (at least looks like) STL format...

ruevs, 2021-08-11:
:-D :-D
Binary: https://www.loc.gov/preservation/digital/formats/fdd/fdd000504.shtml#specs
Text: https://www.loc.gov/preservation/digital/formats/fdd/fdd000506.shtml#specs
```
Format specifications Explanation of format description terms

    StereoLithography Interface Specification, 3D Systems, Inc., October 1989
    The compilers of this resource have not been able to locate a copy of the original specification from 3D Systems, Inc. See Useful References, immediately below, for sources with documentation for the format, often by example.
```

Anyway... they are trivial and we could support both, or at least easily not crash on the text one... The text one just starts with "solid"... I'll work on it if I find time.

ruevs, 2021-08-11:
http://paulbourke.net/dataformats/stl/

phkahler, 2021-08-11:
IIRC there are 2 variations on how to encode color, but the one I implemented is more common and works with STLs exported by Solvespace.

phkahler, 2021-08-18:
The first 80 bytes are read all at once, as that's the header for the binary STL files. So we can check if that starts with "solid" and pop up as error, or branch to a text.STL file reader which could share a lot of code with the existing one.

ruevs, 2021-08-19:
> The first 80 bytes are read all at once, as that's the header for the binary STL files. So we can check if that starts with "solid" and pop up as error, or branch to a text.STL file reader which could share a lot of code with the existing one.

My thoughts exactly.

closed 2022-01-27 by phkahler