solvespace/solvespace #1085: Linked STL has no edges but outlined with "Show edges of solid model"

Symbian9, 2021-08-07:
### System information

- **SolveSpace version:** `3.0~` https://github.com/solvespace/solvespace/commit/1b8e1dec652eef7916a94b6c9db404457dacb0db
- **Operating system:** Debian 10.x buster

### Expected behavior

With *"Show edges of solid model"* linked STL geometry should not be shown as *outlined* - instead there should be shown edges (apply removing vertices doubles is needed to avoid double edging) OR there should be no edges & outlines at all.

For show linked STL geometry *outlined* users should use "Show outline of solid model".

### Actual behavior

Actually linked STL geometry has no edges and just outlined with *"Show edges of solid model"* (same way as with "Show outline of solid model").

### Additional information

- STL files - https://github.com/solvespace/solvespace/files/6950016/blender_cube_monkey_stl.zip

[image]

phkahler, 2021-08-08:
You can turn off the silhouette edges with the toolbar on the text window. STLs will behave like a normal model with all entities disabled. There will not be any edge entities, but you can add lines between the vertices it chose to add. STL contains no surface or edge information. It's just triangle soup.

ruevs, 2021-08-08:
This works exactly as for any other model.

The "thick edges" are ones where - from the current point of view - the visible boundary of the model is "hanging in the air" (in other words a ray from the camera direction is tangent to the solid). Such a boundary may be in front of other parts of the solid.

Symbian9, 2021-08-08:
> You can turn off the silhouette edges with the toolbar on the text window.

@phkahler, As you can on my first screenshot *'silhouette edges'* already turned OFF with "Hide outlines of solid model", but imported mesh still outlined with "Show edges of solid model".

So, it seems like a bug.

phkahler, 2021-08-08:
> As you can on my first screenshot _'silhouette edges'_ already turned OFF with "Hide outlines of solid model", but imported mesh still outlined with "Show edges of solid model".

Nope. 8th toolbar icon from the left - 4th from the right. Click that one to make those edges go away.

AFAICT there is no bug or other issue with this one.

Symbian9, 2021-08-08:
> Nope. 8th toolbar icon from the left - 4th from the right. Click that one to make those edges go away.
>
> AFAICT there is no bug or other issue with this one.

Here is simple usecase:

1. Create any sort of solid group;
2. Link STL;

So, how to disable outlines of STL mesh groups, BUT keep edges of solid groups visible?

[image]

ruevs, 2021-08-09:
@Symbian9 I don't understand. You want to be able to show/hide the outlines of linked STL files independently of other solids?

Lets use the attached file for discussion. It is a cube with a cylinder, exported as STL and the STL imported back into the same SLVS file (STL on the right in the screenshot).

[10mmCube5mmCylinder.zip](https://github.com/solvespace/solvespace/files/6952442/10mmCube5mmCylinder.zip)

[image]

Symbian9, 2021-08-09:
>  I don't understand. You want to be able to show/hide the outlines of linked STL files independently of other solids?

No, I mean that there *should be NO outlines* of linked STL shown IF *"Hide outlines of solid models"* is active (even *"Show edges of solid model"* is active too) — actually it looks like a bug, because if *"Hide outlines of solid models"* + *"Show edges of solid model"* are active together there should be shown only edges of solid body and NO any outlines.

It's OK to show *outlines* of inked STL (same way as for solid body) IF *"Show outlines of solid models"* is active.

> Lets use the attached file for discussion.

Lets clarify, what I see: if linked STL has *no edges* (as you mentioned https://github.com/solvespace/solvespace/issues/1085#issuecomment-894722736), how then STL has *hidden otlines* shown dashed with *"Stipple occluded lines"*? (Outlines in SolveSpace are only visible)

phkahler, 2021-08-09:
> @phkahler, As you can on my first screenshot _'silhouette edges'_ already turned OFF with "Hide outlines of solid model", but imported mesh still outlined with "Show edges of solid model".

@Symbian9  No. I specifically told which toolbar icon would make them go away. It does not have the red X in your screenshot.

I am bothered by @ruevs image where the STL version doesn't show some of the hidden lines. Not sure what's going on but none of this should be STL specific.

Symbian9, 2021-08-09:
>  I specifically told which toolbar icon would make them go away.

The only way to hide *'lines'* (*'edges', 'outlines'*) of linked STL is just hide all outlines & edges in model/scene:

[image]

But I'm talking about all othes situations, where its inclear how 'lines' works for linked STL

@phkahler, Here is comparison of *Solid Cube*, `cube.stl` & `monkey.stl` - as you can see `cube.stl` & `monkey.stl` has outlines even "Hide outlines" is active. Much worse, `monkey.stl` has also hidden (stipple) outlines, but `cube.stl` has no such.

[image]

phkahler, 2021-08-10:
**The problem is that STL files are terrible** I see what's happening now. Solvespace is meant to work with surfaces, each of which may be represented by 1 or more NURBS surfaces, each of which may be represented by one or more triangles. You'll notice that a torus created by using Lathe on a circle will produce a single surface even though it is actually composed of 16 NURBS surfaces and possibly thousands of triangle. Many CAD programs will show seams between those NURBS patches, but we don't. I believe that's because all the triangles are ultimately considered part of the same surface (formed by the swept circle).

STL files just contain triangles. There is no meta-information about the triangles like which ones are part of the same surface. For this reason there is no way to distinguish which edges are at the edge of a surface. That is why the edges on the front of the cube are never shown - there is no difference between those edges and the seams between different triangle on a single face. In other words, there is no useful edge information available in the STL files. The only reason the silhouette edges are show is because they are automatically identified - they have to be because they are view dependent, not part of the model. The silhouette edges that are hidden can be displayed stippled as well.

Due to all of that, I'm going to have to tag this one with won't-fix for now.

ruevs, 2021-08-10:
@phkahler bravo! You explained it much better than I did above when I said:

> And as @phkahler explained there are no other edges to show for STL, since it is not "constructed" from geometry.

But hopefully all is clear now - (linked) STLs simply have no "structure" to show apart from the "edges" from the current point of view.

I am closing this.

closed 2021-08-10 by ruevs