solvespace/solvespace #1086: Feature Request: Backwards compatibility for macOS 10.14.5

kryspetrie, 2021-08-08:
### System information

- **SolveSpace version:** 3.0~0e0b0252 (release)
- **Operating system:** macOS 10.14.5

### Expected behavior

Application opens under macOS 10.14.5

### Actual behavior

Popup saying Solvespace is only compatible with macOS 10.15.x

### Additional information

For organizational reasons I cannot upgrade yet would like to still use this software... Is there any way to maintain a single major OS version of backwards compatibility?

Symbian9, 2021-08-08:
Which version are you using (could be run) now on your macOS 10.14.x?

@kryspetrie, please, try @vespakoen's DMG build:

- https://github.com/vespakoen/solvespace/releases/tag/3.0.0-alpha.5

phkahler, 2021-08-08:
@vespakoen  I also have problems with the 3.0 stuff on my 2013 Macbook Air.  Do you know why newer builds don't run on older OSX?  I'd like them to work, but I don't even know why they don't. Any ideas?

kryspetrie, 2021-08-08:
I don't have previous version information since this is a newly provisioned
laptop. I'll try the suggested build and will let you know.

ruevs, 2021-08-15:
@tomsci since you are actively working on the code on macOS can you look into what is required to support old versions? I don't have a Mac (newer than 2008 :-) ), but as far as I can see in the history of [guimac](https://github.com/solvespace/solvespace/commits/master/src/platform/guimac.mm) we should be able to run on very old macOS versions if compiled "properly".

tomsci, 2021-08-15:
On it :)

closed 2021-08-15 by phkahler

ruevs, 2022-01-24:
Now that I have [macOS in a VirtualBox](https://github.com/solvespace/solvespace/issues/1199) I experimented a bit (did a bisect) and found that the [oldest version](https://en.wikipedia.org/wiki/MacOS_version_history#Releases) for which we can compile with Xcode 11.3.1  is [10.9 Mavericks](https://en.wikipedia.org/wiki/OS_X_Mavericks).

It runs fine on Mojave and the test suite (solvespace-testsuite) passes. Now to see if it is possible/easy to install Mavericks in a virtual machine...

Just did `cmake .. -G Xcode -DCMAKE_OSX_DEPLOYMENT_TARGET="10.9"`

With `-DCMAKE_OSX_DEPLOYMENT_TARGET="10.8"` or older even the CMake configuration can not be generated.

PR #1092 (2021-08-15): Set OSX minimum supported version
As per [cmake docs](https://cmake.org/cmake/help/v3.21/variable/CMAKE_OSX_DEPLOYMENT_TARGET.html#variable:CMAKE_OSX_DEPLOYMENT_TARGET) this is how you configure `-mmacosx-version-min` in cmake land.

To my surprise this is all that was needed to get a binary compiled on 11.5.1 to launch on 10.12.6, which is as far back as I can test. 10.12.6 is supported on macs at least as far back as mid 2012.

Fixes #1086.

It did cause some build warnings from libomp:

    ld: warning: dylib (/usr/local/lib/libomp.dylib) was built for newer macOS version (11.0) than being  [... cut]