solvespace/solvespace #1089: github macOS automated tasks are broken

tomsci, 2021-08-12:
The "Test / Test macOS" github check seems to be broken for all PRs, eg [here](https://github.com/solvespace/solvespace/pull/1057/checks?check_run_id=3176323644) and [here](https://github.com/solvespace/solvespace/pull/1088/checks?check_run_id=3311566506). It'd be nice to have them working again, or commented out I guess.

phkahler, 2021-08-12:
Agreed. I think I broke it by merging a PR that wasn't quite ready.

tomsci, 2021-08-15:
It seems like the failure is due to install-macos.sh trying to fetch the libomp bottle directly from bintray.com, which is now returning 403 for that URL, which the script doesn't recognise is an error and tries to extract the resulting HTML error page as if it were the archive. Visiting bintray.com suggests they're no longer mirroring bottles. So this seems to be a case of an external dependency disappearing. I'm not sure there's a reliable way of installing a homebrew bottle without running `brew install`, I assume there's some reason that isn't done in the `ci` condition? It certainly seems like there is an arm64 bottle available [by default](https://formulae.brew.sh/formula/libomp) in brew now, if that was the reason? @vespakoen do you have any thoughts on how best to fix this?

vespakoen, 2021-08-15:
We build the arm64 binary on a x64 macOS CI image, and there is no way to tell homebrew to fetch the arm64 package, it uses the architecture of the computer it's executed on.

That is the reason I had to download the binary from the bintray url, apparently it changed.

I'd suggest downloading the 2 binaries from the new host and hosting them ourselves in the github repository.

tomsci, 2021-08-15:
Ah, that makes sense. Maybe we should just include the libomp source as a submodule and build it from source like libpng etc? It might be less maintenance hassle than pulling binaries from homebrew's CDN? The brew recipe doesn't look too bad. It's a bigish package, but not as big as cairo which is also built from source on mac I think?

tomsci, 2021-08-15:
> there is no way to tell homebrew to fetch the arm64 package

It seems like there is now:

```
brew fetch --bottle-tag=arm64_big_sur libomp
==> Downloading https://ghcr.io/v2/homebrew/core/libomp/manifests/12.0.1
Already downloaded: /Users/tomsci/Library/Caches/Homebrew/downloads/0b103ea0f29dc69fbea8f376b26fdb05fa5fe230b09eccd4366fced640bbeb63--libomp-12.0.1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/libomp/blobs/sha256:9b4d71ac4e8a8b8d04819b1bfd155bcb266a9fdf1405b24c9e3801858b08d8bf
==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:9b4d71ac4e8a8b8d04819b1bfd155bcb266a9fdf1405b24c9e3801858b08d8bf
######################################################################## 100.0%
Downloaded to: /Users/tomsci/Library/Caches/Homebrew/downloads/30c6123bedb6ecf8abe8c43c8d6035223cc0d5892004c44cc051008006cacc07--libomp--12.0.1.arm64_big_sur.bottle.tar.gz
SHA256: 9b4d71ac4e8a8b8d04819b1bfd155bcb266a9fdf1405b24c9e3801858b08d8bf
```

Although I haven't found a way yet to control where that download ends up, or get brew to easily tell you, because `brew --cache libomp` doesn't accept the `--bottle-tag` option :( Might have to scrape it from the `brew fetch` stdout...

ruevs, 2021-08-17:
Half jokingly - fetch it from China ;-)
https://mirrors.nju.edu.cn/homebrew-bottles/libomp-12.0.1.arm64_big_sur.bottle.tar.gz

https://mirrors.nju.edu.cn/homebrew-bottles/

phkahler, 2021-08-18:
Can we stop trying to build the ARM version until it actually works? CI needs to not fail.

vespakoen, 2021-08-18:
Just fixed CI, sorry that this took so long, thanks @tomsci for finding that --bottle-tag option!

ruevs, 2021-08-18:
And the CI for your pull request passed! Thank you!

closed 2021-08-18 by ruevs