solvespace/solvespace #1091: NURBS failure with countersunk holes

tomsci, 2021-08-14:
### System information

- **SolveSpace version:** master @ 56719415dea4de7af750227653362f4c8cad78f8 (3.0)
- **Operating system:** macOS 11.5.1 (Big Sur)

### Expected behavior

Opened [turc-nurbs-1.slvs.zip](https://github.com/solvespace/solvespace/files/6986416/turc-nurbs-1.slvs.zip)
Expected no NURBS issues

### Actual behavior

The 'board-1-inset' group shows "The Boolean operation failed", there are angry red lines around some of the geometry from this group and Analyze->Naked Edges reports "The mesh has naked edges (NOT okay, invalid). 10 problematic edges, bad." On the console I get:

```
failed: I=20, avoid=7
print 4 edges
failed to assemble polygon to trim nurbs surface in uv space
```

### Additional information

The problematic model has a couple of holes through a solid board (called "top-plate" in the model), modelled as a difference extrude ("board-1-holes" being the extrude of the "board-1" sketch). These holes are countersunk by adding a second sketch of slightly-larger holes and a second difference extrude that doesn't go all the way through the board ("board-1-inset-sketch" and "board-1-inset"). I've simplified board-1-inset-sketch to reduce the number of holes, and only countersink the top-left hole, to hopefully make the test case easier. When I tried removing some more construction geometry from "board-1" the problem went away and I'm not sure why.

The overall project that this model is part of quickly degenerated into having to tick "Force NURBS surfaces to triangle mesh" early on and is pretty complex and incredibly slow to recalculate (Generate::ALL takes 30s). I read a comment on another issue suggesting that the triangle meshing might be contributing to the performance issues, hence why I'm raising this to see if I can get the whole thing to work in NURBS.

Symbian9, 2021-08-14:
JFTR, Increasing `chord tolerance (in percents)` to `0,01 %` prevents this NURBS issue.

[image]

tomsci, 2021-08-14:
The correctness of NURBS operations is predicated on the chord tolerance setting? 🤨 I didn't realise that the chord tolerance would affect such things, I assumed it was only when you forced to triangle mesh (and thus started approximating everything) that the chord tolerance became a factor. My mistake.

The problem I have with this as a solution is that it seems to be a massive house of cards that only is guaranteed to work for simple cases... one problem might be solved by increasing it, the next by decreasing it. I'm currently trying to get my mega-horrible-model to work in NURBS and having tinkered with that setting (which fortunately, is a separate assembly) I might actually have it working. The next problem was that NURBS refused to solve an extrude that brushed past the surface of a linked assembly - if I moved the linked assembly to after the extrude had made a surface for it to "sit" on, things are much happier.

It's a little annoying that keeping NURBS happy seems to be highly dependant on doing your operations in the "right" order - even if that isn't the way you'd prefer your model to be composed. (Plus there being no easy way of reordering groups around, so if you've done it wrong, it can be a pain to fix).

At this point I've no idea what is considered "user error" where NURBS is concerned, and what is worth raising an issue for... is "tinkering around until you hopefully find something that resolves" the expected workflow? Until now I'd given up and clicked "Force to triangle mesh" early in the project, but now I can see the huge performance improvements you gain if you can get NURBS working. I'm currently re-learning what works and what doesn't, in light of that...

phkahler, 2021-08-14:
@tomsci  Your frustration is completely justified. It is not obvious that chord tolerance affects these thing, but it does.  It shouldn't but it does. That doesnt mean this is a non-issue - it will be added to the collection of NURBS failures to be fixed.

tomsci, 2021-08-14:
OK thanks. I'm afraid I have another one though, and no amount of fiddling with chord tolerance seems to fix it :/

This time it's when trying to make a curved edge between two boxes by extruding a quarter-circle:

[image]

[turc-nurbs-2.slvs.zip](https://github.com/solvespace/solvespace/files/6986942/turc-nurbs-2.slvs.zip)

I suspect the workaround would be to create the curved piece before one of the other walls, or to make it all one extrusion, but (as is always the case) the original mode would require a lot of rework to achieve that.

ruevs, 2021-08-14:
Strictly speaking this is a mesh bug, not a NURBS one. It is clear from the error message "failed to assemble polygon to trim nurbs surface in uv space". The problem also disappears if you change the extrusion height of `g009-board-1-inset` to 0.6mm.

It starts working with force triangle mesh, because the algorithm is different, just turn on [image] to see the difference. The NURBS meshes (when they work) are drastically better.

ruevs, 2021-08-14:
With NURBS (chord tolerance 0,1% and 0.6mm extrusion) your model is 280 triangles.
With "force NIRBS surfaces to triangle mesh" in the last group it is 488 triangles.

Just one (of possibly many) reasons the second one is slower.

Anyway; it is a bug and I will add it to the tracking issue https://github.com/solvespace/solvespace/issues/738. All the ones with an `m` in the first column are mesh problems.

ruevs, 2021-08-14:
Your second one on the other hand is a typical NURBS bug. You are creating a "singularity" along the edge where the two parallelograms meet. It is a hard problem in general for 3D software, which does not mean we should not try to handle it. Here is an explanation https://github.com/solvespace/solvespace/issues/470#issuecomment-527175273 with links to better explanations in my following posts :-)

tomsci, 2021-08-14:
Thanks, that's really useful info! I think I understand a bit better what not to do :)

ruevs, 2021-08-14:
By the way here is a simplified version of your second case - fewer surfaces, easier to debug in the future ;-)

ruevs, 2021-08-14:
> Thanks, that's really useful info! I think I understand a bit better what not to do :)

By all means do create singularities if they work - in many cases they do - and usually they are quite hard to avoid (construction wise).

Your example is not one of those cases - just make the arc part of the first sketch ;-) - but it is a nice simple example of a very typical failure.

Symbian9, 2021-08-14:
> This time it's when trying to make a curved edge between two boxes by extruding a quarter-circle

I already played a lot with similar complex features found that most (but not all) of NURBS issues *could be solved* with a little bit different *"topology"*.

Here is example of "box with rounded corner" which NURBS does not fails (even with `chord tolerance (in percents)` set to `0.10 %`)
 
- SLVS - [box_rounded_corner.zip](https://github.com/solvespace/solvespace/files/6987150/box_rounded_corner.zip)
  [pic.1](https://i.imgur.com/Ly9CkIk.png)

Here is my showcase of complex solid body (see *handle*) with a lot of rounded corners workarounded by similar tricks:

- https://twitter.com/i/status/1419455483350503424
  [image]
  [image]
- https://twitter.com/i/status/1419477004836036611
  [image]
  [image]

phkahler, 2021-08-15:
@ruevs  I Finally got a chance to try the sketch here. It is difficult for me to speculate why this simple difference might fail. The problematic curve (shown in red) is at the intersection of a curved surface (90 degree portion of the inside of a hole) and a plane. My best guess is that proximity of the top or botton faces is causing some issue - similar to #907. Indeed, changing the depth of the first extrusion to 10mm make the problem go away. Not that it's a solution, just a hint as to what might be going wrong.

BTW chord tolerance of 0.03 percent works for me as well as 0.02 and 0.01.

phkahler, 2021-08-15:
Oh BTW, the idea behind #955 would also allow you to put these holes in the base sketch with a defined counter-bore. It would then create them fully featured at the time of extrusion using no boolean operations. At least that's one way I envision it.

tomsci, 2021-08-18:
That's a really nice model @Symbian9!

Thanks for looking into it @ruevs and @phkahler, I've managed to nurbsify all the rest of my problematic model (soo much nicer mesh and dramatically better performance!) and I think I've found one more problem that cranking up the chord tolerance or tweaking dimensions doesn't make go away, and seems like it might be the same issue of intersecting circular geometry:

[turc-nurbs-3.slvs.zip](https://github.com/solvespace/solvespace/files/7009874/turc-nurbs-3.slvs.zip)
[image]

Notice how the rectangular hole has no problem cutting into the curved edge, whereas the circular hole does. I've no idea what that indicates, but it seems significant :)

phkahler, 2021-08-19:
I suspect the arcs (2 of the 4 making up the top circle) are not getting split by the line where the curved and flat surfaces are tangent. They need to because part of those arcs should be kept and part discarded.

The boolean process includes a step where all curves in one shell are split by surfaces in the other shell. It seems to fail when the curve intersects the surface right on its edge for reasons unknown. I've been meaning to add tests specifically for that situation. We might also try figuring out why it fails to find those intersections.

TLDR; I have some ideas.

phkahler, 2021-08-20:
Thinking about this during my drive (the second failure, not the proximity one). Those arcs are exact curves at the end of a circular cylinder, so they will be surface-intersected with the new exact method rather than using the intersection of 3 surfaces method. That doesn't matter because they would both fail to find the intersection of the curve and the rounded surface because the tangent plane of the surface contains the curve at the point of intersection, no numerically it breaks down there. Those curves do lie in-plane of the top flat surface, but we don't look for intersections between trim curves in that case. So I'm really leaning toward adding trim-trim intersection tests as there are other cases I think that will fix as well and it was fruitful to add e d-point / curve intersections last year.

tomsci, 2021-08-20:
I admit I didn't understand a lot of that but it certainly sounds like a promising line of inquiry :-) Let me know if there's anything I can do to help!

phkahler, 2021-08-22:
Thinking about a general way to find curve-curve intersection. For curves 1 and 2 I was thinking of checking the bounding boxes. If they overlap, split curve 1 in half to form curves 3 and 4 - then recursively check (2 & 3) and (2 & 4). The recursion would put curve 2 as the first one so it would split next time. Then I realized the the curves are coincident for any length, this will result in a massive number of intersection (2^N where N is the recursion depth). So this may relate to identifying overlapping curves (see issue #171) not just intersecting. Again this is for the second issue in this thread, not the title problem.

ruevs, 2021-08-23:
I on the other hand am thinking about the case with the three prisms:
[image]

The problem is that the highlighted edge of the small prism falls on four edges belonging to the extrude with the two big prisms.

At the very least this code is not at all happy since it gets `edge_inters==4`:
https://github.com/solvespace/solvespace/blob/f47cf65f41442760cfb6120f5b897043d55a0def/src/srf/raycast.cpp#L464

And of course this does not seem to take this case into account:
https://github.com/solvespace/solvespace/blob/f47cf65f41442760cfb6120f5b897043d55a0def/src/srf/raycast.cpp#L446

What is more the only place where `bool SShell::ClassifyEdge()` is called does not take the return value into account
https://github.com/solvespace/solvespace/blob/f47cf65f41442760cfb6120f5b897043d55a0def/src/srf/boolean.cpp#L651
@jwesthues do you recall what might have been the intention? Unsurprisingly you are the last one that touched this in ae35b3595cf646b ;-) It seems that `false` means "classification failed" but what to do if it fails? ;-)

I'll keep thinking about this and debugging when I have time.

The model: [NURBS_ThreePrisms.zip](https://github.com/solvespace/solvespace/files/7030785/NURBS_ThreePrisms.zip)

Symbian9, 2021-08-23:
> The model: `NURBS_ThreePrisms.zip`

It seems like there is some sort of  NURBS fail *much earlier just with two prisms with mutual corner edge*, where "corner edge" itself is not recognized as "edge of solid model" (similar to what I mentioned in https://github.com/solvespace/solvespace/issues/1091#issuecomment-898955336)

[image]

phkahler, 2021-08-23:
@Symbian9  That missing edge is issue #1051 and is a display issue. The NURBS surfaces are actually completely intact in that situation.

jwesthues, 2021-08-23:
@ruevs - All the Booleans assume non-degenerate geometry without regions joining along knife edges. Whenever stuff like that works, it's basically by accident. Looks like here I assume that an edge from shell A can intersect shell B at only zero or two of shell B's coincident edges (since edges on a watertight shell always occur in pairs), when in that knife-edge case it could be any even number. It's possible that just picking any pair and classifying as if n=2 will do the right thing. I'm not sure, though, and it's probably easier to try the experiment than to think it through.

When `ClassifyEdge()` fails, the only real solution is for a developer to go back and fix the bug or unimplemented special case. Of course it would be better to debug-print a warning than to just fail silently. It would also be better to either bail out of the Boolean or proceed with an arbitrary fixed guess for the classification, since I suspect some paths now access uninitialized memory for `{in,out}dir_shell`.

The other problems are different, but in the same basic class of unimplemented special cases. As far as I know the nice robust commercial kernels don't do anything much more clever; they've just spent more time grinding out more cases.

jwesthues, 2021-08-23:
> I suspect the arcs (2 of the 4 making up the top circle) are not getting split by the line where the curved and flat surfaces are tangent. They need to because part of those arcs should be kept and part discarded.

@phkahler - I suspect that you're right. The options are either to make the curve-surface intersection work even with the tangency  or to split trim curves against trim curves as you say. The former is painful in general numerically, since anything Newton-like fails; but you could include hand-solved special cases for common surfaces like cylinders and get most of the benefit plus a speedup. For the latter, you could look for near-intersections (within our chord tolerance) on the piecewise linear segments, and then refine any candidates numerically.

Curve-surface intersection is easy when the curve endpoints lie on opposite sides of the surface, since there's no need to introduce any concept of tolerance. Curve-tangent-surface intersection and curve-curve intersection require a tolerance, so they're trickier. Certainly doable, though.

phkahler, 2021-08-29:
As a crude experiment, I tried a basic curve-curve intersection test here:
https://github.com/phkahler/solvespace/tree/curve-inter

I think it actually solves the second issue here sometimes depending on the location of the hole. It obviously needs to be more robust and should also be able to find multiple intersections between a pair of curves. I'm curious what other open NURBS issues it might fix though.

tomsci, 2021-08-29:
@phkahler your edits beat me to it, I just tried it on a bunch of my problem models including the ones I uploaded on this issue. As you say, it seems to give mixed results, for eg it made the rectangular hole in turc-nurbs-3.slvs fail when before it was fine, but I'm very happy that you're looking at it! I tried stepping through boolean.cpp for a bit before reluctantly concluding I have no chance of being able to help with the code, it's too far outside anything I have a clue about :-)

But as I type this the 3D printer sitting next to my desk is churning out the first properly complicated model (>1000 surfaces) which I've done entirely in SolveSpace, so the work you and all the other devs have put in is very much appreciated!

phkahler, 2021-08-30:
@tomsci  dont worry about the parts of the code that are currently out of reach, there are a lot of complex parts here. If you do want to dig into the /srf folder I recommend reading a little about Bezier and NURBS curves and surfaces (without knots, we dont do knots, just weighted control points) then read ratpoly.cpp. That file IMHO is very clean and educational. But there is plenty to be done in the less mathy areas. Enjoy!

jwesthues, 2021-08-30:
@tomsci And I can't imagine anyone understanding that code just by reading it, so don't get discouraged if you can't. The best/only way I've found to debug such operations is by graphical experiment, modifying the the code to identify edges or surfaces of interest and then highlight them on the model.

phkahler, 2021-08-30:
@ruevs and @jwesthues  I see that SBezier::PointOnThisAndCurve() and  SBezier::ClosestPointTo() don't clamp the t-parameter to the range [0, 1] but let it go off the ends. In the hack I posted here, that seems to lead to false solutions in a lot of places. I haven't tried limiting those. It just feels like adding limits in there might break a bunch of other things that allow that to go past (or maybe that's the cause of "proximity" failures).

Either way, I don't think my quick hack is a useful approach. I did confirm that finding curve-curve intersections will resolve some NURBS issues though, so it was worth doing the experiment.

Symbian9, 2021-10-17:
I just discovered one more NURBS failure, but could you point me is it related to actual issue, or there already some other issues, or this something else:

- SolveSpace version: `3.0~`https://github.com/solvespace/solvespace/commit/2cd0ee4b3352c8fb68921399790d0545ad9b26bf
- SLVS-file: [corner-nurbs_slvs3x.zip](https://github.com/solvespace/solvespace/files/7360794/corner-nurbs_slvs3x.zip)

[image]

ruevs, 2026-11-03:
https://github.com/solvespace/solvespace/files/7009874/turc-nurbs-3.slvs.zip from https://github.com/solvespace/solvespace/issues/1091#issuecomment-901393520 is fixed in #1806.

But I noticed that the original https://github.com/solvespace/solvespace/files/6986416/turc-nurbs-1.slvs.zip still has naked edges - both in 3.3 and in #1806. :-(