solvespace/solvespace #1098: CLA Assistant violates SolveSpace licensing

Symbian9, 2021-08-23:
### System information

- **SolveSpace version:** `3.0~master`
- **Operating system:** All supported

### Expected behavior

CLA Assistant should be fully un-linked/removed from pull request checks for SolveSpace repo due to violating SolveSpace licensing, because *"SolveSpace is licensed under the GPLv3 or later and any contributions must be made available under the terms of that license"* 

### Actual behavior

New SolveSpace contributors may not aware that *CLA Assistant Agreement* (Version: 2018-07-13) violates actual SolveSpace licensing (see https://github.com/solvespace/solvespace/issues/943):

- https://cla-assistant.io/solvespace/solvespace
> ...
>
> 2.3 Outbound License
>
> Based on the grant of rights in Sections 2.1 and 2.2, if We include Your Contribution in a Material, **We may license the Contribution under any license, including copyleft, permissive, commercial, or proprietary licenses**. As a condition on the exercise of this right, We agree to also license the Contribution under the terms of the license or licenses which We are using for the Material on the Submission Date.
>
> ... 

Actually (as has already been finally established by @jwesthues in https://github.com/solvespace/solvespace/commit/2fcc933aa9c7d8f30a35d94fdb760b343e4562bf) all contributions to SolveSpace should just "under the GPLv3 or later", to prevent including of SolveSpace source and SolveSpace contributions to any proprietary products.

- https://github.com/solvespace/solvespace/blob/master/CONTRIBUTING.md#licensing
> SolveSpace is licensed under the GPLv3 or later and any contributions
must be made available under the terms of that license.

### Additional information

phkahler, 2021-08-23:
Yeah, does anyone know how to disable that? We're just GPL3 now since things have been merged without the CLA already.

Symbian9, 2021-08-23:
@phkahler, please, look here:

- https://github.com/cla-assistant/cla-assistant/issues/561

Symbian9, 2022-01-10:
@phkahler, please "pin" this issue to make more visible for new contributors.

ppd, 2022-01-10:
Wouldn't it be sufficient to remove the webhook to get rid of it? There  should be one in the GH settings.

Shegox, 2022-01-10:
So there are quite a few ways to disable it: 
The normal way is to log into https://cla-assistant.io/ and just delete the repository there (... -> unlink) , for this you require admin permissions. 
If this doesn't work/ you don't want to do that you can do following:
- As a repository you can remove the webhook under repository -> settings -> hooks: This will prevent new checks from being triggered.
- As an organization administrator you can remove the third-party permissions of the app, preventing CLA-assistant access to your repository. (For details check "[Denying access to a previously approved OAuth App for your organization](https://docs.github.com/en/organizations/restricting-access-to-your-organizations-data/denying-access-to-a-previously-approved-oauth-app-for-your-organization)"
- As the orginal installer (@whitequark) you can remove the application permissions under [your settings -> Applications -> Authorized OAuth Apps](https://github.com/settings/applications), removing the required permissions.

Symbian9, 2022-01-10:
> As the orginal installer (`@whitequark`)

JFYI, @Shegox, `@whitequark` resigned from SolveSpace dev team years ago.

Shegox, 2022-01-10:
In that case, removing the webhook under repository -> settings -> hooks and removing the OAuth App permissions are the way to get rid of it. Unfortunately this is kinda problem with how CLA-Assistant currently uses user tokens for accessing the GitHub API, in future we want to switch to GitHub Apps, which makes it way more clear that there is an app installed and easier to remove it.

phkahler, 2022-01-10:
I disabled the CLA assistant access, but we should not be telling it to do anything in the first place.

phkahler, 2022-01-10:
Update. Also found the webhook and unchecked "Pull Request" which was the only condition set to trigger it.

Symbian9, 2022-01-10:
> I disabled the CLA assistant access
>
> Also found the webhook and unchecked "Pull Request"

@Shegox, Is there a way to mass remove comments from @CLAassistant already posted in exists pulls threads?

Such like this one: https://github.com/solvespace/solvespace/pull/1187#issuecomment-1008674428

Shegox, 2022-01-10:
I'm not aware of a native UI way to do this, but you can leverage the [GH CLI](https://github.com/cli/cli) and API to do this. This small script should do the trick: 
```bash
gh api --paginate /repos/solvespace/solvespace/issues/comments | jq '.[] | select(.user.login=="CLAassistant") | .id' | while read -r line; do gh api -X DELETE /repos/solvespace/solvespace/issues/comments/$line ;done
```

I have only tested in on a very small dummy repo, so best to first get the list of all comments to remove via getting first all comments in the whole repository and then selecting it down to only the ones done by CLAassistant and putting them into a file.
```bash
gh api --paginate /repos/solvespace/solvespace/issues/comments | jq '.[] | select(.user.login=="CLAassistant") | .id' | tee comment-ids.list
```

Now you can spot check them and then mass delete these with
```bash
cat comment-ids.list | while read -r line; do gh api -X DELETE /repos/solvespace/solvespace/issues/comments/$line ;done
```

ruevs, 2022-01-10:
https://github.com/solvespace/solvespace/pull/1188 "worked" - no CLA assistant message any more.

closed 2022-01-10 by ruevs

Symbian9, 2022-01-10:
Thank Y'all for the fix on this issue finally!

phkahler, 2022-01-10:
> Is there a way to mass remove comments from @CLAassistant already posted in exists pulls threads?

This is something we absolutely will NOT do.  There has been a lot of code contributed to solvespace under CLA and there is no reason to remove github comments to that effect.

Symbian9, 2022-01-10:
> There has been a lot of code contributed to solvespace under CLA and there is no reason to remove github comments to that effect.

Since https://github.com/solvespace/solvespace/commit/2fcc933aa9c7d8f30a35d94fdb760b343e4562bf none of pulls/commits could be CLA assigned where are edited GPL-licensed files.