solvespace/solvespace #1103: "Edge" GitHub releases are not useful

ddevault, 2021-08-29:
I am the maintainer of the solvespace package on Alpine Linux and I cannot subscribe to release notifications because I am overwhelmed by notifications for the "edge" versions all of the time.

vespakoen, 2021-08-30:
What package manager are you getting the notifications from?
Are we not tagging our builds correctly as an "edge" / "unstable" release?

ddevault, 2021-08-30:
I am getting the notifications from GitHub. I am the *maintainer* of a solvespace package in a Linux distribution, so when you release an update it's my job to go and update the Alpine package also. I subscribed to GitHub release notifications so I can do my part after solvespace updates, but getting notified every day for the same edge release makes it useless for this purpose.

ppd, 2021-08-30:
Well, we could opt for a solution where we leave the edge tag where it is and replace the edge release's assets each time instead of re-creating the whole thing. @phkahler ?

ppd, 2021-08-30:
You're right; a separate channel would be a nice solution. I could just be another GitHub repository if we want to keep it simple.

phkahler, 2021-08-30:
> Well, we could opt for a solution where we leave the edge tag where it is and replace the edge release's assets each time instead of re-creating the whole thing. @phkahler ?

@ppd  I'd be OK with that or putting them somewhere else (not sure where). I'll leave the details to you.

In contrast to this issue I find the edge builds quite useful. Some people get to confirm fixes without having to build from source. I get to check out the latest macOS versions even though I only build on Linux. But I can see how churn on the releases page could be bad.

vespakoen, 2021-08-30:
I misread the first message, thanks for clarifying!

I would prefer updating the files only, we actually did that in the past but the commit sha1 would not get updated.

It looks like we should be able to change the "target commitish" though looking at the API: https://docs.github.com/en/rest/reference/repos#update-a-release

I'll give it a try...

ppd, 2021-09-02:
> Another option I see is not releasing edge builds on github releases, but just grab them directly from the action artifacts, for example (see bottom of the page): https://github.com/solvespace/solvespace/actions/runs/1179787122

I think you can't download assets from there if you're not a project member.

Symbian9, 2021-09-02:
> I think you can't download assets from there if you're not a project member.

For get Actions artifacts user shold be just logged into GitHub. It's NO need to be a project member.

BTW, I suppose the best solution would be just upload `edge` builds as assets into separate repo:

- https://github.com/solvespace-builds/releases/tag/edge

ppd, 2021-09-02:
There's a nice solution for getting static links to workflow artifacts (and without requiring the user to be logged in): https://nightly.link/

At first glance, it looks exactly like what we need. I'll check it out tomorrow or this weekend.

Symbian9, 2021-09-02:
> If you'll be publishing a link to your own repository's artifacts, please install the GitHub App (https://github.com/apps/nightly-link) anyway, so that downloads for your repositories don't share the global API rate limit.

@ppd, It may fails as I can see due to rate limits.

I suppose it would be better to use [my suggestion](https://github.com/solvespace/solvespace/issues/1103#issuecomment-911672134) with publish artifacts in `solvespace-builds` as assets to `edge` release using next scripts:

- https://github.com/fnkr/github-action-ghr

ppd, 2021-10-29:
Apologies for the delay. Life got in the way.

> @ppd, It may fails as I can see due to rate limits.

Unlikely, I reckon. If we install the nightly.link app (https://github.com/apps/nightly-link), we get something like 5,000 requests/hour (https://docs.github.com/en/developers/apps/building-github-apps/rate-limits-for-github-apps). I doubt this will ever be a problem.

I think we should just

1. install the nightly.link GitHub app
2. add those links to the README.md (a bit nicer, of course):
[image]
3. Drop the edge release altogether

This is rather trivial, has few edge cases (pun not intended), and requires very little development & testing. I will create a PR.

PR #1137 (2021-10-29): CI: Replace edge releases with links to artifacts
Re-creating the edge release for every push
to master creates many superfluous release notifications.

Stop creating those releases and provide users with direct
links to the workflow artifacts instead via the
nightly.link GitHub app (https://github.com/apps/nightly-link).

Fixes #1103
commit eb17248b (2021-10-29): CI: Replace edge releases with links to artifacts

Re-creating the edge release for every push
to master creates many superfluous release notifications.

Stop creating those releases and provide users with direct
links to the workflow artifacts instead via the
nightly.link GitHub app (https://github.com/apps/nightly-link).

Fixes #1103
closed 2021-10-29 by phkahler