solvespace/solvespace #1105: Cannot extend reference line in simple sketch

ardugitter, 2021-09-03:
### System information

- **SolveSpace version:** 
3.0~0a3504c3

- **Operating system:** 
Debian stable

### Expected behavior
Should be able to change the length of the 1000mm line to 4000mm.

### Actual behavior

Solvespace says that it is a redundant constraint even though all I have done is extend a reference line with no connection on the end.

### Additional information

Please see attached .zip. Simply change the 1000mm line to 4000mm length and see the problem.

ardugitter, 2021-09-03:
[problem.zip](https://github.com/solvespace/solvespace/files/7106553/problem.zip)

phkahler, 2021-09-03:
It's not a line "with no connection on the end" as both its ends are constrained to other things. One is tied to the midpoint of a line and the other is coincident with 2 lines end points. Having said that, the sketch is not fully constrained and the line (along with some others) can be extended. I was able to go from 1000 -> 1200 -> 2000 -> 2500 -> 3000 etc. But it eventually required smaller steps and I couldn't get past 3629.

This seems to be some kind of constraint solver issue which is outside my territory.

phkahler, 2021-09-21:
This might be related to #225 if those changes really do improve the convergence of the solver and not just the speed.

ruevs, 2024-02-07:
https://github.com/solvespace/solvespace/issues/246, for cross reference.