solvespace/solvespace #1106: Mac Magic Mouse scroll gestures are misinterpreted

tomsci, 2021-09-05:
The mac Magic Mouse has a touch-sensitive surface instead of scrollwheel, and its 'scroll' gesture is misinterpreted as a trackpad pan by the logic I added in #1093.

_Originally posted by @nanoant in https://github.com/solvespace/solvespace/issues/1093#issuecomment-913132466_

tomsci, 2021-09-05:
@nanoant can you try using this branch https://github.com/tomsci/solvespace/tree/tomsci-scroll-magic-mouse to see if it fixes the issue for you? I've changed the logic so that single-finger gestures on the Magic Mouse should go back to being interpreted as scroll gestures, hopefully even if you still have another finger resting on the surface of the mouse (ie one moving finger, and any number of stationary fingers, should still scroll and not pan). Moving two fingers will still pan, there isn't any way I can see to stop that without completely refactoring the pan logic, the mac is determined to make it seem indistinguishable from a trackpad.

nanoant, 2021-09-05:
@tomsci It works well for me. Is two finger dragging is good for you to pan, then I think this is a good compromise.

FYI unfortunately it seems there's absolutely no way to distinguish `NSEvent` sent from track-pad and Apple Magic Mouse. See https://developer.blender.org/T28092#97274 - it seems they generate exactly the same events. Theoretically we could query `IOHIDManager` to see which kind of touch device it is, but it looks like a overkill to me.

tomsci, 2021-09-05:
That's great to hear, thanks for trying it out!

> is two finger dragging is good for you to pan

It seems a reasonable behaviour to me, although since I don't have a Magic Mouse in front of me I haven't been able to actually try it out to see whether I like it or not.

As you say though, I don't think we have any choice. I didn't find any way to distinguish a two-finger trackpad gesture from a two-finger Magic Mouse gesture either. The only option would be to either revert the trackpad pan gesture support entirely (which I'd rather not do, I think it's overall an improvement) or reimplement it in terms of raw touch events (which would be a huge pain, requiring a reeimplementation of kinetic scrolling, and generally reinventing the wheel). So I think I'd prefer this workaround, which is relatively straightforward :-) I really wouldn't want to try second-guessing what the events are based on HIDmanager stuff, that sounds like a rabbit hole one could get lost in.

nanoant, 2021-09-05:
> As you say though, I don't think we have any choice. I didn't find any way to distinguish a two-finger trackpad gesture from a two-finger Magic Mouse gesture either.

Well, there's always a way. See https://github.com/nanoant/solvespace/commit/3c04016bf85293a888c79ac1ded516898835e496 Shall I make a PR?

tomsci, 2021-09-06:
@nanoant I think it's an interesting bit of code (I'll definitely bookmark it for future reference) but not really the right thing to do here. For one, if I've understood it right it would prevent trackpad gestures from being recognised _on the trackpad_ if there happened to also be a Magic Mouse connected to that machine? Because there's still no way to associate the event with the device that it originated from? Unless the one-finger filtering in this PR leads to some really unpleasant interaction on the Magic Mouse I think it'd be better to allow it to behave similarly and keep the amount of special-casing code to a minimum.

nanoant, 2021-09-06:
@tomsci I absolutely agree and understand that solution that enumerates connected devices is far from something acceptable. It is a pity that Magic Mouse and trackpads are undistinguishable at `NSEvent` though. Nevertheless it was an interesting experiment.

phkahler, 2021-09-06:
Isn't the issue that trackpad scroll is being interpreted as pan?

nanoant, 2021-09-07:
I think this is other way around. https://github.com/solvespace/solvespace/pull/1093/commits/cabe114bff90168f379fe75ef83e11d4df8271b5 adds special condition in `- (void)scrollWheel:(NSEvent *)nsEvent` that was so far triggering zoom, that is now instead triggering panning via adding `event.button = MouseEvent::Button::RIGHT;` when event is flagged as `nsEvent.subtype == NSEventSubtypeTabletPoint`.

It assumed that `NSEventSubtypeTabletPoint` is sufficient to tell that event came from trackpad. Unfortunately Magic Mouse that is a mouse, had a trackpad on the top instead scrolling wheel, and its wheel scroll events are also flagged with `NSEventSubtypeTabletPoint`.

phkahler, 2021-09-07:
According to this page: https://support.apple.com/en-us/HT204895
sliding two fingers up or down on the trackpad is meant to scroll. We are not doing that, we're panning. We use scrolling to zoom, but on my MBA there doesn't seem to be a way to zoom with the trackpad. It's way better than the recent changes, but I don't think its working the way Apple suggests it should in all cases.

tomsci, 2021-09-07:
> Isn't the issue that trackpad scroll is being interpreted as pan? 

That was the intention of #1093 :-) 

> sliding two fingers up or down on the trackpad is meant to scroll. We are not doing that, we're panning. 

Well, "scroll" is a bit ambiguous in this context. Sliding 2 fingers up or down on the trackpad is generally meant to move the document up and down, which I've tried to be consistent in calling a "pan" gesture (given UIKit has `UIPanGestureRecognizer` but Apple themselves are pretty inconsistent in terminology here) to distinguish from a scroll (wheel) event from an actual mouse. Often the scroll wheel also initiates an up-and-down pan, but not always, such as in the SolveSpace case where it's used to zoom. The fact that on the mac a 2 finger up-and-down gesture is by default treated the same as a scroll _wheel_ event, is what let to to that gesture being interpreted as a zoom rather than a move-the-document-up-and-down event. And that's what I changed in #1093, so that 2-finger-up-and-down (and side-to-side etc), 2-finger-pinch, and 2-finger-rotate all behave the way any standard mac app like Safari or Preview handles them.

> We use scrolling to zoom, but on my MBA there doesn't seem to be a way to zoom with the trackpad

You should be able to use the standard 2-finger pinch gesture to zoom in and out?

phkahler, 2021-09-07:
> You should be able to use the standard 2-finger pinch gesture to zoom in and out?

Oops. You are right, and I knew that somewhere in the dark recesses of my mind when I said otherwise.

PR #1107 (2021-12-10): mac: Don't interpret single-touch scroll events as pan gestures
To handle the Magic Mouse, whose single-finger touch gestures should be interpreted as scrollwheel events rather than pan gestures.

fixes #1106

@vespakoen does this look ok to you? Having to track the current number of active touches as an independent bit of state is slightly messy, but it does seem to do the trick without breaking actual trackpad support. Providing someone doesn't try zooming with a Magic Mouse single-finger gesture at the same time as panning with two fingers on a trackpad, I suppose...
closed 2021-12-10 by phkahler