solvespace/solvespace #1108: Assembly fit fails at tangent arcs despite of same dimensions

nanoant, 2021-09-05:
### System information

- **SolveSpace version:** 3.0~6bc63e9 (latest master as of today)
- **Operating system:** macOS 10.15.7 (19H1323)

### Expected behavior

When constructing enclosure over assembly part using the same dimensions as assembly part there should be no fit errors.

### Actual behavior

Assembly is reporting fit problem of the enclosure using tangent arcs at the corners, despite tangent arcs have exactly same dimensions. It is very likely some numerical issue.

[image]

### Additional information

See attached zip file, open `FT232BL_Case.slvs` that uses linked `FT232BL.slvs`.

[FT232BL_Case.zip](https://github.com/solvespace/solvespace/files/7111925/FT232BL_Case.zip)

phkahler, 2021-09-05:
@nanoant  does changing chord tolerance have any effect? How about changing the dimensions slightly to avoid "perfect" fit - leave a tiny gap.

nanoant, 2021-09-05:
@phkahler Nope. It seems increasing or decreasing it has no effect on fit. It obviously affects the numer of triangles/vertices used for rendering curves though.

Btw. I am surprised why SS is not finding a problem that linked assembly is piercing the wall of newly built solid.

[image]

Additionally it looks like boolean operations apply also to linked "assembly". Is there any way to prevent boolean operation from applying on certain solids within single file?

Here I am linking `Board` as an assembly into `Case` file. I try to model case dimensions according to PCB and properly place holes. Unfortunately boolean operations also affect the linked Board, e.g. USB plus from `Board` is also cut by difference.

[image]

I hope I don't need to model `Board`, `Case` (without linking to Board) and finally a `Board+Case` assembly. This would make it much more complicated.

Symbian9, 2021-09-05:
> does changing chord tolerance have any effect?

@phkahler, nope, changing tolerance does not help. Additionally according assembly design I may reproduce such issue as single part - as for me there is NURBS in coincident surfaces (especially circles with equal radius)

- Here is my minimal testcase on this issue SLVS: [nurbs-union-issue.zip](https://github.com/solvespace/solvespace/files/7112079/nurbs-union-issue.zip)
  [image]

>  How about changing the dimensions slightly to avoid "perfect" fit - leave a tiny gap.

I guess that beside SolveSpace NURBS issue, there is core issue in its topology where case body designed as `union` group without fills between 'green PCB' border - as a result NURBS  coincidents.

@nanoant, as for me, your design should re-designed a little bit to avoid coincidence of NURBS face (if you are interested, I may point you out how to re-design your SLVS-assembly design).

FTR, Switching in `FT232BL_Case.slvs` > `g009-extrude` group type from `union` to `assemble` *may solve this NURBS issue* (partially), BUT that would be "dirty hack" in actual design, so I highly recommend to create case in separate file WITHOUT linking `FT232BL.slvs` (aka PCB) only after that join both `FT232BL_Case.slvs` & `FT232BL.slvs` into newly created `FT232BL_asm.slvs`

[image]

nanoant, 2021-09-05:
> I highly recommend to create case in separate file WITHOUT linking FT232BL.slvs (aka PCB) only after that join both FT232BL_Case.slvs & FT232BL.slvs into newly created FT232BL_asm.slvs

@Symbian9 Oh, this is what I was afraid of actually. This is of course doable, but then I need to model holes in `FT232BL_Case.slvs` without any point of reference (in void). I presume I'd need to leave few DOF for holes and constrain them in final `FT232BL_asm.slvs`, right?

Symbian9, 2021-09-05:
> I presume I'd need to leave few DOF for holes and constrain them in final `FT232BL_asm.slvs`, right?

No. *SolveSpace* does not support editing entities with free DOF from linked SLVS-part inside SLVS-assembly (*yet*: https://github.com/solvespace/solvespace/issues/80).

@nanoant, May you draw some basic image (or show photo) of expected resulting assembly for this "dongle"? Then I may recommend the best solution.

nanoant, 2021-09-05:
Well I already did that case in Fusion 360 and 3D printed it.

[image]

I first measured the physical board with caliper. Then I modeled parts like PCB, connectors. Finally I created the enclosure on top of that with constraints, e.g. for distance to the wall and for holes. And I was not using any extra margins, so it was really tight fit - as I wanted, and it worked just great as my 3D printer produces pretty accurate prints in terms of dimensions.

Since I really love the speed, application size and UI/navigation of SolveSpace, I wanted to see if I can somehow recreate similar workflow I used in Fusion.

I ended up linking `FT232BL_Board.slvs` to `FT232BL_Case.slvs` and it "almost" worked for me, except that my tight fit was make fit test fail (hence this issue report) and moreover I could not stop SolveSpace from applying boolean operations on linked part, which was cutting connectors in half.

This does not stop me from producing valid results (in terms of dimensions) for designed case though. However it prevents me from having nice 3D visualization of the final product.

phkahler, 2021-09-05:
When building parts you should not use the assembly operation for a group. When making an assembly (linking a file), always combine as assembly.

Aside from that, solvespace does have some trouble with coincident surfaces, particularly curved ones.

Symbian9, 2021-09-05:
> I could not stop SolveSpace from applying boolean operations on linked part

Okay, here is a quick solution I found basing on initial `FT232BL_Case.zip` design:

- After linking `FT232BL.slvs` (e.g. board, where all groups was `shown` when file saved) inside `FT232BL_Case.slvs` just **suppress** `g003-FT232BL` group, BUT keep `g003-FT232BL` group `shown` in "Property Browser" — this way you would be able to use `g003-FT232BL` entities as reference wireframes for case design while there *should not* be NURBS issues as linked solid is suppressed;
  [image]
- When case design would be complete hide `g003-FT232BL` group and all other unnecessary groups (but keep `g001-#references` group `shown`) in "Property Browser" and save it (e.g. `FT232BL_Case.slvs`);
> **NOTE**: SolveSpace behavior on saving SLVS files is to preserve actual `shown` group entities keep visible after linking it inside other SLVS assembly files, so I for preparing part for linking into assembly SLVS I may recommend left `shown` only `g001-#references` group and then save it.
- Open `FT232BL.slvs` file hide all groups (just left `shown` only `g001-#references` group) and re-save it as `FT232BL_Board.slvs`
- Create new file for assembly, link `FT232BL_Board.slvs` and `FT232BL_Case.slvs` inside it (where each of linked group should be set to `assemble`), constraint each group position & orientation (using reference point & XYZ-normals) and save it as `FT232BL_asm.slvs`.
- Here is **a set of SLVS files** for resulted assembly: [FT232BL_asm.zip](https://github.com/solvespace/solvespace/files/7112363/FT232BL_asm.zip)

  [image]

<details><description>JFTR, Here is an alternative way to assembly:</description>
0. As `FT232BL_Board.slvs` is main part of assembly, I would not describe how to design it here, BUT recommend to hide all unnecessary groups before save it as SLVS-file (keeping `shown` only `g001-#references` would be enough for constrain its position & orientation as linked part in resulted assembly);
> NOTE: PCB-board could be designed itself as assembly of PCB plate & many separate PCB components where each is a separate SLVS file. But in actual situation lets keep it as single SLVS with all PCB components as internal solid groups.
1. Board reference creating:
1.1. Open `FT232BL_Board.slvs`;
1.2. In all `sketch-in-plane` groups "Toggle Construction" on all entities in style `s001-#def-active-grp`  (so, all should be in `s002-#def-construction` style);
1.3. Set `shown` only solid groups (e.g. uncheck all `sketch-in-plane` groups in "Property Browser");
1.4. Save as `FT232BL_Board_ref.slvs` (it would be used as linked wireframe reference for `FT232BL_Case.slvs`): 
2. Case design:
2.1. Create new file for design solid case (would be saved as `FT232BL_Case.slvs`)
2.2. Link `FT232BL_Board_ref.slvs` and constraint its position & orientation (would appear as `g003-FT232BL-Board-ref` group in "Property Browser");
2.3. Design case basing on linked wireframe & save file as `FT232BL_Case.slvs` (just hide `g003-FT232BL-Board-ref` in "Property Browser" before file saving operation);
3. Assembly design:
3.1. Create new file (which would be `FT232BL_asm.slvs`);
3.2. Link `FT232BL_Case.slvs` and constrain its position & orientation (group type should be `assemble`);
3.3. Link `FT232BL_Board.slvs` and constrain its position & orientation (group type should be `assemble`);
3.4. Save assembly file as `FT232BL_asm.slvs`.
</details>

N.B. I myself have a "feature request" that should help in such situations, and also should prevent applying boolean ops to linked STL mesh too, but that is a thing for another issue thread (would create it tomorrow).

Symbian9, 2021-09-05:
> When building parts you should not use the assembly operation for a group

@phkahler, to be clear, its possible (in comment above I just described how to avoid NURBS issues) & in some cases it may be useful (even linking SLVS with just "flat sketches"), but that workflow *"just is not recommended until you know what you are doing"* ;-)