solvespace/solvespace #1110: 16-bit color PNG gets imported as corrupted

anymouse404, 2021-09-12:
### System information

- **SolveSpace version:** 3.0~0e0b0252
- **Operating system:** Ubuntu 20.04

### Expected behavior

16-bit color picture imports correctly in the same way as the 8-bit (both attached)

### Actual behavior

The 16-bit colored image has weird colors, line artifacts and is cut in half (pictured right in the following picture:)
[image]

Here are the 8- and 16-bit files:

[test-8-and-16-bit-png.zip](https://github.com/solvespace/solvespace/files/7149368/test-8-and-16-bit-png.zip)

Symbian9, 2021-09-14:
Issue probably related to `libpng` (as SolveSpace uses it for import PNG):

- https://stackoverflow.com/questions/51309691/read-16-bit-png-using-libpng

phkahler, 2021-09-14:
> @phkahler, any ideas?

Don't use 16bit PNG files?  Seriously.  But it looks like a valid bug.

anymouse404, 2021-09-14:
Tbh, I feel like an error message would be plenty sufficient solution.

Symbian9, 2021-09-14:
>  I feel like an error message

Something like next?

```
Error
 
Image is broken or this format is not supported.
Convert image to 8-bit PNG using image editor.

[OK]
```

PR #1181 (2022-01-06): Load 16bit PNG images correctly by re-scaling to 8bit
Fixes #1110
commit 3136493a (2022-01-06): Load 16bit PNG images correctly by re-scaling to 8bit

Fixes #1110
closed 2022-01-06 by ruevs